realsenseai/librealsense #12421: RealSense Viewer crashes after many window size changes

ionutnechita-intel, 2023-11-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Ubuntu 22.04 IOT | 
| Kernel Version (Linux Only)    |  5.15.0-1043-intel-iotg                                        | 
| Platform                                 | Industrial PC ASRock RPL-P  |
| SDK Version                            |  2.54.2-0~realsense.10773                          | 
| Language                            |  Default/C/C++/Python                   | 
| Segment			| Robot                         | 

Error in console:
...
```
 17/11 12:47:06,381 INFO [139707347527552] (rs.cpp:2697) Framebuffer size changed to 964 x 586
 17/11 12:47:06,381 INFO [139707347527552] (rs.cpp:2697) Window size changed to 964 x 586
 17/11 12:47:06,414 INFO [139707347527552] (rs.cpp:2697) Framebuffer size changed to 862 x 486
 17/11 12:47:06,414 INFO [139707347527552] (rs.cpp:2697) Window size changed to 862 x 486
 17/11 12:47:06,466 INFO [139707347527552] (rs.cpp:2697) Framebuffer size changed to 872 x 292
 17/11 12:47:06,467 INFO [139707347527552] (rs.cpp:2697) Window size changed to 872 x 292
 17/11 12:47:06,507 INFO [139707347527552] (rs.cpp:2697) Framebuffer size changed to 844 x 65507
 17/11 12:47:06,507 INFO [139707347527552] (rs.cpp:2697) Window size changed to 844 x 65507
X Error of failed request:  BadAlloc (insufficient resources for operation)
  Major opcode of failed request:  146 ()
  Minor opcode of failed request:  7
  Serial number of failed request:  190193
  Current serial number in output stream:  190202
^C
:~$ sudo -E realsense-viewer
 17/11 12:47:25,903 INFO [139767681857408] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.0/video4linux/video0
 17/11 12:47:25,903 INFO [139767681857408] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.0/video4linux/video1
 17/11 12:47:25,903 INFO [139767681857408] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.0/video4linux/video2
 17/11 12:47:25,903 INFO [139767681857408] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.0/video4linux/video3
 17/11 12:47:25,903 INFO [139767681857408] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.3/video4linux/video4
 17/11 12:47:25,903 INFO [139767681857408] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.3/video4linux/video5
 17/11 12:47:26,399 INFO [139767681857408] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=1
 17/11 12:47:26,399 INFO [139767681857408] (synthetic-stream-gl.cpp:89)  0 GPU objects initialized
 17/11 12:47:26,399 INFO [139767681857408] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
 17/11 12:47:26,479 INFO [139767681857408] (synthetic-stream-gl.cpp:146)  0 GPU objects initialized
X Error of failed request:  BadAlloc (insufficient resources for operation)
  Major opcode of failed request:  146 ()
  Minor opcode of failed request:  7
  Serial number of failed request:  309
  Current serial number in output stream:  312
^C
...
```

[realsense-viewer-bug.txt](https://github.com/IntelRealSense/librealsense/files/13390944/realsense-viewer-bug.txt)

MartyG-RealSense, 2023-11-17:
Hi @ionutnechita-intel  Is your ASRock in a "headless" configuration that does not have a display screen attached?  If it is headless then RealSense programs that use graphics, such as realsense-viewer, require a display to be attached or errors will occur.

ionutnechita-intel, 2023-11-17:
This was happened when using display screen attached.

MartyG-RealSense, 2023-11-17:
I have not previously seen this **X Error of failed request:  BadAlloc (insufficient resources for operation)** error in relation to RealSense.  However, the link below suggests that it may occur because of X11 video output.  

https://forums.debian.net/viewtopic.php?t=32259

I note that in your new question at https://github.com/IntelRealSense/librealsense/issues/12422 you are using X11 forwarding.

ionutnechita-intel, 2023-11-17:
Yes, but this ticket is used only display screen attached.

And #12422, is use only SSH connection.

Is two ticket with different behaviours.

MartyG-RealSense, 2023-11-17:
With industrial PCs with an "exotic" hardware / software configuration that differs from a typical consumer PC, it is recommended to try building the librealsense SDK from source code with CMake with the flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction.  This causes the SDK to bypass the kernel, so that the SDK is not dependent on Linux versions or kernel versions and does not require kernel patching.

ionutnechita-intel, 2023-11-17:
Good. Will try to check with this. 
But issue is not because have specific hardware, is only software.
RealSense Viewer interface maybe not working with high resolution.

MartyG-RealSense, 2023-11-17:
It may be worth updating the ASRock's video drivers to ensure that it has up to date OpenGL drivers for the Viewer to use.

closed 2023-11-29 by MartyG-RealSense