realsenseai/librealsense #12422: RealSense Viewer have more frames drops per Second when is started with X11 Forward SSH session

ionutnechita-intel, 2023-11-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Ubuntu 22.04 IOT | 
| Kernel Version (Linux Only)    |  5.15.0-1043-intel-iotg                                        | 
| Platform                                 | Industrial PC ASRock RPL-P  |
| SDK Version                            |  2.54.2-0~realsense.10773                          | 
| Language                            |  Default/C/C++/Python                   | 
| Segment			| Robot                         | 
| Connection                    |   SSH with X11 Forward activated |

Error in console:

```
...
 17/11 13:20:45,525 INFO [140554750441344] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.3/video4linux/video4
 17/11 13:20:45,525 INFO [140554750441344] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.3/video4linux/video5
 17/11 13:20:45,525 INFO [140554750441344] (context.cpp:336) Found 1 RealSense devices (mask 0xfe)
 17/11 13:20:59,684 INFO [140554750441344] (sensor.cpp:1594) Request: RGB8 Color,
Resolved to: YUYV Color,
 17/11 13:20:59,685 INFO [140554750441344] (backend-v4l2.cpp:2067) Video node was successfully configured to YUYV format, fd 9
 17/11 13:20:59,685 INFO [140554750441344] (backend-v4l2.cpp:2069) Trying to configure fourcc YUYV
 17/11 13:20:59,685 INFO [140554750441344] (backend-v4l2.cpp:2270) Metadata node was successfully configured to D4XX format, fd 13
 17/11 13:21:01,460 INFO [140554750441344] (sensor.cpp:1594) Request: Z16 Depth,
Resolved to: Z16 Depth,
 17/11 13:21:01,461 INFO [140554750441344] (backend-v4l2.cpp:2067) Video node was successfully configured to Z16  format, fd 14
 17/11 13:21:01,461 INFO [140554750441344] (backend-v4l2.cpp:2069) Trying to configure fourcc Z16
 17/11 13:21:01,461 INFO [140554750441344] (backend-v4l2.cpp:2270) Metadata node was successfully configured to D4XX format, fd 17
 17/11 13:21:02,287 WARNING [140551210899008] (hdr-merge.cpp:38) HDR Merge filter cannot process frames because relevant metadata params are missing
 17/11 13:21:07,657 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:07,690 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:07,723 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:07,756 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:07,790 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:07,823 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:07,856 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:08,159 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:08,159 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Error occured during execution of the processing block! See the log for more info
 17/11 13:21:25,836 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:25,869 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:25,903 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:38,879 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:38,912 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:38,945 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:38,978 ERROR [140551219291712] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:39,381 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:39,381 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Error occured during execution of the processing block! See the log for more info
 17/11 13:21:39,414 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:39,414 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Error occured during execution of the processing block! See the log for more info
 17/11 13:21:39,448 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:39,448 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Error occured during execution of the processing block! See the log for more info
 17/11 13:21:39,481 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:39,481 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Error occured during execution of the processing block! See the log for more info
 17/11 13:21:39,514 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:39,514 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Error occured during execution of the processing block! See the log for more info
 17/11 13:21:39,548 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:39,548 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Error occured during execution of the processing block! See the log for more info
 17/11 13:21:39,581 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 17/11 13:21:39,581 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Error occured during execution of the processing block! See the log for more info
 17/11 13:21:39,614 ERROR [140551210899008] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
...

```

Photo: [image]

File: [capture1-file-terminal-realsense-ssh-x11-forward-session.txt](https://github.com/IntelRealSense/librealsense/files/13391080/capture1-file-terminal-realsense-ssh-x11-forward-session.txt)

MartyG-RealSense, 2023-11-17:
Hi @ionutnechita-intel  How does it perform if forwarding is disabled, please?

ionutnechita-intel, 2023-11-17:
Without X11 Forwarding is not possible to open new graphical interfaces.

MartyG-RealSense, 2023-11-18:
I conducted some research into X11 forwarding over SSH and the problem of slowness over the SSH connection is reported frequently.  Some articles suggested enabling compression in the SSH instruction with a capital **-C** flag when handling a high volume of data (like the high volumes that RealSense cameras can transmit).

https://www.cyberciti.biz/faq/speeding-up-ssh-x11-forwarding-with-unix-osx-linux-bsd/

ionutnechita-intel, 2023-11-20:
Hello @MartyG-RealSense,

With compression enabled, i have approximately 4 frame drops per seconds.
With compression disabled, i have approximately 6 frame drops per seconds.

MartyG-RealSense, 2023-11-20:
If you click the small 'i' button at the top of the depth and RGB stream panels to overlay real-time information on the panel, what is the FPS speed shown?

ionutnechita-intel, 2023-11-20:
I view at both stream this value: 29.98 FPS.

MartyG-RealSense, 2023-11-20:
If your streams are set to 30 FPS and the live information is showing an FPS of 29.98 then the streams should actually be fine despite the warning messages in the log, and the messages can be disregarded.  The 'frame drops per second' graph can also usually be disregarded as it is not a useful measure for analyzing performance.

MartyG-RealSense, 2023-11-26:
Hi @ionutnechita-intel  Do you require further assistance with this case, please?  Thanks!

ionutnechita-intel, 2023-11-27:
No, is fine for me.

Is good to known this behaviour in documentation page.

Thank you.

closed 2023-11-27 by MartyG-RealSense