realsenseai/librealsense #12429: Double serial number

alejandronespereira, 2023-11-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i| 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  6.20.0-36-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.0                          | 
| Language                            |  None| 
| Segment			|  others                          | 

### Issue Description
When I run the realsense-viewer, I get a serial number for my camera that differs from the one that dmesg is giving me. 

realsense-viewer: 047322071071
dmesg log:
```
[ 2208.170989] usb 2-1.4.3.1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[ 2208.170993] usb 2-1.4.3.1: SerialNumber: 047423051028
```

I am using the one from the realsense-viewer on my application using the SDK so I guess it's a dmesg problem, but I was curious about whether someone else has found the same issue or if there is an easy explanation I am missing.

ionutnechita-intel, 2023-11-20:
Please verify Asic Serial Number in realsense-viewer and compare with dmesg command.
Because this is a real serial number.

MartyG-RealSense, 2023-11-20:
Hi @alejandronespereira  You can find the camera serial number using the **Info** option near the top of the RealSense Viewer's options side-panel.  It is also on a sticker on the base of the camera beside the large tripod thread-hole.

alejandronespereira, 2023-11-20:
You are right, there are two serial numbers:

- 047322071071: Seen in
  - rs-enumerate-devices under Serial Number
  - realsense-viewer under Serial Number
  - Sticker on the camer
- 047423051028: Seen in
 - rs-enumerate-devices under Asic Serial Number and Firmware Update Id
 - dmesg message
 - realsense-viewer under Asic Serial Number and Firmware Update Id
 
I am using the first one in my launch files (for the ros2 wrapper). What is the difference between them?

MartyG-RealSense, 2023-11-20:
A RealSense 400 Series camera is made up of two circuit boards, the **depth camera module** and the **Vision Processor D4 Board**.  The ASIC refers to the Vision Processor D4 chip.  The other serial number (the one on the sticker) is used as the unique identification code for the camera.

alejandronespereira, 2023-11-20:
I see, thanks for the clarification.

closed 2023-11-23 by alejandronespereira