realsenseai/librealsense #12433: realsense-viewer launched with no errors but cannot display components correctly

Yinshideguanghui, 2023-11-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | **Cannot** Open RealSense Viewer --> Click info | 
| Operating System & Version |   Ubuntu 20  | 
| Kernel Version (Linux Only)    |  5.15.0                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2                     | 

### Issue Description
I followed the "Installing the packages" at https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md to install librealsense, but when I launched `realsense-viewer` from a terminal, I get a huge button (even not whole) and nothing more.
[image]
logs of terminal:
```
 20/11 21:33:38,717 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
 20/11 21:33:38,718 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video1
 20/11 21:33:38,718 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2
 20/11 21:33:38,718 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video3
 20/11 21:33:38,718 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video4
 20/11 21:33:38,718 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video5
 20/11 21:33:38,885 INFO [139745807927168] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=1
 20/11 21:33:38,885 INFO [139745807927168] (synthetic-stream-gl.cpp:89)  0 GPU objects initialized
 20/11 21:33:38,885 INFO [139745807927168] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
 20/11 21:33:38,899 INFO [139745807927168] (synthetic-stream-gl.cpp:146)  0 GPU objects initialized
 20/11 21:33:38,971 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
 20/11 21:33:38,971 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video1
 20/11 21:33:38,971 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2
 20/11 21:33:38,971 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video3
 20/11 21:33:38,971 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video4
 20/11 21:33:38,971 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video5
 20/11 21:33:38,971 INFO [139745807927168] (context.cpp:336) Found 1 RealSense devices (mask 0xff)
 20/11 21:33:38,987 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
 20/11 21:33:38,987 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video1
 20/11 21:33:38,987 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2
 20/11 21:33:38,987 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video3
 20/11 21:33:38,987 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video4
 20/11 21:33:38,987 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video5
 20/11 21:33:38,987 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
 20/11 21:33:38,988 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video1
 20/11 21:33:38,988 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2
 20/11 21:33:38,988 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video3
 20/11 21:33:38,988 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video4
 20/11 21:33:38,988 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video5
 20/11 21:33:38,989 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
 20/11 21:33:38,989 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video1
 20/11 21:33:38,989 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2
 20/11 21:33:38,989 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video3
 20/11 21:33:38,989 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video4
 20/11 21:33:38,989 INFO [139745668499200] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video5
 20/11 21:33:39,005 INFO [139745807927168] (rs.cpp:2697) Framebuffer size changed to 2048 x 1349
 20/11 21:33:39,005 INFO [139745807927168] (rs.cpp:2697) Window size changed to 64 x 42
 20/11 21:33:39,005 INFO [139745807927168] (rs.cpp:2697) Scale Factor is now 32
 20/11 21:33:39,273 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
 20/11 21:33:39,273 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video1
 20/11 21:33:39,273 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video2
 20/11 21:33:39,273 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video3
 20/11 21:33:39,273 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video4
 20/11 21:33:39,273 INFO [139745807927168] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/video4linux/video5
 20/11 21:33:39,273 INFO [139745807927168] (context.cpp:336) Found 1 RealSense devices (mask 0xfe)
```

MartyG-RealSense, 2023-11-20:
Hi @Yinshideguanghui  Does it make a difference if you press the **F8** key on the top of the PC keyboard to toggle **Fullscreen** display mode on and off, please?

Yinshideguanghui, 2023-11-20:
> Hi @Yinshideguanghui Does it make a difference if you press the **F8** key on the top of the PC keyboard to toggle **Fullscreen** display mode on and off, please?

F8 changes the window to fullscreen mode, but there is no difference. Just made a fullscreen "Add" button
[image]
logs when I pressed F8 twice with a two-second interval:
```
20/11 22:42:48,538 INFO [140437744093056] (synthetic-stream-gl.cpp:97) Shutting down rendering
 20/11 22:42:48,538 INFO [140437744093056] (synthetic-stream-gl.cpp:103)  2 GPU objects cleaned-up
 20/11 22:42:48,538 INFO [140437744093056] (synthetic-stream-gl.cpp:153) Shutting down processing
 20/11 22:42:48,539 INFO [140437744093056] (synthetic-stream-gl.cpp:164)  386 GPU objects cleaned-up
 20/11 22:42:48,763 INFO [140437744093056] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=1
 20/11 22:42:48,764 INFO [140437744093056] (synthetic-stream-gl.cpp:89)  2 GPU objects initialized
 20/11 22:42:48,764 INFO [140437744093056] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
 20/11 22:42:48,783 INFO [140437744093056] (synthetic-stream-gl.cpp:146)  386 GPU objects initialized
 20/11 22:42:48,789 INFO [140437744093056] (rs.cpp:2697) Framebuffer size changed to 2488 x 1536
 20/11 22:42:48,789 INFO [140437744093056] (rs.cpp:2697) Window size changed to 77 x 48
 20/11 22:42:48,868 INFO [140437744093056] (rs.cpp:2697) Framebuffer size changed to 2560 x 1600
 20/11 22:42:48,868 INFO [140437744093056] (rs.cpp:2697) Window size changed to 80 x 50
 20/11 22:42:50,233 INFO [140437744093056] (synthetic-stream-gl.cpp:97) Shutting down rendering
 20/11 22:42:50,233 INFO [140437744093056] (synthetic-stream-gl.cpp:103)  2 GPU objects cleaned-up
 20/11 22:42:50,233 INFO [140437744093056] (synthetic-stream-gl.cpp:153) Shutting down processing
 20/11 22:42:50,234 INFO [140437744093056] (synthetic-stream-gl.cpp:164)  386 GPU objects cleaned-up
 20/11 22:42:50,435 INFO [140437744093056] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=1
 20/11 22:42:50,436 INFO [140437744093056] (synthetic-stream-gl.cpp:89)  2 GPU objects initialized
 20/11 22:42:50,436 INFO [140437744093056] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
 20/11 22:42:50,501 INFO [140437744093056] (synthetic-stream-gl.cpp:146)  386 GPU objects initialized
 20/11 22:42:50,528 INFO [140437744093056] (rs.cpp:2697) Framebuffer size changed to 2488 x 1536
 20/11 22:42:50,528 INFO [140437744093056] (rs.cpp:2697) Window size changed to 77 x 48
```

MartyG-RealSense, 2023-11-20:
Thanks very much for trying the fullscreen toggle option.  

A few other RealSense users at https://github.com/IntelRealSense/librealsense/issues/10879 and https://github.com/IntelRealSense/librealsense/issues/5511 experienced this scale problem with the Add Source button.  

If you are using Linux then the solutions suggested by an Intel RealSense team member at https://github.com/IntelRealSense/librealsense/issues/5511 are likely to be the most appropriate for you.

Yinshideguanghui, 2023-11-20:
> Thanks very much for trying the fullscreen toggle option.
>
> A few other RealSense users at #10879 and #5511 experienced this scale problem with the Add Source button.
>
> If you are using Linux then the [...]

Thanks for the suggestions, but I'm afraid this caused another problem. I tried the solution at [#5511](https://github.com/IntelRealSense/librealsense/issues/5511), built from source, and now it shows all components nested at the top-left corner (still, F8 makes no difference, sad).
[image]

MartyG-RealSense, 2023-11-21:
Would it be possible to update the graphics drivers on your computer please in order to eliminate the possibility that this display issue is caused by the computer's OpenGL graphics support being out of date (as realsense-viewer relies on this support).

Yinshideguanghui, 2023-11-21:
> Would it be possible to update the graphics drivers on your computer please in order to eliminate the possibility that this display issue is caused by the computer's OpenGL graphics support being out of date (as realsense-viewer relies on this support).

I've been working with an NVIDIA 3080 Ti Laptop GPU. I updated the driver to the newest (nvidia-driver-535 metapackage), but the problem still exists. I also tried sudo apt install --reinstall the opengl, glfw3, e.t.c., but that made no difference.

I am now trying to develop without using realsense-viewer to test my camera.  Hope it works so that we don't have to deal with this tough imgui problem anymore . T_T

MartyG-RealSense, 2023-11-21:
Thanks very much for the update.  Please let me know how your work progresses.  Good luck!

Yinshideguanghui, 2023-11-22:
> Thanks very much for the update. Please let me know how your work progresses. Good luck!

I used OpenCV to show the RGB images and depths. It works fine. Thanks again for your kind and patient help! Hope this realsense-viewer bug could be figured out and fixed in the future.

closed 2023-11-29 by MartyG-RealSense