realsenseai/librealsense #12435: No MATLAB Wrapper in Windows installer.

saroshpatel, 2023-11-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |       D405                                                 | 
| Firmware Version                   |       5.5                                                    | 
| Operating System & Version |    Win 10                                                | 
| Kernel Version (Linux Only)    |  --------------                                        | 
| Platform                                 |   PC                                                       |
| SDK Version                            |  [SDK 2.0 (v2.54.2)                                | 
| Language                               |                         MATLAB                         | 
| Segment		             	|  Robot                                                    | 

### Issue Description

Hello,

I am using a D405 cam, I installed the [SDK 2.0 (v2.54.2)]  (https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2) using the windows installer and also upgraded the firmware to the latest. 

I could not find any MATLAB folder for the wrapper in the installed folder. Am I doing something  wrong ? These are the only folders I see ..
[image]

Please help ...

Thank you

MartyG-RealSense, 2023-11-22:
Hi @saroshpatel  I replied to your question at https://github.com/IntelRealSense/librealsense/issues/6014#issuecomment-1820445396

saroshpatel, 2023-11-25:
Thank you. I saw that. I was solved.

MartyG-RealSense, 2023-11-25:
You are very welcome.  As you have achieved a solution, I will close this case.  Thanks very much for the update!

closed 2023-11-25 by MartyG-RealSense