realsenseai/librealsense #12442: using D435i with VOXL2

ta-jetson, 2023-11-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|--------------,
-------------------|------------------------------------------- |
| Camera Model                       | D435i |
| Operating System                  |Debian Buster -Linux Kernel v4.19      |
|CPU                                       | QRB5165
                                                  8 cores up to
                                                  3.091GHz
                                                  8GB LPDDR5
                                                  128GB Flash                                         |
|GPU                                      |  Adreno 650 GPU –1024 ALU |
| ROS compatibility                 | Yes |
| Platform                                 | VOXL2 |

Hi @MartyG-RealSense ,
I plan on using Realsense D435i with the VOXL2 board. This is because I have had excellent results with the pointcloud genererated from d435i. I am planning on connecting the D435i to VOXL2 using USB expansion port provided by VOXL, they have a 10 pin connecter to USB 3.0 . i read your comments on "compatibility with VOXL autonomy board #8595", and I am completely fine with using the Realsense SDK and RSUSB method. I plan on using the realsense-ros wrapper to access information for pointcloud data.

How do I proceed and can I proceed? 

Should I follow this ?
"In order to build the SDK using the RSUSB method and avoid the kernel patching procedure, please refer to [libuvc_installation.sh](https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh) script for details. If you have CUDA dev-kit installed, don't forget to add -DBUILD_WITH_CUDA=true for optimal performance."

And another doubt, since it is not a Nvidia GPU, there is no CUDA. I there anyway I can make the SDK use the current GPU?

Any advice would be appreciated

MartyG-RealSense, 2023-11-24:
Hi @ta-jetson  If the computer / computing device does not have an Nvidia GPU then you can use the RealSense SDK's **GLSL Processing Blocks** function if you are writing a C++ language application.  Like with CUDA, GLSL offloads processing from the CPU onto the graphics GPU.  It is 'vendor neutral', meaning that it should work with any GPU brand, though improvement may not be noticeable on devices with a low-end hardware specification.  https://github.com/IntelRealSense/librealsense/pull/3654 advises on the pros and cons of GLSL, and there is also an **rs-gl** C++ example program at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

GLSL support in a program is enabled by editing **RS2::** instructions in a script - for example, changing RS2::POINTCLOUD to RS2::GL::POINTCLOUD - so it does not require the use of a CMake build flag like the CUDA support does when building librealsense from source code.

******

In regard to how to proceed with installing the SDK and the ROS wrapper: if you are using ROS1 (Kinetic, Melodic or Noetic) then you can install librealsense and the wrapper together at the same time with the command below.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

This easy and convenient installation method is based on RSUSB = true, though it does not install the RealSense Viewer tool.  If you require the Viewer then you will need to build librealsense and the ROS wrapper separately - librealsense first and the wrapper secondly afterwards.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#method-1-the-ros-distribution

If you are going to use ROS2 (Humble or Iron) then librealsense and the ROS2 wrapper must be installed separately.  

https://github.com/IntelRealSense/realsense-ros#installation

ta-jetson, 2023-12-11:
Thanks for your assistance @MartyG-RealSense ,
I have a couple of additional questions
1. Do I need a docker to proceed or building on the existing environment will be good?
2. VOXL is a headless system, so I can't view the stream in realsense viewer. However they have a web-portal. Can you provide a suggestion on how to get a realsense-viewer "like" interface through a web-portal?

ta-jetson, 2023-12-11:
Understood
So,
I'll carry forward with the command (apt install) and view it in an HTML5 browser window.
I'm currently doing the project, so i'll update you with my progress and close this issue. 
Thanks for all of your advice.

MartyG-RealSense, 2023-12-11:
You are very welcome.  Good luck!

ta-jetson, 2023-12-11:
Hey @MartyG-RealSense, I ran into some problems with the build.
So VOXL has there own fine tuned opencv (4.5.5) and ros melodic. This version of ros-melodic doesn't have cv-bridge. 

So the command you suggested gives the following error:
||
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-melodic-realsense2-camera : Depends: ros-melodic-cv-bridge but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
||

I couldn't apt-install cv-bridge. So i built it from source for the opencv in VOXL2.

I tried to install librealsense from source. Using
||
./scripts/patch-realsense-ubuntu-lts.sh
./scripts/patch-realsense-ubuntu-lts-hwe.sh
||

 But I ran into this error:
Unsupported kernel version 4.19.125 . The Bionic patches are maintained for Bionic Beaver LTS kernels 4.1[5/8], 5.[0/3/4]

My kernel version is 4.19.125

Unable to install this as well:
||
sudo apt-get install librealsense2-utils
sudo apt-get install librealsense2-dev
||

Could you please provide any suggestions on how to proceed?

ta-jetson, 2023-12-11:
Hey @MartyG-RealSense ,
So
i tried building it using 
$cmake .. -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true
output:
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.54.2
-- Setting Unix configurations
-- Building libcurl enabled
-- using RS2_USE_LIBUVC_BACKEND
-- GLFW 3.3 not found; using internal version
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
-- Using X11 for window creation
-- Check for updates capability added to realsense-viewer
-- Check for updates capability added to realsense-depth-quality
-- Fetching recommended firmwares:
-- disable link time optimization for fw project
-- ... https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-5.15.1.0.bin
-- ... https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- ... https://librealsense.intel.com/Releases/L5xx/FW/L51X_FW_Image-1.5.8.1.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/root/librealsense/build

$sudo make uninstall and make clean
output:
CMake Warning at cmake_uninstall.cmake:2 (MESSAGE):
  Cannot find install manifest:
  "/home/root/librealsense/build/install_manifest.txt"

-- Uninstall targets will be skipped
Built target uninstall

# Got an error here
$time make -j8 && sudo make install
output (ERROR) :
[image]
[image]

Could you please provide any suggestions on how to proceed?

MartyG-RealSense, 2023-12-12:
On some computers, such as ones with a low-end configuration, using a 'j' number such as -j8 can cause a random crash during building.  Please try using a lower j-number such as -j4, or not using a j-number at all.

**make && sudo make install**

MartyG-RealSense, 2023-12-22:
Hi @ta-jetson  Do you require further assistance with this case, please?  Thanks!

ta-jetson, 2023-12-22:
Hi @MartyG-RealSense , I have not tested what you told earlier, I faced some other issues with VOXL hardware itself, I'll test it out as soon as possible and get back to you.

MartyG-RealSense, 2023-12-22:
That's no problem at all - thanks very much for the update.  Good luck!

MartyG-RealSense, 2023-12-30:
Hi @ta-jetson  Do you have an update about this case that you can provide, please?  Thanks!

ta-jetson, 2024-01-11:
Hey @MartyG-RealSense ,
I was facing some unrelated hardware issues. The VOXL is under service and hence the project is on hold. I will update you as soon as I get it back.
Until then, I'll close this issue
Thank you for all of the help @MartyG-RealSense

closed 2024-01-11 by ta-jetson

MartyG-RealSense, 2024-01-11:
You are very welcome.  Please do feel free to re-open the issue or create a new one when you are ready to resume.  Good luck!

Jm7752, 2024-06-24:
Completely different person, but I have succesfully built the realsense SDK and realsense_camera in ROS2. However the realsense_camera in ROS2 says no intel reasnse device found 

and running realsense-viewer I get the followign output

With D435i plugged in 
voxl2:/$ realsense-viewer
 24/06 19:52:47,385 ERROR [548009005072] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation already in progress
 24/06 19:52:47,385 INFO [548009005072] (backend-v4l2.cpp:905) Not a USB video device: Mipi device capability could not be grabbed Last Error: Operation already in progress
 24/06 19:52:47,386 ERROR [548009005072] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation already in progress
 24/06 19:52:47,386 INFO [548009005072] (backend-v4l2.cpp:905) Not a USB video device: Mipi device capability could not be grabbed Last Error: Operation already in progress
 24/06 19:52:47,386 INFO [548009005072] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/platform/soc/a800000.ssusb/a800000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.2/1-1.2:1.0/video4linux/video2
 24/06 19:52:47,386 INFO [548009005072] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/platform/soc/a800000.ssusb/a800000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.2/1-1.2:1.0/video4linux/video3
 24/06 19:52:47,386 INFO [548009005072] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/platform/soc/a800000.ssusb/a800000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.2/1-1.2:1.0/video4linux/video4
 24/06 19:52:47,387 INFO [548009005072] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/platform/soc/a800000.ssusb/a800000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.2/1-1.2:1.0/video4linux/video5
 24/06 19:52:47,387 INFO [548009005072] (backend-v4l2.cpp:657) Enumerating UVC video6 realpath=/sys/devices/platform/soc/a800000.ssusb/a800000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.2/1-1.2:1.3/video4linux/video6
 24/06 19:52:47,387 INFO [548009005072] (backend-v4l2.cpp:657) Enumerating UVC video7 realpath=/sys/devices/platform/soc/a800000.ssusb/a800000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.2/1-1.2:1.3/video4linux/video7
 24/06 19:52:47,390 INFO [548009005072] (backend-hid.cpp:1175) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device1
 24/06 19:52:47,390 INFO [548009005072] (backend-hid.cpp:1175) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device2
 24/06 19:52:47,390 INFO [548009005072] (backend-hid.cpp:1175) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0

Do you have any suggestions for testing. The Voxl2 itself has several MIPI cameras also attached so it doesn't suprise me it ignores those, but it seems there exists some UVC stream while the d435i is plugged in. 

Without D435i plugged in 
 24/06 20:01:56,968 ERROR [547684057104] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation already in progress
 24/06 20:01:56,968 INFO [547684057104] (backend-v4l2.cpp:905) Not a USB video device: Mipi device capability could not be grabbed Last Error: Operation already in progress
 24/06 20:01:56,968 ERROR [547684057104] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation already in progress
 24/06 20:01:56,968 INFO [547684057104] (backend-v4l2.cpp:905) Not a USB video device: Mipi device capability could not be grabbed Last Error: Operation already in progress
 24/06 20:01:56,970 INFO [547684057104] (backend-hid.cpp:1175) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device1
 24/06 20:01:56,970 INFO [547684057104] (backend-hid.cpp:1175) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device2
 24/06 20:01:56,971 INFO [547684057104] (backend-hid.cpp:1175) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0

Jm7752, 2024-06-25:
I solved the issue......... I feel like a list of build options in cmake would be better apprecaited somwehere. Anyway it works with Voxl2 completely. I can give detailed instructions on how to do so on request next week if someone pings me, but it mostly inovles changes to the libraries to hook to Voxl2 openCV which is user dependent.

I.e do they value Voxl2's openCV library which allows the QVIO aglrotihm to run versus building with their own openCV.

MartyG-RealSense, 2024-06-25:
Thanks so much @Jm7752 for sharing your experience!

Jm7752, 2024-06-26:
Ok I will describe the full procedure to install D435i with VOXL2 below. It is very similar to following these steps. By the end of this you should have ROS2 working with streaming video images. We are using a custom humble version for our ROS2, but I think ti should be similar for Foxy.

Note this procedure is ONLY valid for Voxl2 which has Ubuntu support. Also it is highly recommended you upgrade the cmake on it to 3.20 just in case. 
Steps to update ROS2
https://gist.github.com/bmegli/4049b7394f9cfa016c24ed67e5041930

Build realsense driver from source
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk
(We use branch v2.54.1)

1. First follow the steps and install dependencies in the steps here if some don't install don't worry about it for a bit. 
2. Do the following commands
`mkdir build && cd build`
3. Select the following options for CMKAE IT IS IMPERATIVE DFORCE RSUSB BACKEND IS ENABLED
`cmake ../ -DBUILD_GRAPHICAL_EXAMPLES=false -DFORCE_RSUSB_BACKEND=true`
4.Install the driver
`sudo make uninstall && make clean && make && sudo make install`

ROS2 Package. So Voxl2 as mentioned above has their own version of custom openCV. This enables their QVIO or onboard VIO server. You have 2 options with this. Either remove their openCV and install your own fairly trivial just right sudo apt get install some specific openCV. In which case the realsense ROS2 should work as is

https://github.com/IntelRealSense/realsense-ros

IF YOU WANT TO USE MODALAI OpenCV you have to link the pacakges. We have our version of vision_opencv liked to modalai openCV
[git](https://github.com/arplaboratory/vision_opencv/tree/voxl2)

Next replace the realsense_camera CMakelist to link to the VOxl2 openCV libraries 

`file(GLOB OpenCV_LIBS "/usr/lib64/libopencv*.so")
include_directories(include /usr/include/opencv4/)`

MartyG-RealSense, 2024-06-27:
Thanks again, @Jm7752   :)