realsenseai/librealsense #12443: Device busy when start flask app

niknadzirie, 2023-11-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      | 
| Firmware Version                   |          5.15.1                                                       | 
| Operating System & Version |   Ubuntu 22.04.3 LTS  | 
| Kernel Version (Linux Only)    |  5.10.110-rockchip-rk3588                                         | 
| Platform                                 | Orange Pi 5B  |
| SDK Version                            |  2.54.2                         | 
| Language                            |  Python                       | 
| Segment			|  IoT                      | 

### Issue Description
The issue is every time i start flask app, it will state that device is busy, then i need to unplug then plug the camera back, then restart the flask unless it still state device is busy.

MartyG-RealSense, 2023-11-24:
Hi @niknadzirie  Is your flask application a pyrealsense2 script?  If it is, does the problem still occur if you add a **hardware_reset()** instruction before the pipe start line to reset the camera automatically when the script is launched?

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

niknadzirie, 2023-11-24:
> Hi @niknadzirie Is your flask application a pyrealsense2 script? If it is, does the problem still occur if you add a **hardware_reset()** instruction before the pipe start line to reset the camera [...]

So basically i've got seperate python file, one is app.py and the other one is camera.py. I've tried put the code you given in camera.py file to reset the camera but still not working. Still same i need to unplug the camera and plug it back.

MartyG-RealSense, 2023-11-24:
If a program script starts the pipeline then it becomes 'busy' until the pipeline is stopped or the camera is disconnected by some means (such as unplugging the camera).  Whilst the pipeline is enabled, issuing another start instruction will result in a busy error until the pipeline has been stopped, after which a start instruction will work again. 

Do you have a pipeline start instruction in both scripts, please?

For example, if app.py contains an instruction to start the pipeline then camera.py will get a busy error if it afterwards tries to start the pipeline.  The reverse is also true.  If camera.py is first to start the pipeline then app.py will get a busy error if it tries to do so.

niknadzirie, 2023-11-24:
> If a program script starts the pipeline then it becomes 'busy' until the pipeline is stopped or the camera is disconnected by some means (such as unplugging the camera). Whilst the pipeline is enabled, issuing another start instruction will result in a busy error until the pipeline has been stopped, after which a start instruction will work again.
> 
> Do you have a pipeline start instruction in both scripts, please? If app.py contains an instruction to start the pipeline then camera.py will get a busy error if it afterwards tries to start the pipeline. The reverse is also true. If camera.py is first to start the pipeline then app.py will get a busy error if it tries to do so.

I start pipeline in camera.py, in app.py i called the function in camera.py such generate_frame() to stream on web.

alex-carneiro, 2023-11-24:
Hi @niknadzirie, I faced the same problem than you described.

In my case, it was solved by turning debug off in Flask.
app.run(debug=False, host='0.0.0.0', port=5000)

I hope this might be useful.

MartyG-RealSense, 2023-11-25:
Hi @alex-carneiro  Thanks so much for your advice to @niknadzirie  :)

@niknadzirie  Does turning off debug resolve your issue, please?

niknadzirie, 2023-11-27:
@MartyG-RealSense Yes the advice given by @alex-carneiro is worked for me, thank you so much!!!

closed 2023-11-27 by niknadzirie

MartyG-RealSense, 2023-11-27:
That's great to hear that the solution provided by @alex-carneiro worked for you.  Thanks very much for the update!

niknadzirie, 2023-11-27:
@alex-carneiro do you know how to run it using gunicorn?