realsenseai/librealsense #12444: Intel D4 Board V3 doesn´t work and no valid SKU

albertoo22, 2023-11-24:
Hello, I am using the Intel D450 camera with the D4 V3 module. 

When I connect both via USB, it says "Intel Corp. Intel(R) RealSense(TM) Invalid SKU."
When I connect only the D4 V3 module windows says "it does not recognize the USB device, it is not working correctly"

I have tried updating the firmware in two ways as I have seen in other issues:

Way 1: I have installed RealSense Viewer SDK on WIndows 10, but when I connect the camera it does not detect it, so I cannot update the firmware

Way 2: I have tried using the command rs-fw-update -l on Ubuntu 20.04.06 LTS but this returns:
"rs-fw-update: command not found" (I have tried other tools, but they all return the same).

¿What can I do? I think the problem is the firmware in D4 V3 but I don´t know how I can update it.

KERNEL
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D450 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (5.15.0-89-generic)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2023-11-24:
Hi @albertoo22  When combining a D450 depth module with the Vision Processor D4 V3, the **Invalid SKU** error can occur if a newer version of the D450 module is being used with an older version of the D4 V3, as described at https://github.com/IntelRealSense/librealsense/issues/11621#issuecomment-1536539701

There are two approaches to solving this.

1.  Connect the older **K83121-100** version of the D450 depth module to the D4 board, update the firmware and then re-connect the depth module that you were using before.  

This option would not work though unless you already owned an older K83121-100 version of the D450, as there is no way to know which version you are getting when ordering from a retailer.

Or the other approach is:

2.  Disconnect the D450 depth module and try to perform a firmware update (the firmware is stored on the D4 V3, so it can update firmware without a depth module attached).  The D4 on its own should be recognized as **D405i Tester** instead of D450.  

A problem with this method though is that 'D405i Tester' is not currently recognized as a RealSense camera model ID when attempting firmware update and so it does not work.  Intel are planning to add support for 'D405i Tester' to allow firmware update in this mode, so this invalid SKU issue should hopefully be fixable in future without having to obtain an older board.  

The other approach of changing the D450 board for an older version did work for a RealSense user who had such a board and tried it, as described at https://github.com/IntelRealSense/librealsense/issues/12205

albertoo22, 2023-11-24:
When I connect D4 V3 only, Windows doesn't detect the USB, so I can not make the second way that you say :/.
Are there other option to update the firmware?

Also, if i conect the camera and write this command: 'roslaunch realsense2_camera rs_camera.launch' . I get this error:

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [3908]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 18beac90-8ab2-11ee-be1d-99d3306807f9
process[rosout-1]: started with pid [3918]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [3925]
process[camera/realsense2_camera-3]: started with pid [3926]
[ INFO] [1700820803.170722498]: Initializing nodelet with 3 worker threads.
[ INFO] [1700820803.379232005]: RealSense ROS v2.3.2
[ INFO] [1700820803.379377640]: Built with LibRealSense v2.50.0
[ INFO] [1700820803.379408279]: Running with LibRealSense v2.50.0
[ INFO] [1700820803.440641909]:  
[ WARN] [1700820803.773818755]: Device 1/1 failed with exception: hwmon command 0x10( 0 0 0 0 ) failed (response -844431340= unknown)
[ERROR] [1700820803.774141915]: The requested device with  is NOT found. Will Try again.

Do you know if this error is related to the problem of 'Invalid SKU'?

MartyG-RealSense, 2023-11-24:
The ROS wrapper likely cannot detect the camera because the D450 module is not being identified.

There is nothing that you can do yourself to correct this issue with firmware updating, unfortunately.  I will discuss your situation with my Intel RealSense colleagues in the coming week.  Thanks very much for your patience until then.

albertoo22, 2023-11-27:
Good morning!! Is there any news on the issue?

MartyG-RealSense, 2023-11-27:
I am awaiting a response.   Thanks very much for your patience.

albertoo22, 2023-12-04:
Hello MartyG-RealSense, any news? Thank you

MartyG-RealSense, 2023-12-05:
Hi @albertoo22  My colleagues have responded to me.  Can you send me your postal address and the number of D450 depth modules that you have to my email address below please so that I can continue the support discussion with you privately.

martyx.grover@intel.com

developajarit, 2023-12-20:
Now, in order to update the firmware and be able to use the D450 module, how can I do it?

I connect the module D430 with the Visor Processor D4v3 and I receive this notification (photo attached)
[image]

MartyG-RealSense, 2023-12-20:
Hi @developajarit  The message box indicates that you have version **2.48.0** of the RealSense SDK and firmware driver version **5.12.7.150**.  

You should be using firmware version **5.12.14.50** with SDK 2.48.0.

The appropriate firmware update procedure would be:

1.  Download the **bin** file for firmware 5.12.14.50 from the firmware archive page at the link below.

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

2.  In the RealSense Viewer tool, go to the **More** option near the top of the Viewer's options side-panel and select the **Update Firmware** option from its drop-down menu.  
. 
3. A file selector window will pop up.  Navigate to the location on your computer that you downloaded the bin file to and select it.  The firmware update should then automatically begin.

developajarit, 2023-12-20:
[image]

MartyG-RealSense, 2023-12-20:
Your Viewer images shows that you are using the D430 module.  The D4 V3 board will only work with the **D450** module.  If you have a D430 board then you need to use the V1 board if you don't need an IMU (Motion Module), or the V2 board if you need an IMU.

developajarit, 2023-12-20:
I am currently using the D450 K83121-101 module with the D4V3 K94249-110 vision processor.
Connected to the computer, it shows what I mentioned above.

[image]

I repeat the steps I have followed:

1. I have connected the vision processor to D4V3 with the D430 module to be able to update the firmware in order to later use the D450 module with the D4V3 vision processor, since it had 'invalid SKU'.

2. I have updated the firmware to 5.12.14.50.

3. Once the firmware was updated, I connected the D4V3 vision processor to the D450 module again. The 'Invalid SKU' message no longer appears, but the module is detected as: Intel RealSense USB2 D4xx Recovered

developajarit, 2023-12-20:
I forgot to say that I am the same user as albertoo22, just with a different account!!

MartyG-RealSense, 2023-12-20:
Thank you for the confirmation of the module that you are using.  Your Viewer was identifying the camera as a D430i.

[image]

My understanding from previous Invalid SKU cases is that the K83121-101 version of the D450 module prefers the latest firmware to be used with it, as described at https://github.com/IntelRealSense/librealsense/issues/11621#issuecomment-1536539701

Could you try updating the firmware to **5.15.1.0** please.

developajarit, 2023-12-20:
Now it is recognized correctly, thank you very much!!

Now there is a problem receveing frames form the cameras, I was already seeing something related to this and I read that it probably because it is being detected as USB 2.0 instead of USB 3.0, right?

When I active the Stereo Module I recived this error:

[image]

MartyG-RealSense, 2023-12-20:
This error is likely because you have SDK 2.48.0 installed.  Using the latest firmware with old SDK versions results in this Auto Exposure Limit error message.

Please visit the link below and download and run the file **Intel.RealSense.Viewer.exe** in the 'Assets' file list at the bottom of the page.  This will launch the 2.54.2 version of the Viewer that is designed for use with firmware 5.15.1.0.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

Installing this Viewer will not overwrite the 2.48.0 SDK installation already on your computer.

developajarit, 2023-12-20:
Perfect, it seems to be working correctly now.

[image]

One last question, can the fact that the connection is being detected as USB 2.0 have any influence on the use of this camera?

I also have the T265 tracking module and it is also detected as USB 2.0 and in this case the fisheye cameras do not start when I try to start it through ROS and RVIZ

MartyG-RealSense, 2023-12-20:
If you are using a depth module then you would not have had an official RealSense USB cable supplied with it.  If you are using a USB cable that you purchased yourself then there is a chance that it could be a USB2 cable rather than a USB3 cable.  USB2 cables are missing extra wires that USB3 cables have, and that prevents USB3 devices from operating at USB3 speed.

If it is a USB3 cable, the camera would also be detected as USB2 if the camera was plugged into a USB2 hub or in a USB2 port on the computer.

developajarit, 2023-12-20:
Okay, thank you very much for the clarification. What is the specific USB 3.0 cable for RealSense cameras?

MartyG-RealSense, 2023-12-20:
The official cable is not available for purchase separately but a company called **Newnex** supplies RealSense-compatible USB3 cables of equal quality to the official one.

https://www.newnex.com/realsense-3d-camera-connectivity.php

However, other USB3 cables should work so long as they are described in their store listing as **data transfer** cables and are not charging cables.

closed 2024-01-05 by MartyG-RealSense