realsenseai/librealsense #12445: memory leak when we allocate realsense class

seodongmin-kor, 2023-11-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |  Win10 |                              | 
| Platform                                 | PC |
| SDK Version                            |  2.54.1                        | 
| Language                            |  C/C++                          | 
| Segment			|  Robot                          | 

### Issue Description
<issue / question>

As shown in the code below, there is a REALSENSE class, and when repeatedly assigned and released from the main function, memory leakage is found. I think it's a simple member variable allocation, so which part is wrong?

----------------------------------------------------------------------------------------------------
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>   // Include OpenCV API
#include <../samples/cv-helpers.hpp>
#include "open3d/Open3D.h"

class realsense {
public:
	rs2::pipeline pipe;
	rs2::config cfg;
	rs2::pipeline_profile profile;
	rs2::sensor sensor;
	rs2::threshold_filter filter;
	rs2::pointcloud pc;
	rs2::points p;
	const rs2::vertex* vp;
	rs2_intrinsics depth_intrin;
	rs2_intrinsics color_intrin;
	rs2_extrinsics depth_to_color_extrin;
	rs2_extrinsics color_to_depth_extrin;
	float depth_scale;
	cv::Mat image_d;
	cv::Mat image_c;
	open3d::geometry::PointCloud o3dp;

	realsense(float threshold, float exposure) {};
	~realsense() {};
	void capture() {};};
----------------------------------------------------------------------------------------------------

#include "rs_realsense.h"

int main() {
	while (1) {
		realsense* rs_ptr = new realsense(0.8, 150);
		delete rs_ptr;
	}
	return 0;
}

----------------------------------------------------------------------------------------------------

MartyG-RealSense, 2023-11-26:
Hi @seodongmin-kor  Please first check whether the pointcloud is being created by a loop because this can cause a memory leak as advised by a Realense team member at https://github.com/IntelRealSense/librealsense/issues/2860

seodongmin-kor, 2023-11-26:
Hi @MartyG-RealSense. I've already made the code as below and tested it repeatedly, but there was no memory leak. It is expected to leak within the realsense2.dll code.

```
#include "rs_realsense.h"

class test {
public:
	cv::Mat image_d;
	cv::Mat image_c;
	open3d::geometry::PointCloud o3dp;
};

int main() {

	while (1) {
		test* test_ptr = new test;
		delete test_ptr;
	}

	return 0;
}
```

MartyG-RealSense, 2023-11-26:
Wht happens if you comment out `delete rs_ptr;`

seodongmin-kor, 2023-11-26:
I'm debugging with the heap profiling tool of the visual studio. 
Running line 96 allocates 1618.99 kb and executing line 97 deallocates 1527.45 kb.

Repeated execution of the code as a while statement will cause memory to leak continuously.

[image]

seodongmin-kor, 2023-11-26:
If i comment out `delete rs_ptr`, The memory continues to leak as shown in the picture below.

[image]

And then, The test* ptr test result with the while loop removed is as follows. Allocation and deallocation are done properly. There is no memory leak.

[image]

MartyG-RealSense, 2023-11-26:
Is the loop necessary to your program, please?  Using looping in a RealSense pointcloud can result in a memory leak, as mentioned earlier.

seodongmin-kor, 2023-11-26:
I will change the code structure to allocate only the first time. However, i have to periodically capture the screen, but it leaks as shown in the picture below. Can you check if there is anything wrong with my code?

#ifndef __RS_REALSENSE_H__
#define __RS_REALSENSE_H__

```
//#include "MyHeader.h"
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>   // Include OpenCV API
#include <../samples/cv-helpers.hpp>
#include "open3d/Open3D.h"

class realsense {
public:
	rs2::pipeline pipe;
	rs2::config cfg;
	rs2::pipeline_profile profile;
	rs2::sensor sensor;
	rs2::threshold_filter filter;
	rs2::pointcloud pc;
	rs2::points p;
	rs2::device dv;
	const rs2::vertex* vp;
	rs2_intrinsics depth_intrin;
	rs2_intrinsics color_intrin;
	rs2_extrinsics depth_to_color_extrin;
	rs2_extrinsics color_to_depth_extrin;
	float depth_scale;
	cv::Mat image_d;
	cv::Mat image_c;
	open3d::geometry::PointCloud o3dp;

	realsense(float threshold, float exposure);
	~realsense();
	void capture();
};

#endif
```

```
#include "rs_realsense.h"

realsense::realsense(float threshold, float exposure) {
	std::cout << "execute realsense" << std::endl;
	this->cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
	this->cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);

	this->profile = this->pipe.start(this->cfg);

	this->sensor = this->pipe.get_active_profile().get_device().query_sensors()[1];
	this->sensor.set_option(RS2_OPTION_EXPOSURE, exposure);
	this->filter.set_option(RS2_OPTION_MAX_DISTANCE, threshold);

	this->depth_scale = this->profile.get_device().first<rs2::depth_sensor>().get_depth_scale();
}

void realsense::capture() {
	for (int i = 0; i < 10; i++) this->pipe.wait_for_frames(); // Skips some frames to allow for auto-exposure stabilization

	rs2::frameset fr = this->pipe.wait_for_frames();
	rs2::align align_to_color(RS2_STREAM_COLOR);
	fr = align_to_color.process(fr);

	rs2::depth_frame frame_d(fr.get_depth_frame());
	frame_d = this->filter.process(frame_d);
	this->image_d = depth_frame_to_meters(frame_d);

	rs2::video_frame frame_c(fr.get_color_frame());
	this->image_c = frame_to_mat(fr.get_color_frame());

	this->p = this->pc.calculate(frame_d);
	this->pc.map_to(frame_c);

	// Intrinsics & Extrinsics
	this->depth_intrin = frame_d.get_profile().as<rs2::video_stream_profile>().get_intrinsics();
	this->color_intrin = frame_c.get_profile().as<rs2::video_stream_profile>().get_intrinsics();
	this->depth_to_color_extrin = frame_d.get_profile().get_extrinsics_to(frame_c.get_profile());
	this->color_to_depth_extrin = frame_c.get_profile().get_extrinsics_to(frame_d.get_profile());

	this->vp = this->p.get_vertices();
	
	// 획득된 pointcloud 중 0 인 값 제외
	for (int i = 0; i < this->p.size(); i++) {
		if ((this->vp + i)->x != 0 && (this->vp + i)->y != 0 && (this->vp + i)->z != 0) {
			this->o3dp.points_.push_back(Eigen::Vector3d((this->vp + i)->x, (this->vp + i)->y, (this->vp + i)->z));
		}
	}
	this->o3dp.PaintUniformColor(Eigen::Vector3d(0.4, 0.4, 0.4));
}

realsense::~realsense() {
	std::cout << "terminate realsense" << std::endl;
	this->pipe.stop();
}
```

```
#include "rs_realsense.h"

int main() {
	realsense* rs_ptr = new realsense(0.8, 150);
	
	rs_ptr->capture();
	rs_ptr->capture();
	rs_ptr->capture();
	rs_ptr->capture();
	rs_ptr->capture();
	rs_ptr->capture();
	rs_ptr->capture();

	delete rs_ptr;

	return 0;
}
```

[image]

MartyG-RealSense, 2023-11-26:
I checked through the program code that you provided and it mostly looks fine.  The only recommendation that I would make is to not have the alignment-applying **fr = align_to_color.process(fr);** instruction within a loop, so place the line above wait_for_frames() instead of below it.

seodongmin-kor, 2023-11-26:
@MartyG-RealSense Thanks! In my realsense::capture() code, i added `"this->o3dp.Clear()" `
**this->o3dp** is a class member variable so that it can not be deleted automatically untill deallocate class object.

MartyG-RealSense, 2023-11-26:
Thanks very much for the update.  Does the addition of `"this->o3dp.Clear()"` solve your problem, please?

seodongmin-kor, 2023-11-26:
It seems to have been solved, but when I checked the memory leak for about 8 hours with the Windows performance monitor, it was gradually increasing as follows. Can you double check if there's anything wrong with my code?

[image]

```
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>   // Include OpenCV API
#include <../samples/cv-helpers.hpp>
#include "open3d/Open3D.h"

class realsense {
public:
	rs2::pipeline pipe;
	rs2::config cfg;
	rs2::pipeline_profile profile;
	rs2::sensor sensor;
	rs2::threshold_filter filter;
	rs2::pointcloud pc;
	rs2::points p;
	rs2::device dv;
	const rs2::vertex* vp;
	rs2_intrinsics depth_intrin;
	rs2_intrinsics color_intrin;
	rs2_extrinsics depth_to_color_extrin;
	rs2_extrinsics color_to_depth_extrin;
	float depth_scale;
	cv::Mat image_d;
	cv::Mat image_c;
	open3d::geometry::PointCloud o3dp;

	realsense(float threshold, float exposure);
	~realsense();
	void capture();
};
```

```
realsense::realsense(float threshold, float exposure) {
	std::cout << "execute realsense" << std::endl;
	this->cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
	this->cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);

	this->profile = this->pipe.start(this->cfg);

	this->sensor = this->pipe.get_active_profile().get_device().query_sensors()[1];
	this->sensor.set_option(RS2_OPTION_EXPOSURE, exposure);
	this->filter.set_option(RS2_OPTION_MAX_DISTANCE, threshold);

	this->depth_scale = this->profile.get_device().first<rs2::depth_sensor>().get_depth_scale();
}

void realsense::capture() {
	for (int i = 0; i < 10; i++) this->pipe.wait_for_frames(); // Skips some frames to allow for auto-exposure stabilization

	rs2::frameset fr = this->pipe.wait_for_frames();
	rs2::align align_to_color(RS2_STREAM_COLOR);
	fr = align_to_color.process(fr);

	rs2::depth_frame frame_d(fr.get_depth_frame());
	frame_d = this->filter.process(frame_d);
	this->image_d = depth_frame_to_meters(frame_d);

	rs2::video_frame frame_c(fr.get_color_frame());
	this->image_c = frame_to_mat(fr.get_color_frame());

	this->p = this->pc.calculate(frame_d);
	this->pc.map_to(frame_c);

	// Intrinsics & Extrinsics
	this->depth_intrin = frame_d.get_profile().as<rs2::video_stream_profile>().get_intrinsics();
	this->color_intrin = frame_c.get_profile().as<rs2::video_stream_profile>().get_intrinsics();
	this->depth_to_color_extrin = frame_d.get_profile().get_extrinsics_to(frame_c.get_profile());
	this->color_to_depth_extrin = frame_c.get_profile().get_extrinsics_to(frame_d.get_profile());

	this->vp = this->p.get_vertices();

	// 초기화(클래스 멤버변수 이므로 인스턴스를 삭제하기 전까지 소멸되지 않으므로 명시적으로 삭제 시킴)
	this->o3dp.Clear(); 
	this->o3dp.points_.clear();
	this->o3dp.points_.shrink_to_fit();
	// 획득된 pointcloud 중 0 인 값 제외
	for (int i = 0; i < this->p.size(); i++) {
		if ((this->vp + i)->x != 0 && (this->vp + i)->y != 0 && (this->vp + i)->z != 0) {
			this->o3dp.points_.push_back(Eigen::Vector3d((this->vp + i)->x, (this->vp + i)->y, (this->vp + i)->z));
		}
	}
	this->o3dp.PaintUniformColor(Eigen::Vector3d(0.4, 0.4, 0.4));
}

realsense::~realsense() {
	std::cout << "terminate realsense" << std::endl;
	this->pipe.stop();
}
```

MartyG-RealSense, 2023-11-27:
The code looks fine.  You could try dealing with the leak by adding the **rs2_release_frame** instruction to release frames and free up memory.  A RealSense team member provides C++ code for doing so at https://github.com/IntelRealSense/librealsense/issues/4006#issuecomment-494033057

seodongmin-kor, 2023-11-27:
Which frame do i have to release? Should I release frame_d and frame_c? Or should I release fr? If i release the three together, an error will occur. I think both of them will be the same because fr_d and fr_c are included in fr, is that correct?

1.
```
rs2_release_frame(fr.get()); 
//rs2_release_frame(frame_d.get());
//rs2_release_frame(frame_c.get());
```

2.
```
//rs2_release_frame(fr.get()); 
rs2_release_frame(frame_d.get());
rs2_release_frame(frame_c.get());
```

MartyG-RealSense, 2023-11-27:
I would usually say that it would be frame **fr** (the one used in the wait_for_frames() instruction) that should be released, but in this particular case with multiple frame references, the choice that provides the best, most stable results (frame_c and frame_d) is likely to be the best one.

seodongmin-kor, 2023-11-27:
After testing the memory leak for about an hour with case 2 (frame_d and frame_c → release), it was founded that the memory leak occurred as follows. I want to apply case number 1 (fr → release), but can I know the cause of the error like the image attached above?

[image]

seodongmin-kor, 2023-11-28:
I have one more question. 

If i move **this->profile = this->pipe.start(this->cfg);** code in the capture() function and add **this->pipe.stop()** at the end like below, will the memory be released without a frame release instruction?(**rs2_release_frame()**)

```
void realsense::capture() {

	this->profile = this->pipe.start(this->cfg);

        .....

        this->pipe.stop();
}
```

MartyG-RealSense, 2023-11-28:
Some RealSense users have used a free tool called **Valgrind** to analyze memory leaks.  https://valgrind.org/

Does the program function correctly if you tick the **Except when thrown from realsense2.dll** box to allow the program to continue if the exception is thrown?

The documentation link below indicates that stopping the pipeline releases resources used by the camera (such as disabling active streams), but the application is responsible for releasing frames.  Closing down an application would be an example of releasing memory that the application has consumed.

You could try placing the release instruction after the pipe stop line if you only want to release once.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html#a142ee1adb798f9a03f86eb90895dd8a5

MartyG-RealSense, 2023-12-02:
1.  I am not able to test code or use Valgrind on my computer, unfortunately.  I do apologize.

2.  Repeated start and stop of the pipeline can be a cause of memory leak.  Examples of this are https://github.com/IntelRealSense/librealsense/issues/8091 and https://github.com/IntelRealSense/librealsense/issues/10757

seodongmin-kor, 2023-12-04:
Is it right that Repeated start and stop of the pipeline can be a cause of memory leak? In #8091, @VadimAspirin says "Starting and stopping pipeline with and without catching frames seems to not leak any memory"

[image]

closed 2023-12-22 by MartyG-RealSense