realsenseai/librealsense #12450: Intel RealSense D415 - Automatically switch to night vision

rajrajeswaran, 2023-11-27:
Camera Model: Intel D415
Firmware: 05.12.07.100
Operating System: Windows 10
Platform: PC
SDK version: C# RealSense SDK 2.38
Language: C#
Segment: Windows Forms application.

We have an application that consumes to RGB, IR, and 3D data from a Intel RealSense D415.  We are currently using C# SDK 2.38 version.  Currently, we are having to manually switch to night vision when lighting condition is low in the room.  Essentially, our application broadcast either RGB or IR based on user input.  We are looking to automate this process.  Is there any event handler in  C# SDK that can notify the application when the lighting condition is low (lights OFF) from RGB sensor?  In the same way, notify lighting conditions are high when in IR mode (lights ON)?

MartyG-RealSense, 2023-11-28:
Hi @rajrajeswaran  The IR projector component's laser controls have a setting that is usually not used called **Laser Auto**.  It was deprecated (not recommended for use) as far back as 2018 but remains selectable in the RealSense SDK.  My understanding of this setting is that it was intended to toggle the laser on or off depending on the current lighting level.

[image]

When controlling the status of the projector, the 'Laser Auto' setting has a value of '2'.  So you could try the following C# code.

```
var pipe = new Pipeline();
PipelineProfile selection = pipe.Start();
var selected_device = selection.Device;
var depth_sensor = selected_device.Sensors[0];

if (depth_sensor.Options.Supports(Option.EmitterEnabled))
{
depth_sensor.Options[Option.EmitterEnabled].Value = 2f; // Enable emitter Auto mode
}
```

It may be easier though to simply have the emitter permanently enabled.  Because its light is IR, it will not be visible to the human eye or on the RGB image but will illuminate the IR image and enhance the depth image quality.

rajrajeswaran, 2023-11-28:
@MartyG-RealSense Thank you for the quick response.  I did not state my question clearly.  In our code, we consume both IR and RGB data but only broadcast RGB or IR based on user selection.  When the lighting condition is low, the user will select to receive IR (night vision).  When appropriate lighting condition exists, user will select to receive RGB frames.  I would like to automate this transition from IR to RGB and vice versa.  In order to achieve that I need a notification from SDK that tells me the lighting condition have a reached a low threshold or high thresh hold in the room.  Is that possible?  If a low threshold is received, I will automatically send the IR frames to user.  When high threshold is reached, I will send RGB frames to end user.  and our Option.EmitterEnabled is always set to 1.

MartyG-RealSense, 2023-11-29:
The only way I can think of that you may be able to implement such a mechanism in the RealSense SDK if you have depth auto-exposure enabled is to monitor the value of the camera metadata parameter **RS2_FRAME_METADATA_ACTUAL_EXPOSURE**.  The auto-exposure should adapt depending on the current lighting level.  You could set up a check to send a notification when the exposure falls below or rises above certain values.

Camera metadata is accessible in the C# wrapper - as confirmed at https://github.com/IntelRealSense/librealsense/issues/3349 - but there is unfortunately not an existing example of a C# script for doing so.

rajrajeswaran, 2023-11-29:
Thank you @MartyG-RealSense .  I will review the API and open a new ticket if further information is needed.

closed 2023-11-29 by rajrajeswaran