realsenseai/librealsense #12452: RealSense IMU/Motion Module: Failed to open device sensor

milan-r-shah, 2023-11-28:
| Required Info                         |                                                     |
|----------------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.12.14.50 | 
| Operating System & Version  | Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)   |  5.4.0 | 
| Platform                                 | Computer  |
| SDK Version                          | 2.44.0 | 
| Language                              | C++ | 
| Segment			        | {Robot/Smartphone/VR/AR/others } | 

### Issue Description
On my machine, the `realsense-viewer` works without any issues. However, when I try to access the IMU, depth camera, and RGB camera data, using my own code, I get the following error:
```
WARNING: RealSense IMU/Motion Module could not be opened: 'Failed to open device sensor. /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0034/HID-SENSOR-200076.3.auto/iio:device1 Last Error: No such file or directory'
```
If I don't enable the IMU/Motion Module, I don't get the aforementioned error!

MartyG-RealSense, 2023-11-28:
Hi @milan-r-shah  Which version of the Vision Processor D4 Board are you using with your D430 depth module, please?  The IMU is located on the D4 board.  So if it is the **V1** version of the board then it will not have an IMU.  If it is the **V2** version of the D4 board then it will have an IMU.

A D430 board also does not have an RGB sensor.  So an RGB stream would have to come from somewhere else, such as a USB webcam or a camera built in to a laptop screen.  That RGB stream would also only be accessible using the **Add Source** button at the top of the realsense-viewer options side-panel.

Would it be possible to share your C++ script in a comment?

MartyG-RealSense, 2023-12-05:
Hi @milan-r-shah  Do you require further assistance with this case, please?  Thanks!

milan-r-shah, 2023-12-06:
Sorry for not getting back to you sooner, @MartyG-RealSense. I'm not super sure which version of the Vision Processor I'm using. But, I'm pretty sure it has IMU because otherwise, I wouldn't have been able to see the motion module data in the `realsense-viewer` correct?

Yup, the RGB stream is coming from a separate sensor. I don't have any issues accessing it either via the viewer or by my code.

I can't share my C++ script directly. Let me see what I can do. I'll try creating a sample code that reproduces the error.

MartyG-RealSense, 2023-12-06:
Yes, your D4 board would have to be a version that has an IMU component on it in order for it to be detected in the Viewer as the Motion Module.  For the D430 module it should be the V2 version of D4, as the V3 version that also has an IMU is for use with the D450 depth module only.

closed 2023-12-28 by MartyG-RealSense