realsenseai/librealsense #12453: RuntimeError: No device connected

sesamestengg, 2023-11-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | F455
| Firmware Version                   | Open RealSense Viewer v2.54.2
| Operating System & Version | Win 10
| Platform                                 | PC
| SDK Version                           | SDK 2.0 version 2.54.2 |
| Language                               | Python
| Segment		       	         | others

### Issue Description
opencv-python 4.8.1.78
pip 23.3.1
pylint 3.0.2
pyrealsense 2.54.2.5684
setuptools 56.0.0

Hi, i am continuously running into "RuntimeError: No device connected" when i run several scripts including https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py. The camera i used is labelled F455 but is recognize on Device and Manager as F450. Its attached by USB3.0 and i have tried scouring solutions in the forum but to no avail. 

I am a student and have been programming for fun on the sideline for 2 years so i might not be that advanced in terms of software engineering. 

Any chance there might be a forum/solution i might have missed?

RealSenseCustomerSupport, 2024-06-24:
----------------------------------------------

@sesamestengg If you're using F455 please refer to github link https://github.com/IntelRealSense/RealSenseID  Thanks!

RealSenseCustomerSupport, 2024-07-02:
----------------------------------------------

@sesamestengg Did you get chance to try F455 github link? Any more questions about this ticket? Looking forward to your reply. Thanks!

closed 2026-10-23 by MartyG-RealSense