realsenseai/librealsense #12457: Depth error has increased without changing the environment

avadosf, 2023-11-30:
Hi, 
Previously the camera D455, attached perpendicular to the floor at a height of 3m gave a depth of 3m, now without changing any algorithms, the camera at the same place gives a depth of 3.3m. The camera has not been calibrated, the calibration is factory default, the laser power has not been changed, the value is 150, the firmware version has not been changed and the client for obtaining the depth has not been changed either. Could you please tell me what could be the reason?

avadosf, 2023-12-01:
Yeah, I tried that. The calibration values didn't change.

MartyG-RealSense, 2023-12-02:
Would it be possible to take a depth reading at different times of day to confirm whether environmental temperature is a factor in the different measurement?

avadosf, 2023-12-04:
The room is kept at the same temperature

avadosf, 2023-12-05:
[image]
[image]

avadosf, 2023-12-05:
There are no shadows on the floor

MartyG-RealSense, 2023-12-05:
What happens to the depth value if you maximize the Laser Power value to '360'.  If Laser Power is increased then the infrared dot pattern projected by the camera onto the surface of objects becomes more visible to the camera, improving depth analysis.

avadosf, 2023-12-06:
The matter may be in the calibration of the camera, when checked on another camera the error was less. I will let you know when I recheck everything.

MartyG-RealSense, 2023-12-06:
Thanks very much.  I look forward to your next report after testing.  Good luck!

avadosf, 2023-12-07:
Tare calibration solved my problem, the results are more accurate. But I encountered another problem, because the camera is without housing, it overheats and its accuracy decreases. 
Now I am trying to calibrate another camera, the fully aligned camera in Depth Quality Tool shows an angle of 2-3 degrees, also the depth is slightly tilted.
[image]
I have attached an example image, the studies were conducted under different conditions, but the results were similar.

MartyG-RealSense, 2023-12-07:
A method of reducing heat on a caseless depth module is to attach a **heat sink** to the back of the module.  A heat sink is a block of metal that draws heat away from electronics.  The cased version of RealSense cameras contain one, like the image of a D435's internal components below.  

[image]

Page 119 of the current edition of the data sheet document for the 400 Series cameras contains information about heat sinks.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

The incorrect area of depth on one side seems to correspond to the presence of a stronger light source being cast onto the wall on that side of the scene, as shown by the areas of white compared to the darker area on the other (correct) side of the wall.

It could also be that the light is making the wall's pattern of a stack of boards more visible to the camera.  Regular horizontal or vertical patterns can cause a **repetitive pattern** to form that confuses the camera's depth sensing.  Intel have a guide at the link below for negating repetitive patterns.

https://dev.intelrealsense.com/docs/mitigate-repetitive-pattern-effect-stereo-depth-cameras

In regard to the tilt, you could check whether the camera is fully level, especially if it is mounted on a tripod.

closed 2023-12-12 by avadosf