realsenseai/librealsense #12460: D430 RuntimeError: Frame didn't arrive within 5000

MACTEPBEKA, 2023-12-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430 | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |  WIN10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.2.5684 }                          | 
| Language                            |  Python 3.11 (also tested at 3.10 3.7)                        | 
| Segment			|  Machine Vision                    | 

After launching [script](https://pastebin.com/1ib651YH) for getting frames from camera I get error 

frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000

In RealSense Viewer v2.54.2 i can connect to camera and watch depth image, but can not connect at python script.

I tryed also launch .pyd lib in to the "site-packages", tryed to launch lib from file in current directory, reinstall sdk, reconnect usb(3.2 version) but problem always active

MartyG-RealSense, 2023-12-01:
Hi @MACTEPBEKA  When the **pc.calculate**  point cloud instruction is used, the line directly before it is usually a **map_to** instruction to map depth and color together.  Your script does not have a map_to instruction.

```
pc.map_to(color_frame)
points = pc.calculate(depth_frame)
```

MACTEPBEKA, 2023-12-01:
> Hi @MACTEPBEKA When the **pc.calculate** point cloud instruction is used, the line directly before it is usually a **map_to** instruction to map depth and color together. Your script does not have a map_to instruction. Does adding one make a difference to your RuntimeError, please?
> 
> ```
> pc.map_to(color_frame)
> points = pc.calculate(depth_frame)
> ```

RuntimeError before this part of code. But I will use your advice at future

MartyG-RealSense, 2023-12-02:
Do you get a RunTime error if you run the RealSense SDK's pointcloud example script that also uses pc.calculate, please?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py

MartyG-RealSense, 2023-12-04:
Is the RealSense Viewer closed when you run your Python script?  If the Viewer is streaming depth then another program launched afterwards will not be able to use the depth stream until the Viewer's depth stream is stopped.

If you do not have the Viewer active when launching your Python program, does the RunTimeError still occur if you insert a **hardware_reset()** instruction before the pipe start line in order to reset the camera when the script runs?

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

MACTEPBEKA, 2023-12-05:
After deep search in issues I found a solution of my problem. 

https://github.com/IntelRealSense/librealsense/issues/7048

closed 2023-12-05 by MACTEPBEKA

MartyG-RealSense, 2023-12-05:
I'm pleased to hear that you found a solution.  Thanks very much for the update!