realsenseai/librealsense #12462: Pyrealsense2-D435i：frames = pipeline.wait_for_frames() RuntimeError: Frame didn't arrive within 5000

Wnm0706, 2023-12-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |5.15.1 | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    | 5.4.0-1098-raspi  | 
| Platform                                 | Raspberry Pi 4B |
| SDK Version                            | 2.54.2                      | 
| Language                            | Python     | 
| Segment			|  Others                       | 

### Issue Description
The following is the relevant code named "image-collect.py" for capturing and saving images using Intel Realsense D435i in Python:

```
import pyrealsense2 as rs
import numpy as np
import cv2
import time
import os

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipeline.start(config)

depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: ", depth_scale)

align_to = rs.stream.color
align = rs.align(align_to)

save_path = os.path.join(os.getcwd(), "out", time.strftime("%Y_%m_%d_%H_%M_%S", time.localtime()))
os.mkdir(save_path)
os.mkdir(os.path.join(save_path, "color"))
os.mkdir(os.path.join(save_path, "depth"))
os.mkdir(os.path.join(save_path, "depth_map"))
os.mkdir(os.path.join(save_path, "data"))

cv2.namedWindow("live", cv2.WINDOW_AUTOSIZE)
cv2.namedWindow("save", cv2.WINDOW_AUTOSIZE)
saved_color_image = None
saved_depth_mapped_image = None
saved_count = 0

try:
    while True:
        frames = pipeline.wait_for_frames()
        aligned_frames = align.process(frames)

        aligned_depth_frame = aligned_frames.get_depth_frame()
        color_frame = aligned_frames.get_color_frame()

        if not aligned_depth_frame or not color_frame:
            continue

        depth_data = np.asanyarray(aligned_depth_frame.get_data(), dtype="float16")
        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        depth_mapped_image = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        cv2.imshow("live", np.hstack((color_image, depth_mapped_image)))
        key = cv2.waitKey(30)

        if key & 0xFF == ord('s'):
            saved_color_image = color_image
            saved_depth_mapped_image = depth_mapped_image
            saved_depth_image = depth_image

            cv2.imwrite(os.path.join((save_path), "color", "{}.png".format(saved_count)), saved_color_image)
            cv2.imwrite(os.path.join((save_path), "depth", "{}.png".format(saved_count)), saved_depth_image)
            cv2.imwrite(os.path.join((save_path), "depth_map", "{}.png".format(saved_count)), saved_depth_mapped_image)
            np.save(os.path.join((save_path), "data", "{}".format(saved_count)), depth_data)
            saved_count += 1
            cv2.imshow("save", np.hstack((saved_color_image, saved_depth_mapped_image)))

        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()
```
    
    
When running this "image_collect.py" program on a Raspberry Pi 4B with Ubuntu 20.04 installed, you encounter an error:

```
frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000

```

Additionally, when attempting to resolve the issue by upgrading the firmware of D435i using "rs_fw_update," you encounter another error:
```
Connected devices:
1) Name: Intel RealSense D435I, serial number: 135122077394, update serial number: 134523065209, firmware version: 5.15.1, USB type: 3.2
 01/12 16:06:21,733 ERROR [281473802507328] (librealsense-exception.h:52) /dev/video10 is no video capture device Last Error: No such file or directory
 01/12 16:06:21,733 ERROR [281473802507328] (sensor.cpp:661) acquire_power failed: /dev/video10 is no video capture device Last Error: No such file or directory
 01/12 16:06:21,825 WARNING [281473802507328] (rs.cpp:312) null pointer passed for argument "device"
RealSense error calling rs2_create_device(info_list:0xaaaaef4a3510, index:1):
    /dev/video10 is no video capture device Last Error: No such file or directory
```
It's worth mentioning that during the installation of librealsense, the package "librealsense2-dpkm" couldn't be located. You installed the librealsense-related packages using cmake, and after installation, "realsense-viewer" and "rs-capture" work fine. The D435i is connected to the USB 3.0 port on the Raspberry Pi, and the same Python file runs successfully on a PC with Windows installed. However, you are unsure how to resolve the mentioned issues and seek assistance.
[image]

Wnm0706, 2023-12-02:
Thank you for your reply! I understand what you mean. Currently, I am using Python 3.8 in the Ubuntu 20.04 system in Raspberry Pi. I can install realsense related software packages in Python through the pip install pyrealsense 2 you mentioned. However, the problem I am currently facing is when I run image_ When collecting. py, the following error occurs: ```
```
frame=pipeline.wait_ For_ Frames()
RuntimeError: Frame didn't drive within 5000
```

And when I run rs ww update, the following content appears:
```
Connected devices:
1) Name: Intel RealSense D435I, serial number: 135122077394, update serial number: 134523065209, firmware version: 5.15.1, USB type: 3.2
 01/12 16:06:21,733 ERROR [281473802507328] (librealsense-exception.h:52) /dev/video10 is no video capture device Last Error: No such file or directory
 01/12 16:06:21,733 ERROR [281473802507328] (sensor.cpp:661) acquire_power failed: /dev/video10 is no video capture device Last Error: No such file or directory
 01/12 16:06:21,825 WARNING [281473802507328] (rs.cpp:312) null pointer passed for argument "device"
RealSense error calling rs2_create_device(info_list:0xaaaaef4a3510, index:1):
    /dev/video10 is no video capture device Last Error: No such file or directory
```

I don't know how to solve my current problem, as you can see, I am able to run realsense viewer normally and turn on the camera, and I am able to obtain realsense videos in ROS. I'm not sure if reinstalling the SDK can solve the issue of RuntimeError: Frame didn't drive within 5000 that I encountered. I would like to try using the
```
sudo apt-get install librelsense2-dkms
sudo apt-get install librelsense2 utils
```
I came to install the SDK, but during the installation process, I was unable to locate the package librelsense2-dkms

MartyG-RealSense, 2023-12-02:
The librealsense2-dkms and librelsense2 utils packages are not suitable for use with Arm devices like Pi as they are designed for use with x86 / x64 computers such as laptop and desktop PCs.  The installation method for using Ubuntu with Pi that is most successful is to build from source code with CMake and include the flag **-DFORCE_RSUSB_BACKEND=TRUE** in the CMake build instruction.

An alternative, simpler approach for building from source code on Pi is to use an installation script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

MartyG-RealSense, 2023-12-03:
If the Python code has been confirmed to be correct then it is likely that the problem is with the Pi's installation of the RealSense SDK.  The SDK is difficult to install and get working on Python compared to other computers / computing boards and can experience errors with Python scripts even if the RealSense Viewer is working.  An example of this is that the IMU sometimes cannot be accessed on Pi, whilst depth and color work.

Instead of updating the firmware with rs-fw-update, you could try performing the update in the Viewer by going to the 'More' option near the top of the Viewer's options side-panel and selecting 'Update Firmware' from its drop-down menu, then navigating to the downloaded firmware bin file on your coputer to select it and initiate the firmware update.

Wnm0706, 2023-12-05:
Thank you for your reply, fortunately I found a solution to this problem, the reason is that the pyrealsense2 library is not available for the arm architecture raspberry pi, so it can't be installed by pip install pyrealsense2. The correct way is to import the pyrealsense related files from the compiled librealsense/build directory into ~/.bashrc and copy the pyrealsense2 related so, so.2.54, so.2.54.2 files into python3/site-packages. directory to call realsense in python.

MartyG-RealSense, 2023-12-05:
Thanks so much @Wnm0706 for sharing your solution for installing pyrealsense2 on Raspberry Pi  :)

MartyG-RealSense, 2023-12-11:
Hi @Wnm0706  After having found a solution, do you require further assistance with this case please?  Thanks!

Wnm0706, 2023-12-15:
No more for now, thanks for the reply! @MartyG-RealSense

MartyG-RealSense, 2023-12-15:
You are very welcome, @Wnm0706 - thanks very much for the update!

Wnm0706, 2024-03-05:
> > 谢谢你的回复，好在我找到了这个问题的解决方案，原因是pyrealsense2库不适用于arm架构树莓派，所以pip install pyrealsense2无法安装。正确的方法是将编译好的 librealsense/build 目录下的 pyrealsense 相关文件导入到 ~/.bashrc 中，并将 pyrealsense2 相关的 so、so.2.54、so.2.54.2 文件复制到 python3/site-packages 中。在 Python 中调用实感的目录。
> 
> 能写一下具体的流程吗？谢谢

Specifically, when you compile and install realsense related software packages in the Ubuntu system, the librealsense folder will exist in your Ubuntu home directory. Copy files with the so, so.2.54, and so.2.54.2 suffixes from **/home/librealsense/build** to the root directory of Python in your Ubuntu system, usually located in the **python3/site-packages** directory, to solve this problem.

If you encounter any problems during this process, please feel free to contact me.

ikyuocwc, 2024-03-06:
> Specifically, when you compile and install realsense related software packages in the Ubuntu system, the librealsense folder will exist in your Ubuntu home directory. Copy files with the so, [...]

感谢您的回答，我在我的树莓派4B+的ubuntu20.04LTS系统，可以通过relasense-viewer正常打开摄像头，但是我用的是的miniconda环境（python3.8.8），因为我想通过python代码实现yolo检测，返回三维坐标。但是我遇到了报错：frames = pipeline.wait_for_frames() RuntimeError: Frame didn't arrive within 5000

按照你的回答，将编译好的 librealsense/build 目录下的 pyrealsense 相关文件导入到 ~/.bashrc 中的命令是下面这个吗
nano ~/.bashrc
export PYTHONPATH=$PYTHONPATH:/home/librealsense/build/pyrealsense2

将 pyrealsense2 相关的 so、so.2.54、so.2.54.2 文件复制到 python3/site-packages 中，我并没有找到python3/site-packages目录，有的只是/usr/lib/python/dist-packages，在我复制之后，依旧没有解决。要是您有空的话，能不能写一篇详细的配置教程吗（带有图文），感谢

Wnm0706, 2024-03-07:
> 感谢您的回答，我在我的树莓派4B+的ubuntu20.04LTS系统，可以通过relasense-viewer正常打开摄像头，但是我用的是的miniconda环境（python3.8.8），因为我想通过python代码实现yolo检测，返回三维坐标。但是我遇到了报错：frames = pipeline.wait_for_frames() RuntimeError: Frame didn't [...]

First of all, I'm sorry that my raspberry pi is not with me right now, so I can't give you a more detailed configuration tutorial. Since I am using the pre installed Python 3 interpreter for Ubuntu 20.04, I have solved this problem using the above method. I noticed that you are using the Python 3.8 environment in Miniconda. You can try copying the files related to Pyrealsense 2 to the **site-package of the Python 3.8** installation in Miniconda, which should solve this problem.

ikyuocwc, 2024-03-08:
> First of all, I'm sorry that my raspberry pi is not with me right now, so I can't give you a more detailed configuration tutorial. Since I am using the pre installed Python 3 interpreter for Ubuntu [...]

感谢您的回复，我已经解决了，能够使用python打开摄像机。如下是我的详细过程（我的环境是ubuntu20.04LTS）：
sudo apt-get update 
sudo apt-get upgrade
sudo apt-get install nano
sudo apt-get install git
git clone https://github.com/IntelRealSense/librealsense
cd librealsense/
mkdir build && cd build
sudo apt-get install libxrandr-dev
sudo apt-get install libxinerama-dev
sudo apt-get install libsdl2-dev
cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
sudo apt-get install libusb-1.0-0-dev
sudo make uninstall && make clean && make && sudo make install
cd ..
sudo ./scripts/setup_udev_rules.sh
测试代码
realsense-viewer

#安装pyrealsense2
cd ~/librealsense/build
cmake .. -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python3)
make -j1
sudo make install
# add python path（添加python路径，该条不是指令）
nano ~/.zshrc # 复制下面语句到~/.zshrc中
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
source ~/.zshrc
到librealsense/build/wrappers/python目录下看发现有了一些so、so.2、so.2.50、so.2.50.0结尾的文件，我们将这些文件全部拷贝至 /usr/lib/python3/dist-packages/ 文件夹下

facerrr, 2024-03-09:
> 感谢您的回复，我已经解决了，能够使用python打开摄像机。如下是我的详细过程（我的环境是ubuntu20.04LTS）： sudo apt-get update sudo apt-get upgrade sudo apt-get install nano sudo apt-get install git git clone [...]

Hello, I followed your code and compiled it, but I didn't find any so file in the wrappers/python folder. Only CMakeFiles  cmake_install.cmake  Makefile  pyrealsense2Config.cmake  pyrealsense2ConfigVersion.cmake. Have you ever had this situation?

ikyuocwc, 2024-03-10:
> > > > > > > 谢谢你的回复，好在我找到了这个问题的解决方案，原因是pyrealsense2库不适用于arm架构树莓派，所以pip install pyrealsense2无法安装。正确的方法是将编译好的 librealsense/build 目录下的 pyrealsense 相关文件导入到 ~/.bashrc 中，并将 pyrealsense2 相关的 so、so.2.54、so.2.54.2 文件复制到 python3/site-packages 中。在 Python 中调用实感的目录。
> > > > > > 
> > > > > > 
> > > > > > 能写一下具体的流程吗？谢谢
> > > > > 
> > > > > 
> > > > > 具体来说，当你在Ubuntu系统中编译安装实感相关软件包时，librealsense文件夹会存在于你的Ubuntu主目录下。将带有 so、so.2.54 和 so.2.54.2 后缀的文件从 **/home/librealsense/build** 复制到 Ubuntu 系统中 Python 的根目录下，通常位于 **python3/site-packages** 目录下，以解决此问题。如果您在此过程中遇到任何问题，请随时与我联系。
> > > > 
> > > > 
> > > > 感谢您的回答，我在我的树莓派4B+的ubuntu20.04LTS系统，可以通过relasense-viewer正常打开摄像头，但是我用的是的miniconda环境（python3.8.8），因为我想通过python代码实现yolo检测，返回三维坐标。但是我遇到了报错：frames = pipeline.wait_for_frames（） RuntimeError： Frame didn't arrival within 5000 按照你的回答，将编译好的 librealsense/build 目录下的 pyrealsense 相关文件导入到 ~/.bashrc 中的命令是下面这个吗 nano ~/.bashrc export PYTHONPATH=$PYTHONPATH：/home/librealsense/build/pyrealsense2 将 pyrealsense2 相关的 so、so.2.54、so.2.54.2 文件复制到 python3/site-packages 中，我并没有找到python3/site-packages目录，有的只是/usr/lib/python/dist-packages，在我复制之后，依旧没有解决。要是您有空的话，能不能写一篇详细的配置教程吗（带有图文），感谢
> > > 
> > > 
> > > 首先，很抱歉我的树莓派现在不在我身边，所以我不能给你更详细的配置教程。由于我使用的是 Ubuntu 3 预装的 Python 20.04 解释器，因此我使用上述方法解决了这个问题。我注意到您在 Miniconda 中使用了 Python 3.8 环境。您可以尝试将与 Pyrealsense 2 相关的文件复制到 Miniconda 中 **Python 3.8 安装的站点包**中，这应该可以解决这个问题。
> > 
> > 
> > 感谢您的回复，我已经解决了，能够使用python打开摄像机。如下是我的详细过程（我的环境是ubuntu20.04LTS）： sudo apt-get update sudo apt-get upgrade sudo apt-get install nano sudo apt-get install git git clone https://github.com/IntelRealSense/librealsense cd librealsense/ mkdir build & cd build sudo apt-get install libxrandr-dev sudo apt-get install libxinerama-dev sudo apt-get install libsdl2-dev cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true sudo apt-get install libusb-1.0-0-dev sudo make uninstall && make clean && make && sudo make install cd ..sudo ./scripts/setup_udev_rules.sh 测试代码 realsense-viewer
> > #安装pyrealsense2 cd ~/librealsense/build cmake ..-DBUILD_PYTHON_BINDINGS=bool：true -DPYTHON_EXECUTABLE=$（which python3） make -j1 sudo make install
> > # add python path（添加python路径，该条不是指令）
> > nano ~/.zshrc # 复制下面语句到~/.zshrc中 export PYTHONPATH=$PYTHONPATH：/usr/local/lib source ~/.zshrc 到librealsense/build/wrappers/python目录下看发现有了一些so、so.2、so.2.50、so.2.50.0结尾的文件，我们将这些文件全部拷贝至 /usr/lib/python3/dist-packages/ 文件夹下
> 
> 您好，我按照您的代码进行了编译，但是在wrappers/python文件夹中没有找到任何so文件。只有 CMakeFiles cmake_install.cmake Makefile pyrealsense2Config.cmake pyrealsense2ConfigVersion.cmake。你有过这种情况吗？

我的配置教程如下：
https://blog.csdn.net/weixin_73703571/article/details/136575520?spm=1001.2014.3001.5502

MartyG-RealSense, 2024-03-16:
Hi @ikyuocwc Thanks very much for the advice you have been providing to @Wnm0706  :)

Hi @Wnm0706  Do you require further assistance with this case, please?  Thanks!

closed 2024-03-23 by MartyG-RealSense