realsenseai/librealsense #12474: Dynamically setting a ROI or AE set point

mjc6fe, 2023-12-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|----------------------------------------|------------------------------------------- |
| Camera Model                       | D435f | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version  | Linux Ubuntu 20  | 
| Kernel Version (Linux Only)   |  5.15.0-89-generic | 
| Platform                                 | PC |
| SDK Version                          |  2.50.0 | 
| Language                              |  C++ + OpenCV | 
| Segment			        |  Robot | 

### Issue Description
We are thinking of transitioning from D435i to D435f to improve the behaviour with repetitive structures. While doing a performance comparison between both cameras, we have noticed that while the D435f yields much better performance when bright spots (like lighting on the ceiling) exists in the scene, it performs quite poorly when relatively small objects like hands are close to the camera (less than 0.5 meters) in large scenes. In contrast, the D435i still performs pretty well on the same scene.

This of course happens because the D435f filters quite a bit of the visible light and then on short distances, pretty much only IR light from the projector is present, over-exposing the area, and consequently the depth information is lost. Our assessment is that the AE is actually working pretty well when the ROI is correct. The problem is that in such a large scene (e.g., a long room), objects that are within the detection range of the camera have the same consideration as objects that are outside the detection range of the camera. The former become overexposed and lost in the depth because the IR imagers are simply compensating for overall luminance.

### Approach
We have dived into the depths of the issues here and the white papers and have tried a few approaches based:
1. Set a ROI dynamically to some calculated tiles
2. Set the AE set point dynamically to prevent overexposing some of our areas of interest

From a qualitative perspective, both approaches seem to work -depth is not lost and the overall image quality is preserved, however, both are dramatically slow (approximately, a 66% of the frames are dropped if you set either values on every frame).

### Questions
1. Is there a way to achieve what we need and keep a decent frame rate?
2. Is there any other setting we need to adjust to prevent dropping as many frames? - Not counting tricks like setting it every other frame and whatnot
3. Are there other recommended settings to compensate for over-exposed areas?

Some of our constraints:
* Must use AE since lighting changes a lot throughout the year
* Setting gain or exposure limits are not an option because large lighting changes
* Cannot set a static ROI since objects can show up on any area of the image
* Scenes have a lot of motions from many agents happening at the same time
* Repetitive structures are present
* Hardware is fixed and the re-positioning of the cameras is not an option

MartyG-RealSense, 2023-12-05:
Hi @mjc6fe  I would recommend dropping the processing-heavy method of updating the ROI on every frame and instead explore the possibility of helping the camera to handle situations of under-exposure and over-exposure better by using the **High-Dynamic Range (HDR)**.feature, which allows two distinct exposure values to be used.

https://dev.intelrealsense.com/docs/high-dynamic-range-with-stereoscopic-depth-cameras

In HDR mode auto-exposure is disabled, though you may find that the ability of the HDR mode to alternate between two exposures can still enable the camera to cope with lighting changes throughout the year.

MartyG-RealSense, 2023-12-11:
Hi @mjc6fe  Do you require further assistance with this case, please?  Thanks!

mjc6fe, 2023-12-12:
Hi @MartyG-RealSense, we don't need any further assistance for the time being, but thanks for your input!

closed 2023-12-12 by mjc6fe