realsenseai/librealsense #12476: Error building and applying patched kernel modules for:  Ubuntu 22 (jammy) with LTS kernel 5.15

benpeart, 2023-12-04:
I'm following the instructions at (https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md) and have run into an error at the step:

```
Build and apply patched kernel modules for:

Ubuntu 20/22 (focal/jammy) with LTS kernel 5.13, 5.15
./scripts/patch-realsense-ubuntu-lts-hwe.sh
```

The error is:

```
Fetching Ubuntu LTS tag   to the local kernel sources folder
fatal: You need to specify a tag name.

```
Full output of the script below:

```
ben@raspberrypi:~/librealsense$ ./scripts/patch-realsense-ubuntu-lts-hwe.sh

Remove all RealSense cameras attached. Hit any key when ready

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
bc is already the newest version (1.07.1-3build1).
build-essential is already the newest version (12.9ubuntu3).
git is already the newest version (1:2.34.1-1ubuntu1.10).
linux-headers-generic is already the newest version (5.15.0.89.86).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-jammy folder

Package required libelf-dev:  - found
Package required elfutils:  - found
Package required bison:  - found
Package required flex:  - found
Package required dwarves:  - found
Fetching Ubuntu LTS tag   to the local kernel sources folder
fatal: You need to specify a tag name.
```

Note: I had to run the entire script as super user to get this far, I had other failures earlier in the script if I didn't but that is a separate issue.

Here is the output of uname on my system:

```
uname -a
Linux raspberrypi 5.15.0-1043-raspi #46-Ubuntu SMP PREEMPT Mon Nov 13 15:08:58 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

```

MartyG-RealSense, 2023-12-05:
Hi @benpeart  I note at the bottom of your opening post that you are using Raspberry Pi.  The kernel patch scripts listed on the installation.md instruction page are not designed for computing devices with Arm archiecture such as Pi, but instead for computers with x86 / x64 architecture such as laptop and desktop PCs.

The best method for installing the librealsense SDK for Ubuntu on Pi is to build from source code but include the build flag **-DFORCE_RSUSB_BACKEND=TRUE** in the CMake build instruction.  This builds the SDK in RSUSB mode, which bypasses the Linux kernel.  The SDK build is therefore not dependent on Linux versions or kernel versions and does not require kernel patching.

A simple method of building from source on Arm devices without needing to apply a kernel patch is to use a build script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The **libuvc_installation.sh** script builds the SDK with the flag **-DFORCE_LIBUVC=true**, which was the predecessor of the -DFORCE_RSUSB_BACKEND flag.

benpeart, 2023-12-05:
Yea, I started with Raspbian as the OS but when it didn't work (probably the Bullseye issue you mentioned) I switched to Ubuntu MATE per the [Raspberry Pi 3 ](https://dev.intelrealsense.com/docs/using-depth-camera-with-raspberry-pi-3) instructions. 

Sounds like Intel no longer has an official working/supported version for Raspberry Pi. When looking for ROS2 support, I noticed the Intel Robotics SDK was depreciated about 4 years ago. Looks like bought I an Intel RealSense camera for my robotic project a few years too late. :)

I'll try the steps in #11506 and see if I have success. If not, I guess I'll have to shop around for a different pointcloud/LIDAR/skeletal tracking solution for my project. Thanks for your help.

MartyG-RealSense, 2023-12-11:
Hi @benpeart  Do you require further assistance with this case, please?  Thanks!

benpeart, 2023-12-11:
No, I've given up on trying to get it to work on Raspberry Pi. Raspbian, Ubuntu, Ubuntu MATE. Don't work. I even tried going back to the last Intel supported version of Raspbian but it is so old, I can't download packages for the OS (they return 404 errors indicating they no longer exist).

closed 2023-12-11 by benpeart

MartyG-RealSense, 2023-12-12:
Okay, thanks very much for the update.  I'm sincerely sorry that you did not get the outcome that you were seeking.

flabrosse, 2024-03-07:
I'm taking over from a system that was left unfinished a while back.  The system uses a D455 connected to a Raspberry Pi4B running ubuntu 22.04 (and ROS humble).

Is the conclusion of this thread that the D455 (or any realsense device) will indeed not work on the RPI?

If this is the case, this is hugely disappointing and (1) will likely stop many small robotics projects and (2) needs to be made very clear to people as there is nothing in any documentation that states that, apart from places like this thread, that people will not read until too late.  We are now left with a bunch of small robots that have no way forward (no pun intended) because of that.

MartyG-RealSense, 2024-03-07:
Hi @flabrosse  RealSense camera hardware, including D455, can work with Raspberry Pi.  When problems occur, it is usually with the **software** side (the installation of the RealSense SDK on Pi boards).  Installation is certainly possible but may be challenging.

It is worth mentioning that whilst depth and color streams can work correctly on Raspberry Pi, the IMU streams (accel and gyro) of IMU-equipped RealSense camera models such as D435i and D455 may not be accessible on Pi.

If SDK installation problems are experienced, I recommend trying the **libuvc backend** build script at the link below as libuvc has a high past success rate on Pi boards.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

flabrosse, 2024-03-07:
Thanks @MartyG-RealSense.  Your comment from 05/12/2023 seemed to indicate that the script libuvc_installation was the old way of doing this: 

> The libuvc_installation.sh script builds the SDK with the flag -DFORCE_LIBUVC=true, which was the predecessor of the -DFORCE_RSUSB_BACKEND flag.

So I instead installed from sources using `-DFORCE_RSUSB_BACKEND=true` as a cmake argument.  This worked (compiled and installed) but no camera is being detected.  I'll try the script mentioned above now.