realsenseai/librealsense #12477: Failed to extract target information from the captured frames

MarkSmith-Bodd, 2023-12-04:
I have a lot of the D435 and D415 lens, and about 25% of them fail to perform the focal length calibration using the Relasense Viewer app.
The configuration is exactly the same each time, as I have built a test jig to maintain consistency where I can. The camera firmware is identical in all situations, and the lighting is the same.

However, I receive the dreaded "Failed to extract target information from the captured frames" on about 25% of my cameras.

[image]
)

MartyG-RealSense, 2023-12-05:
Hi @MarkSmith-Bodd  Does this error still occur if you first reset the calibration of affected cameras to their factory-new default settings using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 please?

MightyMaestro, 2023-12-05:
Hi @MartyG-RealSense .
After factory reset, the calibrations worked. Thank you for your help.

MartyG-RealSense, 2023-12-06:
You are very welcome.  It's excellent to hear that you were successful in calibrating the affected cameras.  Thanks very much for the update!

MartyG-RealSense, 2023-12-11:
Hi @MightyMaestro  After achieving a solution, do you require further assistance with this case please?  Thanks!

MightyMaestro, 2023-12-11:
Hi. @MartyG-RealSense. I am happy with the result. I do not require any further assistance on this matter. Cheers.

MartyG-RealSense, 2023-12-12:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks very much for the update!

closed 2023-12-12 by MartyG-RealSense