realsenseai/librealsense #12478: The problem of error reporting when two programs call d435i at the same time

RTrain111, 2023-12-05:
I encountered some problems when using d435i. I used ros to call d435i
"cd ~/catkin_ws
source devel/setup.bash
roslaunch aruco_ros single_realsense.launch"
At the same time, using the program Python to control the d435i shooting, such an error occurred.
"RuntimeError: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy"
I would like to know if this problem can be solved?

RTrain111, 2023-12-05:
In ubuntu18.04 system

MartyG-RealSense, 2023-12-05:
Hi @RTrain111  The librealsense SDK operates under rules called the **Multi-Streaming Model** at the link below that dictate that once a particular stream on a particular camera has been 'claimed' by a program, a second RealSense application launched subsequently cannot access that stream until the first program releases its claim on it (for example, by disabling the stream or closing the application).

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

A workaround may be to write a Python **node script** that can subscribe to topics generated by a roslaunch and use that topic's data.  The RealSense ROS wrapper has an example Python node script called **show_center_depth.py** that subscribes to the topic `/camera/depth/image_rect_raw` and `/camera/depth/camera_info` in order to calculate the distance to the coordinate at the center of the camera's view.

**ROS1**
https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/scripts/show_center_depth.py

**ROS2**
https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/scripts/show_center_depth.py

closed 2023-12-22 by MartyG-RealSense