realsenseai/librealsense #12486: Depth Quality Tool has crashed

peacerior, 2023-12-05:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2023-12-05:
Hi @peacerior  Please first check whether permission to access the camera hardware and permission for applications to access the camera are both set to 'On' in the Windows camera privacy settings.  This can be done using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

If the crash occurred wth the Depth Quality Tool, does the **RealSense Viewer** tool also crash with the same error?

MartyG-RealSense, 2023-12-15:
Hi @peacerior  Do you require further assistance with this case, please?  Thanks!

peacerior, 2023-12-16:
Hi! it helped a bitt to find the cause = Enabling the depth module in
camera settings solved the issue. When using the RealSense cam for the
webcam, it disables the depth module. I just enabled the module.

MartyG-RealSense, 2023-12-17:
Hi @peacerior  I'm pleased to hear that you achieved a solution.  Thanks very much for the update!    As your problem is solved I will close this case.

closed 2023-12-17 by MartyG-RealSense

peacerior, 2023-12-17:
Thanks for your support.