realsenseai/librealsense #12487: color and depth image quality (d435)

jooseuk, 2023-12-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> ) | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC|
| SDK Version                            |                    | 
| Language                            |  python }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
In the example code below, after a 5-second wait, images (both color and depth) are captured every 3 seconds. This code is intended to create a color point cloud from the color and depth images. Are there any ways to improve the quality and resolution of the depth and color images for better formation of the color point cloud? Please let me know if there are any parts in my current code that need modification or if there are any issues.
#timer
import pyrealsense2 as rs
import numpy as np
import cv2
import time

# RealSense pipeline
pipeline = rs.pipeline()
config = rs.config()

# RGB depth stream
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30) 

# start pipeline
pipeline.start(config)

# camera intrinsic
profile = pipeline.get_active_profile()
color_profile = profile.get_stream(rs.stream.color)
color_intrinsics = color_profile.as_video_stream_profile().get_intrinsics()
depth_profile = profile.get_stream(rs.stream.depth)
depth_intrinsics = depth_profile.as_video_stream_profile().get_intrinsics()

print("Color Intrinsics: ", color_intrinsics)
print("Depth Intrinsics: ", depth_intrinsics)

# align rgb depth
align_to = rs.stream.color
align = rs.align(align_to)

# wait 5sec
time.sleep(5)

# time setting
save_interval = 3
last_saved_time = time.time()
num_saved_images = 0
max_images_to_save = 5  # max imgs

try:
    while True:
        frames = pipeline.wait_for_frames()
        aligned_frames = align.process(frames)

        aligned_depth_frame = aligned_frames.get_depth_frame()
        color_frame = aligned_frames.get_color_frame()

        if not aligned_depth_frame or not color_frame:
            continue

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        if time.time() - last_saved_time > save_interval and num_saved_images < max_images_to_save:
            cv2.imwrite(f'C:/Users/juseok/Desktop/example/color/sample1_{num_saved_images}.png', color_image)
            cv2.imwrite(f'C:/Users/juseok/Desktop/example/depth/sample1_{num_saved_images}.png', depth_image)
            last_saved_time = time.time()
            num_saved_images += 1

        if num_saved_images >= max_images_to_save:
            break

        cv2.imshow('Depth Image', depth_image)
        cv2.imshow('Color Image', color_image)

        if cv2.waitKey(1) & 0xFF == ord('q'):
            break

finally:
    pipeline.stop()
    cv2.destroyAllWindows()

MartyG-RealSense, 2023-12-06:
Hi @jooseuk  It looks as though your script is already optimizing depth and color quality almost as much as is possible to do so.  You could though try maximizing the RGB image's sharpness using Python code at https://github.com/IntelRealSense/librealsense/issues/12030#issuecomment-1647640046

jooseuk, 2023-12-08:
> Hi @jooseuk It looks as though your script is already optimizing depth and color quality almost as much as is possible to do so. You could though try maximizing the RGB image's sharpness using Python code at [#12030 (comment)](https://github.com/IntelRealSense/librealsense/issues/12030#issuecomment-1647640046)

Thank you. Also, for experimental purposes, if I want to get color and depth images, do I need to calibrate the D435 camera?

MartyG-RealSense, 2023-12-08:
It is not necessary to calibrate the camera in order to obtain depth and color images.  However, the **On-chip calibration** tool can improve depth quality, whilst the **Tare calibration** tool can improve depth measurement accuracy.

closed 2023-12-22 by MartyG-RealSense