realsenseai/librealsense #12492: Question: How to get the Accelerometer and Gyroscope sensitivity values?

milan-r-shah, 2023-12-06:
| Required Info                         |                                                     |
|----------------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.12.14.50 | 
| Operating System & Version  | Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)   |  5.4.0 | 
| Platform                                 | Computer  |
| SDK Version                          | 2.44.0 | 
| Language                              | C++ | 
| Segment			        | {Robot/Smartphone/VR/AR/others} | 

### Issue Description
- As per this page, [How-to: Getting IMU data from D435i and T265](https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/), RealSense D435i (and certain version of D430 modules) have **Bosch BMI055** 6-axis inertial sensor. I'm working on calculating an orientation using the raw data I'm getting from the inertial sensor. In the calculation, one of the required parameters are Accelerometer and Gyroscope sensitivity.

- I tried taking a look into the BMI055 [specs sheet](https://www.bosch-sensortec.com/products/motion-sensors/imus/bmi055/) and found this: 

[image]

How can I get the range values so that I can get the sensitivity values? I couldn't find any functions in the API for that.

MartyG-RealSense, 2023-12-06:
Hi @milan-r-shah  Although the Bosch IMU component supports multiple sensitivity values as described above, the component's implementation in RealSense only supports one value for accel and gyro.  That is +- 1000 deg/s for gyro and +4 g for accel.

In regard to accel, the ideal situation is to have the range as low as possible.  The lower the range, the more sensitive the measurement is.  If the range is set high then it can cause problems, such as a flying drone vehicle that the camera is attached to moving faster than the sensor can keep up with (e.g the vehicle reaches the ground before the sensor detects the arrival).

milan-r-shah, 2023-12-07:
Got it. Thank you so much @MartyG-RealSense for such a prompt response!

closed 2023-12-07 by milan-r-shah