realsenseai/librealsense #12493: API documentation examples

jpfsaunders, 2023-12-06:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 2.54.2.0 | 
| Operating System & Version |   {Linux (Ubuntu )  | 
| Kernel Version (Linux Only)    |                       | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  {  2.<?>.<?> }    | 
| Language                            |  {C++ }                          | 
| Segment			|  { }                          | 

### Issue Description
Is there better (ie, more in-depth) information available for the API?  I have gone through the examples several times and generally have had success over the past 1.5 years initially adapting them to create my own applications in C++ based off what I can find from the github examples or from others who may have posted on the web.  However, very few of all the available class functions are documented with more info than just their name (ie, I am looking for info as to how to actually use them, the options available, etc - more like the online documentation for openCV for example).  It seems weird that somebody went to all the work to design the API for developers to use, but there is minimal documentation for the member functions other than to list their names.  Is there an internal Intel document that describes these in more detail, and which could be made available to show why I would use one function vs some other similar-sounding one?  The existing examples and the API How-To are fine for getting started, but they only cover the basics.  In my case I specifically would like this for the D455 camera.

MartyG-RealSense, 2023-12-06:
Hi @jpfsaunders  You may find the menu-driven version of the API C++ documentation at the link below to be an easy way to navigate the RealSense API.  It draws data directly from the official documentation pages and formats it into a user-friendly interface.

https://unanancyowen.github.io/librealsense2_apireference/classes.html

Sometimes the Python pyrealsense2 version of the official documentation has better descriptions for a particular API instruction than the C++ site.

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.html

The documentation hub-page is also an excellent all-in-one place for RealSense resources, including an extensive list of white-paper guides.

https://dev.intelrealsense.com/docs

Other useful API pages can be found here:

https://dev.intelrealsense.com/docs/api-architecture

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md

The 'Projection' page also contains key information.

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20

jpfsaunders, 2023-12-06:
Hi @MartyG-RealSense
Thank you for your quick response.

closed 2023-12-06 by jpfsaunders