realsenseai/librealsense #12495: Noisy pointcloud when getting too close to an object (in the range of 8-5 cm) when Laser is on

hatem-darweesh, 2023-12-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.15.01 | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  |                                   
| Platform                                 | several platforms |
| SDK Version                            |        2.53.1                    | 
| Language                            |  C++, ROS, realsenseviewer                         | 
| Segment			|  robot                         | 

### Issue Description

How to check: 
1- any configuration 
2- output stream pointcloud. (also the noise appear in the depth output)
3- choose 3D view in realsense viewer or view pointcloud in ROS. 
4- Next image shows when the camera is > 10 cm from the wall. 
[image]

5- Next image shows output of getting between 8-6 cm close to the wall: 
7.5 cm:
[image]
6.0 cm
[image]

6- Next image shows when we are closer than 5 cm. 
[image]

I understand that the minimum range is 10, even when setting minimum range to 15  or 20. The problem is the noise appear later when the camera get closer to the wall or an object. 

Then noise is considered as valid pointcloud frame ! 

Expected output: No data on the pointcloud stream line below the defined minimum range. 

This happens only when the laser projector is on. 

Thank, appreciate your support.

MartyG-RealSense, 2023-12-07:
Hi @hatem-darweesh  The D435 camera model is not designed to display distance at distances lower than 10 cm.  You can try to force it to do so with settings such as **Disparity Shift**.  But once you get to 6 cm distance or less, the camera's baseline (the distance between the centerlines of the left and right infrared sensors) is so big that you will be trying to triangulate on objects seen from opposite sides, so it will only work with nearly flat objects.

As the camera's distance from the observed surface gets closer and closer after going below the minimum distance, the depth image progressively breaks up as the camera moves towards the surface.  This is normal and happens with all RealSense 400 Series camera models.

The **D405** camera model is best suited for very close range sensing, as it was designed to provide high quality, high accuracy images at a minimum distance of 7 cm (and as low as 4 cm with Disparity Shift).

Having the projector enabled at very close range is likely to blind the camera due to the intensity of the IR laser light on the surface at that range.  The D405 is not equipped with a projector.

hatem-darweesh, 2023-12-13:
Hi @MartyG-RealSense Thank you very much for your answer. 
I just want to clarify that My objective isn't getting good depth below 10cm. 
I want to prevent any data from being published below 10cm. 
I tried to create a post processing function to limit that, but I discovered that I would have no idea if the noise is actually valid data or not. One reason for that is the closest point distance from the noisy data is > 15cm. 

Is there any option to detect this edge case and stop the camera from generating depth information below 10cm ? 
Thanks again

MartyG-RealSense, 2023-12-14:
You could try setting the Visual Preset camera configuration to **High Accuracy** so that the depth image is more strict about which depth values it has confidence in and permits to be included on the depth image.  More information about setting the preset with C++ code can be found at https://github.com/IntelRealSense/librealsense/issues/10630#issuecomment-1167495608

MartyG-RealSense, 2023-12-28:
Case closed due to no further comments received.

closed 2023-12-28 by MartyG-RealSense

hatem-darweesh, 2023-12-29:
@MartyG-RealSense  It is okay, 
I can't find any logical or easy solution to this issue. 
Tried all pre-settings, the only solution is to disable the projector. But by doing that all data will be noisy not just when I get too close (<8cm). 
Thanks again.

MartyG-RealSense, 2023-12-29:
Hi @hatem-darweesh  If depth quality reduces due to the projector being disabled then increasing the strength of artificial light illumination in the scene may help if it is possible for you to do so.  This is because RealSense 400 Series cameras can alternatively use light to aid depth analysis of surfaces instead of the dot pattern projection.