realsenseai/librealsense #12500: Can't get depth information when facing the  human body plastic model using D405

Jasonwyh, 2023-12-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 5.15.0.2 and other version | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  /                                         | 
| Platform                                 | PC |
| Language                            |  {C++ }                          | 

Hi, I have tested an D405 camera to capture information from the human body model. And the model surface is matte and non-reflective. But the results seems to be very bad. But when I tried the real human skin like my hand and arm, the results was fine. It makes me confused because I need to test on the model and I need it to work out, too. Please help me thanks.

[image]
[image]

MartyG-RealSense, 2023-12-07:
Hi @Jasonwyh  A smooth mannequin body lacks the texture that a real body's skin has, and RealSense cameras have difficulty reading plain / smooth surfaces without assistance.  That assistance could be a pattern of dots cast onto a surface by the camera's projector, or casting a strong light source onto the surface to bring out texture from it.  

The D405 camera model is not equipped with a built-in projector though, so increasing the strength of the light shining onto the mannequin may be the best option.  

Another option would be to use an external infrared pattern projector to cast an invisible dot pattern onto the mannequin that only the camera can see and would not appear on RGB color images of.the mannequin.

MartyG-RealSense, 2023-12-22:
Hi @Jasonwyh  Do you require further assistance with this case, please?  Thanks!

Jasonwyh, 2023-12-22:
> Hi @Jasonwyh Do you require further assistance with this case, please? Thanks!

No and thanks~

MartyG-RealSense, 2023-12-22:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-12-22 by MartyG-RealSense