realsenseai/librealsense #12501: not work now

ranjithlesoko, 2023-12-08:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|241222076966|
|**Firmware Version**|5.15.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|237523063305|
|**Firmware Update Id**|237523063305|

Please provide a description of the problem

ranjithlesoko, 2023-12-11:
hi @MartyG-RealSense  i not change anything still now this is not working   i using this with jetson nx xeiver and my system

ranjithlesoko, 2023-12-11:
i try new version in another system also same error nothing change please help to solve and i try all your  previous step

ranjithlesoko, 2023-12-11:
@MartyG-RealSense yes first time i'm getting this error  after complete the work i put this to intel realsense box

MartyG-RealSense, 2023-12-11:
If you are getting this error for the first time after taking the camera out of storage for a period of time then the camera should be be returned to the retailer that you purchased it from for a refund or replacement, unfortunately, as you will not be able to correct the problem.

MartyG-RealSense, 2023-12-11:
It may be a little more complicated if purchased through Amazon.  In https://github.com/IntelRealSense/realsense-ros/issues/2919#issuecomment-1813826529 I suggested to a RealSense customer who purchased through Amazon to talk to an Amazon customer support representative if they could not return it through their Amazon 'Your Orders' section.

ranjithlesoko, 2023-12-11:
@MartyG-RealSense if they said no mean what i can do for it?

MartyG-RealSense, 2023-12-11:
If they refuse then I will discuss your case with my Intel RealSense colleagues.

ranjithlesoko, 2023-12-11:
@MartyG-RealSense this massege recived  from amazon  see below 
Hi,
Please contact Intel customer care at 000 800 440 2319.
Thanks
Kochar Computers

MartyG-RealSense, 2023-12-11:
I will raise your case with my colleagues and provide them with the information that you have given me.  I will let you know as soon as I have a reply.  Thanks very much for your patience!

ranjithlesoko, 2023-12-11:
thanks for your help if  you want  more details  let me know

ranjithlesoko, 2023-12-11:
@MartyG-RealSense India

ranjithlesoko, 2023-12-12:
@MartyG-RealSense  i want the reason for this issue and how to avoid

MartyG-RealSense, 2023-12-12:
Hardware failure after a period of storage is a very highly rarely occurring problem, so there is not currently a known reason for it, though the environmental conditions in the place where the camera is stored may be a factor.  In one of the very few cases where this issue has occurred, the camera was also located in India.

ranjithlesoko, 2023-12-12:
@MartyG-RealSense any update from your colleagues

MartyG-RealSense, 2023-12-12:
Not yet but it should be soon.

ranjithlesoko, 2023-12-12:
thanks lot and please soon

MartyG-RealSense, 2023-12-13:
Can you confirm please if your camera model is a D435i

ranjithlesoko, 2023-12-13:
@MartyG-RealSense  yes

ranjithlesoko, 2023-12-14:
@MartyG-RealSense any related update

MartyG-RealSense, 2023-12-14:
Yes, please email your contact details to me at the address below and we can discuss further about replacing your camera.  To do this we require your name, a telephone number and a postal address, please.

racsoaurora@gmail.com

ranjithlesoko, 2023-12-29:
i send a mail please reply as soon as @MartyG-RealSense

ranjithlesoko, 2024-01-03:
@MartyG-RealSense please reply as soon as possible

MartyG-RealSense, 2024-01-09:
Hi @ranjithlesoko  Have you had any messages from customs since Intel emailed them, please?

ranjithlesoko, 2024-01-10:
no, i paid the custom duty

MartyG-RealSense, 2024-01-10:
I am sincerely very sorry for the difficulty that you had with customs.

ranjithlesoko, 2024-01-10:
its ok , marty all over now fine

closed 2024-01-15 by MartyG-RealSense