realsenseai/librealsense #12509: Realsense D435 Tare Calibration error

MarkSmith-Bodd, 2023-12-11:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2023-12-12:
Hi @MarkSmith-Bodd  Please first check whether permission to access the camera hardware and permission for applications to access the camera are both set to 'On' in the Windows camera privacy settings.  This can be done using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

MartyG-RealSense, 2023-12-22:
Hi @MarkSmith-Bodd  Do you require further assistance with this case, please?  Thanks!

MarkSmith-Bodd, 2023-12-22:
No. Thank you.

(from phone)

MartyG-RealSense, 2023-12-22:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-12-22 by MartyG-RealSense