realsenseai/librealsense #12512: Repository not working for Ubuntu 18.04

silasalves, 2023-12-12:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Ubuntu 18  | 
| Kernel Version (Linux Only)    |  5.4.0-150-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  v2.54.2                         | 
| Language                            |  C/python                      | 
| Segment			|  Robot                          | 

### Issue Description
I can't install librealsense2 using the instructions in https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

It seems like the apt certificate for Ubuntu 18.04 has expired:

```
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.102.74 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' is not signed.
```

I found the following workaround in issue #11878 (thanks, @bmegli ) which worked for me:

```
sudo apt -o Acquire::https::librealsense.intel.com::Verify-Peer=false update
sudo apt -o Acquire::https::librealsense.intel.com::Verify-Peer=false install librealsense2 librealsense2-net librealsense2-utils librealsense2-gl
```

MartyG-RealSense, 2023-12-13:
Thanks very much @silasalves for the report and the workaround that worked for you.  Package installation problems such as this typically resolve themselves within 24 hours.

MartyG-RealSense, 2023-12-22:
Hi @silasalves  Do you require further assistance with this case, please?  Thanks!

silasalves, 2023-12-24:
@MartyG-RealSense The problem persists. I ran (as per [these instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)):

```bash
sudo mkdir -p /etc/apt/keyrings
curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | sudo tee /etc/apt/keyrings/librealsense.pgp > /dev/null
sudo apt-get install apt-transport-https
echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo `lsb_release -cs` main" | \
sudo tee /etc/apt/sources.list.d/librealsense.list
sudo apt-get update
```

But I still get this errors:

```
Err:3 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease                                         
  403  Forbidden [IP: 52.218.98.98 80]

...

E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.98.98 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' is not signed.
```

MartyG-RealSense, 2023-12-25:
Does it work if you use the curl command below, please?

`curl -s https://librealsense.intel.com/Debian/librealsense.pgp | gpg --show-keys`

silasalves, 2023-12-25:
Ubuntu 18.04 ships with gpg 2.2.4 (no `-- show-keys` option), so I modified the command. This is the command used and its response:

```bash
$ curl -s https://librealsense.intel.com/Debian/librealsense.pgp | gpg --with-colons --import-options show-only --import
pub:-:2048:1:C8B3A55A6F3EFCDE:1515497521:1830943921::-:::scSC::::::23::0:
fpr:::::::::F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE:
uid:-::::1515497521::DF6F359E9DE2C8BB41BA8FB94026C86B0283745E::"CN = Intel(R) Intel(R) Realsense", O=Intel Corporation::::::::::0:
gpg: Total number processed: 1
```

MartyG-RealSense, 2023-12-26:
If package installation is not working for you, are you able to install by running the **/libuvc_installation.sh** build script at the link below?

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

silasalves, 2024-01-05:
@MartyG-RealSense I solved the issue using the workaround by @bmegli. Thanks.

closed 2024-01-05 by silasalves

MartyG-RealSense, 2024-01-05:
@silasalves  I'm pleased to hear about the solution.  Thanks very much for the update!