realsenseai/librealsense #12516: librealsense2-dkms debian package for LTS 6.6 Linux Kernel

ionutnechita-intel, 2023-12-14:
No template - general description.

librealsense2-dkms debian package for LTS 6.6 Linux Kernel

MartyG-RealSense, 2023-12-14:
Hi @ionutnechita-intel  Support for installing from DKMS packages with kernel 6.2 has now been added.  

For newer kernels that are not yet officially supported at the time of writing this, you can bypass the kernel by building the librealsense from source code with CMake and including the flag **-DFORCE_RSUSB_BACKEND=TRUE** in the CMake build instruction.  

An RSUSB = true build of librealsense is not dependent on Linux versions or kernel versions and does not need the application of a patch to the kernel. You can therefore have kernel 6.6 on your computer as librealsense will not be affected by the kernel version.

Instructions for an RSUSB source code build can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

ionutnechita-intel, 2023-12-18:
Hi @MartyG-RealSense ,

I tested today and is working correctly.
I use release archive with -DFORCE_RSUSB_BACKEND=TRUE and is fine with this.

Thank you for all support.

MartyG-RealSense, 2023-12-18:
You are very welcome.  I'm pleased to hear that RSUSB = true is a suitable solution for you.  Thanks very much for the update!

closed 2024-01-09 by MartyG-RealSense