realsenseai/librealsense #12520: Multiple D455 cameras cause computer to crash

jasatron9000, 2023-12-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.15.0-91-generic (uvcvideo: 1.1.1-realsense-1.3.24)                                        | 
| Platform                                 |  PC  |
| SDK Version                            |  2.54.2 (realsense-ros v.2.54.1)| 
| Language                            |  C++ ros2-humble                          | 
| Segment			|  Robot                          |

### Issue Description
 
I made a previous post here regarding an issue with the Linux PC crashing when running a multi-camera setup which can be found here: https://github.com/IntelRealSense/librealsense/issues/12016
 
We're running five D455 Realsense cameras in two seperate machines that are both configured similarly, but we are still getting this issue regarding the entire machine crashing. Three of the cameras are connected together to a USB Hub and the other two are connected to another USB Hub. The cameras are operating at 10Hz and are all being synced using an external signal to all five cameras. 

The hubs we are using is the OWC Thunderbolt Hub OWCTB4HUB5P.
 
Here is the PC specs of both systems:
- **CPU**: Intel i7-12700F
- **Motherboard**: MSI MEG Z690I UNIFY (MS-7D29)
- **GPU**: Nvidia GeForce RTX 3070
- **RAM**: 2x16GB DDR5
 
The librealsense library was installed using the pre-built binaries, but the realsense-ros package was installed from source.

I've also attached the logs generated from journalctl.
[pc18_logs.log](https://github.com/IntelRealSense/librealsense/files/13680934/pc18_logs.log)
[pc18_logs2.txt](https://github.com/IntelRealSense/librealsense/files/13681382/pc18_logs2.txt)
[pc25_logs.log](https://github.com/IntelRealSense/librealsense/files/13680935/pc25_logs.log)

This is also what running `lsusb -t` reports where the hub reports 10GB and the cameras report 5GB:
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 20000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 10000M
        ID 8087:0b40 Intel Corp.
        |__ Port 1: Dev 4, If 5, Class=Human Interface Device, Driver=usbhid, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 1: Dev 4, If 3, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 1: Dev 4, If 1, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 1: Dev 4, If 4, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 1: Dev 4, If 2, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 1: Dev 4, If 0, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 2: Dev 6, If 0, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 2: Dev 6, If 1, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 2: Dev 6, If 2, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 2: Dev 6, If 3, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 2: Dev 6, If 4, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 2: Dev 6, If 5, Class=Human Interface Device, Driver=usbhid, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 3: Dev 8, If 4, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 3: Dev 8, If 2, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 3: Dev 8, If 0, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 3: Dev 8, If 5, Class=Human Interface Device, Driver=usbhid, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 3: Dev 8, If 3, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
        |__ Port 3: Dev 8, If 1, Class=Video, Driver=uvcvideo, 5000M
            ID 8086:0b5c Intel Corp.
```

Would appreciate some help with this issue.
Thanks in advance.

MartyG-RealSense, 2023-12-15:
Hi @jasatron9000  Your previous post does not mention texting for a **memory leak**, where a computer's available memory capacity is progressively consumed over a period of time whilst an application is running until the application becomes more unstable and then can crash.

If you have not already checked for a memory leak, would it be possible for you to do so whilst your application is running using an Ubuntu system monitoring tool such as **htop** please?

jasatron9000, 2023-12-18:
@MartyG-RealSense We've checked the memory cosumption overtime and we didn't see htop report an increase in memory capacity over time. The memory capacity reported by htop was hovering between 11 - 13 GB of usage out of a total of 30 GB. 

Here's a screenshots of htop with the application running

[image]
[image]

MartyG-RealSense, 2023-12-19:
Are the two hubs connected to separate USB ports on the computer, or are they 'daisy-chained' (where the second hub is plugged into a port on the first hub).  The USB standard supports the chaining together of up to 5 hubs.  If your hubs are not daisy-chained then I would recommend trying plugging the second hub into the first if it has an available port on it and seeing how it affects performance.

danieldymondnz, 2023-12-20:
Greetings @MartyG-RealSense, I'm working with @jasatron9000 on this issue. 

We currently have the hubs connected separately:
- 3x camera Thunderbolt hub connected to a USB 3.2 (Type A) port via an optical USB cable
- 2x camera Thunderbolt hub connected to a Thunderbolt 4 (Type C) port via a USB-C to -C cable

I'll make an attempt to link-up the hubs in the coming days and let you know how it goes. We really appreciate the help on this!

MartyG-RealSense, 2023-12-20:
You are very welcome, @danieldymondnz - I look forward to your next report.  Good luck!

I would add that problems with RealSense cameras are more likely to occur with USB C to C cables than a Type-C (USB A to USB C) cable like the official one supplied with the camera.

MartyG-RealSense, 2023-12-30:
Hi @jasatron9000 and @danieldymondnz  Do you have an update about this case that you can provide, please?  Thanks!

danieldymondnz, 2023-12-30:
Hi @MartyG-RealSense, we don't have an update as of yet. We're currently collecting data with the hubs daisy-chained and will be able to report back within the next couple days. Cheers!

MartyG-RealSense, 2023-12-30:
That's no problem at all.   Thanks very much for the update!

danieldymondnz, 2024-01-02:
Hi @MartyG-RealSense, appreciate your patience! We've ran the daisy-chained configuration for a couple of days and haven't observed any notable change in crash frequency. Some notes:

- The 3x camera hub was connected to the 2x camera hub via the optical USB cable to a USB 3.2 10 Gb/s port.
- We can only get a single camera on the 3x camera hub to work reliably. It would appear the system fails to negotate an address with the other two cameras:
```
[    7.140037] usb 4-1.4.1: Device not responding to setup address.
[    7.351989] usb 4-1.4.1: Device not responding to setup address.
[    7.556971] usb 4-1.4.1: device not accepting address 10, error -71
[    7.557444] usb 4-1.4-port1: attempt power cycle
[    8.767940] usb 4-1.4.1: Device not responding to setup address.
[    8.980000] usb 4-1.4.1: Device not responding to setup address.
[    9.184972] usb 4-1.4.1: device not accepting address 11, error -71
[    9.523912] usb 4-1.4.1: Device not responding to setup address.
[    9.739988] usb 4-1.4.1: Device not responding to setup address.
[    9.945010] usb 4-1.4.1: device not accepting address 12, error -71
[    9.945597] usb 4-1.4-port1: unable to enumerate USB device
```
- In one of the system crashes, a kworker with workqueue `usb_hub_wq hub_event` triggers an Oops. This seems to be shortly after a camera is re-discovered by the system. 
[kworker__usb_hub_wq__hub_event.log](https://github.com/IntelRealSense/librealsense/files/13813312/kworker__usb_hub_wq__hub_event.log)

>I would add that problems with RealSense cameras are more likely to occur with USB C to C cables than a Type-C (USB A to USB C) cable like the official one supplied with the camera.

As we're using Thunderbolt 4 Hubs, all the cameras are connected via USB-C to -C cables. Are the issues using USB-C to -C cables related to cable quality or the interface? Is it worth trying to use USB-C to -A cables, with an -A to -C adapter for each camera such that we can preserve the Thunderbolt hubs?

Thanks!

MartyG-RealSense, 2024-01-03:
Cable quality can be an issue, especially if using cables longer than 2 meters.  The chosen cable, which should be a high quality one, should also be one that is designed for data transfer and not just device charging.

I would not recommend using adaptors, as it may be exchanging one problem for another.  It would be better to continue with USB C to C cables and ensure that they are high quality ones.

Are your hubs powered by the electricity mains socket or are they unpowered ones (which draw the power for their attached devices from the computer).  Mains powered hubs are typically more stable as the power for the devices comes from the mains plug, whilst an unpowered hub draws all the device power from the computer.

danieldymondnz, 2024-01-03:
Thanks for the response!

Our cameras are connected using 2m cables capable of Thunderbolt 3 transfer and charging (Cruxtec CTC-40G-2MBK). Both hubs are mains powered and provide up to 15W per port so should be well in-spec for each camera.

MartyG-RealSense, 2024-01-04:
As the cameras can run for an hour before a problem occurs, the hubs and cabling are likely okay.

You checked already for the possibility of a memory leak and found that one was not occurring.

Have you tried having the cameras run without the external sync signal?

I note that you are operating the cameras at 10Hz (equivalent to 10 FPS).  Problems related to timing are more likely to occur at FPS speeds lower than 15.  Are you able to test your synced cameras at 15 Hz?

danieldymondnz, 2024-01-05:
We have not run the cameras with a higher frequency external sync signal. It will be a few days before I can make the changes to modify the sync frequency and hub setup but I will be able to report back mid next week! Is there a particular reason under 15 FPS is an issue? I can also try running the cameras without external sync to see if it changes the behaviour.

As an aside, we're also considering a revision to the PC hardware and wanted to know whether there are verified PC hardware configurations which RealSense cameras can operate on in a similar configuration?

MartyG-RealSense, 2024-01-06:
When using a low FPS, there is a chance that new frames will sometimes not arrive fast enough and the camera connection will 'time out' whilst waiting for a new frame to arrive, causing an error.

There is not a validated pre-made computer product for RealSense multicam.  In general though, for the number of cameras that you are using on the same computer, at least an i7 is recommended.

For 5 cameras running at low speed, your current computer configuration should be fine.  If you want more processing power then you could consider upgrading from an i7 to an i9 though.  The more cameras that are attached to the same computer, the more of the computer's system resources that are consumed. 

As your current PC has Nvidia graphics, if your application uses pointclouds or depth-color alignment then you may be able to improve performance by enabling **CUDA** support.  This can be done by building librealsense from source code with CMake and including the flag **-DBUILD_WITH_CUDA=TRUE** in the CMake build instruction.  

When librealsense's CUDA support is enabled on a computer with Nvidia graphics, processing work for pointclouds and alignment is offloaded from the CPU onto the graphics GPU.  

However, CUDA support is usually used with Nvidia Jetson computing boards rather than PCs, so there is not a clear guide available for getting CUDA support working on a PC computer.  If your application does not use pointclouds or alignment then there will be no benefit from enabling CUDA support.

danieldymondnz, 2024-01-10:
Hi @MartyG-RealSense, I've began testing with a higher sync frequency (16 Hz) and the cameras restored to a non-daisy-chain configuration. System stability has worsened.

In 8 boots:
- 4 boots had the system stay up for more than 2 hours, but cameras would begin to become unavailable with "device not accepting address" error (like that when the cameras were all daisy-chained). Cameras would become available again once both the PC and the Thunderbolt Hubs are power-cycled.
- 4 boots had the system crash within 2 hours, averaging ~52 mins (01h 30m max, 15m min). In one crash, a kernel oops relating to `iwlwifi` displayed (I was unable to capture this before the machine rebooted). kdump is not triggered on any boot, and there is no useful information in the system logs after the system is rebooted.

We've maintained our resolution settings: 1280x720 for depth and 848x480 for color. At 16 Hz, I'm not seeing saturation of the USB interfaces or hubs when profiling usage via `usbtop` (maximum combined in/out traffic per-camera is ~50000kbps).
 
I note we posted logs from several crashes at the start of this thread. Have you been able to review them and derive any useful information?

> There is not a validated pre-made computer product for RealSense multicam. In general though, for the number of cameras that you are using on the same computer, at least an i7 is recommended.

Are there any known working setups for multicam? It doesn't have to be pre-made or formally validated, just a set of components that has previously been used in a stable multicam setup? We're quite heavily invested in the Realsense product and this stability issue is currently the most critical issue blocking us from deploying our product.

Thanks!

MartyG-RealSense, 2024-01-11:
The most useful message provided by the logs is "Failed to query (GET_CUR) UVC control 1 on unit 3", indicating a potential conflict between librealsense and the kernel.  There was a case with this error in 2020 where I consulted my Intel RealSense colleagues about it.  They provided the response below.

**************

The message “uvcvideo: Failed to query (GET_CUR) UVC control” indicates a failure on the kernel driver level to read the requested parameter, for instance if the device was busy with internal operation or that the incoming requests queue is full. 
 In most cases this message can be disregarded as the SDK has a built-in auto-retry mechanism to compensate for the temporalities.

**************

Whilst building librealsense from source code with the build flag **-DFORCE_RSUSB_BACKEND=TRUE** can help to resolve kernel conflicts by bypassing the kernel, RSUSB = true is suited to single-camera applications rather than multiple camera ones.  For multicam applications, installing librealsense from DKMS packages or building from source code and patching the kernel is preferable.  

At the beginning of this case, it is stated that "the librealsense library was installed using the pre-built binaries", which sounds like the DKMS packages were used. Is that correct, please?  If it is, were the 'distribution_linux.md' instructions at the link below used?

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

You built librealsense from binaries and the RealSense ROS wrapper from source code.  Are you using the ROS1 or ROS2 wrapper?

If the ROS1 wrapper was used and the wrapper had been installed with the command `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` then this could create a problem if librealsense has already been installed from binaries before the wrapper was installed.  This is because the sudo apt-get method installs both librealsense and the ROS wrapper at the same time, so if librealsense had already been installed then you could end up with two conflicting installations of librealsense on the computer.

It is also important to ensure that the ROS wrapper version that you are using matches with a specific librealsense version.  The ROS1 wrapper is not compatible with librealsense 2.54.2 and only supports up to 2.51.1.  For the ROS2 wrapper, 2.54.2 must be matched with wrapper **4.54.1**.  There is not a wrapper for librealsense 2.54.1, but 2.54.2 was mostly bug-fixes rather than new features and so should be compatible with wrapper 4.54.1.

Intel has previously tested a multicam setup in their multiple camera white-paper document using Amazon's own-brand **AmazonBasics** hub.  I have used this hub brand with RealSense cameras for years based on this recommendation and it has proven highly reliable.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#2-multi-camera-considerations

danieldymondnz, 2024-01-16:
Thanks @MartyG-RealSense and apologies for the delayed response!

Firstly to answer those questions:

> ... "the librealsense library was installed using the pre-built binaries", which sounds like the DKMS packages were used. Is that correct, please? If it is, were the 'distribution_linux.md' instructions at the link below used?

That's correct. I've double-checked the documentation against our setup notes. I've also confirmed those DKMS packages are installed, and the uvcvideo module has the RealSense patch applied.

> You built librealsense from binaries and the RealSense ROS wrapper from source code. Are you using the ROS1 or ROS2 wrapper?

> The ROS1 wrapper is not compatible with librealsense 2.54.2 and only supports up to 2.51.1. For the ROS2 wrapper, 2.54.2 must be matched with wrapper 4.54.1. 

Our ROS2 wrapper v4.54.1 is running with librealsense v2.54.2. I've noted a warning when running the wrapper that it was built with librealsense v2.53.1.

I've done a re-build of the wrapper to fix this, but subsequent testing has shown no effect on system stability.

> Intel has previously tested a multicam setup in their multiple camera white-paper document using Amazon's own-brand AmazonBasics hub. I have used this hub brand with RealSense cameras for years based on this recommendation and it has proven highly reliable.

Am I correct in thinking the multi-cam documentation refers to this [4-port hub](https://www.amazon.com/AmazonBasics-Port-2-5A-power-adapter/dp/B00DQFGH80)?

In the meantime, we've collected more data with two systems:

On the current system, the following changes were made:
* Inter Cam Sync Mode set to 0
* External Sync generator disconnected
* Depth and RGB streams set to 15 FPS

Performance of the system is similar to the previous test, with either crashes in the <2hr time frame or cameras becoming unavailable.

On another system, the following changes were made:
* Inter Cam Sync Mode set to 0
* External Sync generator disconnected
* Depth and RGB streams set to 15 FPS
* Connected cameras directly to USB 3.1/3.2 ports on the Motherboard (using [Belkin 1m USB-A to USB-C 10 Gbps cables](https://www.amazon.com/BELKIN-F2CU029bt1M-BLK-USB-USB-C-Cable/dp/B00WJSPB5A))

Performance of this system is again similar to the previous tests.

I have got both systems running at 5 Hz now to see if this has an impact also.

I've also began configuring a NUC11TNBi7 to best replicate the setup trialled in the multi-cam documentation to see if the issue can be re-produced.

As connecting the cameras directly to the PC doesn't appear to have any impact on stability, I would assume that changing to the AmazonBasics hub would not make a difference? The only reason I could think it might is if the Motherboard's USB ports did not provide sufficient power output.

MartyG-RealSense, 2024-01-17:
> Am I correct in thinking the multi-cam documentation refers to this 4-port hub?

It has been over 5 years at the time of writing this since that multi-cam document was first authored but that is the **current** version of the AmazonBasics 4-port hub, yes.

A key benefit of a mains powered hub is greater stability of power supply to the USB ports, since the NUC's ports will draw power from the computer's power supply rather than from the mains power socket.

MartyG-RealSense, 2024-01-26:
Hi @danieldymondnz  Do you require further assistance with this case, please?  Thanks!

danieldymondnz, 2024-07-05:
Following up on this one for future reference.

We managed to isolate the issue down to something either at the kernel or hardware level. We suspect the issue is related to the Intel xHCI controller implementation on our kernel, which can panic under heavy load. Some similar issues can be found online for 5.15 and later kernels: [1](https://bugs.launchpad.net/ubuntu/+source/linux-signed-hwe-5.15/+bug/2013390), [2](https://bbs.archlinux.org/viewtopic.php?pid=2093265#p2093265), [3](https://bugzilla.kernel.org/show_bug.cgi?id=217242).

We have moved to an AMD-based platform, which eliminates the Intel xHCI Controller and have been unable to replicate the issue.