realsenseai/librealsense #12522: [L515] specific point transformation from in color image to depth image

JeongJunLee, 2023-12-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515                                                       | 
| Firmware Version                   | 1.5.8.1                                                    | 
| Operating System & Version |   Win 10                                                 | 
| Kernel Version (Linux Only)    |  N/A                                                      | 
| Platform                                 | PC                                                         |
| SDK Version                            |  Intel.RealSense.SDK-WIN10-2.54.2.5684  | 
| Language                            |  C/C++                                                     | 
| Segment			|  Robot                       | 

I could successfully align color image with depth image using rs2::align class. 
But "rs2::align class" alignment provides only the whole image level transformation.  

I want to some specific points (x, y) in color image to be transformed to depth image frame. 
I could not find any such API until now. 

Please, can you let me know any such method to do it? or any API, if provided? 

Thanks in advance.

JeongJunLee, 2023-12-18:
More specifically, 
I want to know the coordinate (x_depth, y_depth) in depth image, provided the coordinate (x_color, y_color) in color image. 
Regards.

MartyG-RealSense, 2023-12-18:
Hi @JeongJunLee  The RealSense SDK instruction **rs2_project_color_pixel_to_depth_pixel** converts a single specific color pixel to a depth pixel.

https://intelrealsense.github.io/librealsense/doxygen/rsutil_8h.html#af04b7b1ba50b1d9e03089d1e5186d2f8

https://github.com/IntelRealSense/librealsense/issues/6239#issuecomment-615249398 has a C++ script that demonstrates its use.

```
 rs2_project_color_pixel_to_depth_pixel(dpt_tgt_pixel, reinterpret_cast<const uint16_t*>(depth_frame.get_data()), scale, 0.1f, 2,
&depth_intrin, &color_intrin,
&color2depth_extrin, &depth2color_extrin, rgb_src_pixel);
```

JeongJunLee, 2023-12-19:
Hi MartyG-RealSense, 

Thank you very much for your great help!!! 
That's what I needed. 
I appreciate~~

MartyG-RealSense, 2023-12-28:
Hi @JeongJunLee  Do you require further assistance with this case, please?  Thanks!

JeongJunLee, 2023-12-29:
No further assistance needed. 
Thanks a lot. 
Issue closed.

closed 2023-12-29 by JeongJunLee