realsenseai/librealsense #12523: Field of view at different resolutions

avadosf, 2023-12-18:
Hi, 
I'm interested in what the viewing angle will be when shooting at 640 by 480 resolution. If you can tell me what FOV at other resolutions, I would be grateful. 
I have seen this information.
[image]

MartyG-RealSense, 2023-12-18:
Hi @avadosf  The field of view of the physical sensors listed in tables such as the one above is fixed and cannot be changed.

However, each resolution has its own **image** field of view size as part of the camera's **intrinsics** information.  The intrinsics of each individual RealSense camera are unique due to the manufacturing process in the factory.  

You can list the full intrinsics of each supported resolution for your particular camera, including that resolution's image field of view, by using the 'rs-enumerate-devices' tool.  Launching the tool with the instruction below will open it in calibration information mode.

**rs-enumerate-devices -c**

Below is a listing of the intrinsics from my own D455 using this calibration information mode.

[image]

avadosf, 2023-12-22:
[image]
Could you please explain what this information is?

MartyG-RealSense, 2023-12-22:
The extrinsic information describes the difference in physical position between two sensors on the same camera.  For example, the image above describes the positional difference between the depth sensor (the left infrared sensor) and the RGB color sensor.

avadosf, 2023-12-22:
Can you please tell me how to find the distance between sensors in centimeters using the information above?
I calculated the distance between them to be 5.9 centimeters based on this drawing.
[image]

MartyG-RealSense, 2023-12-22:
The distance between sensors is measured from the centerline of the sensors.  On the diagram, the distance between the left and right infrared sensors is 95 mm (9.5 cm).  The distance between left stereo (the origin of depth) and RGB is 59 mm (5.9 cm).

avadosf, 2023-12-22:
yeah, I know.
I want to know the physical distance between sensors using rotation matrix, is it possible?

MartyG-RealSense, 2023-12-22:
I am not certain but as your depth to color translation vector is -0.059, this likely represents the horizontal distance between depth and color sensors in meters (59 mm / 1000 = 0.059 m), with the minus symbol indicating that the position of the color sensor is to the left of the depth sensor's centerline (if it was positioned to the right of the depth centerline then it would be a positive value).

avadosf, 2023-12-22:
what does the second and third number in the translation vector mean?

MartyG-RealSense, 2023-12-22:
As the translation is a 3D vector, the other values are likely vertical y-height and forward-back z-distance parameters.  Y has -6 on the end of its value, indicating that it is not -8 but instead is a very small number, close to zero.  Likewise, Z is -0.0005, indicating that there is virtually no positional difference vertically and back-forward between the depth and color sensors, and the main difference is in the horizontal distance.

avadosf, 2023-12-25:
Please comment on what the Intrinsic values mean.
[image]
The parameters of interest are  PPX, PPY, Fx, Fy.

closed 2023-12-25 by avadosf