realsenseai/librealsense #12527: T265 does not appear in the realsense-viewer in Qualcomm RB5 : No devices was found

buicongnguyen80, 2023-12-19:
I found that the issue is quite similar to my issue: 
https://github.com/IntelRealSense/librealsense/issues/11890

Then I do as the guide here: 
https://developer.qualcomm.com/forum/qdn-forums/hardware/qualcomm-robotics-rb5-development-kit/68234#comment-22103
and also with this link:
https://www.macnica.co.jp/en/business/semiconductor/articles/qualcomm/135331/

I do as this guide but when I connect T265 to Qualcomm RB5 device and run realsense-viewer. I receive message:
There are 0 connected RealSense devices.
What should I do to have T265 displayed on realsense-viewer ?

When I use dmsg and lsusb and usb-devices to check the connected device, 
It look like this problem :
https://github.com/IntelRealSense/librealsense/issues/4681

However, in RB5, I plug and unplug the T265 but there is no change. No device was detected.
Thank you and still wait for the support.

pat19218, 2023-12-19:
Ensure that you are using the correct version of realsense-viewer. T265 was discontinued and removed from the reading software.

buicongnguyen80, 2023-12-21:
Yes, I use according to the guide from Qualcomm : https://developer.qualcomm.com/forum/qdn-forums/hardware/qualcomm-robotics-rb5-development-kit/68234#comment-22103 
In this link, they clone with the branch 2.53 which still support T265. 
The remain issue is: 
T265 still recognized as another device: 
16532.032728] usb 1-2.3: new high-speed USB device number 88 using tegra-xusb
[16532.054736] usb 1-2.3: New USB device found, idVendor=03e7, idProduct=2150
[16532.054753] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16532.054765] usb 1-2.3: Product: Movidius MA2X5X
[16532.054777] usb 1-2.3: Manufacturer: Movidius Ltd.
[16532.054787] usb 1-2.3: SerialNumber: 03e72150

pat19218, 2023-12-21:
You should try version (v2.50.0) or (v2.51.1).
If the problem persists, you should try updating the firmware using realsense-viewer.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

buicongnguyen80, 2023-12-22:
Yes, I have just reset RB5 board, install again from beginning. 
Then I do: 
https://github.com/IntelRealSense/librealsense/blob/R/251/doc/libuvc_installation.md 
Make sure no RealSense device is connected
Open the terminal, run:
$ wget https://github.com/IntelRealSense/librealsense/raw/master/scripts/libuvc_installation.sh
$ chmod +x ./libuvc_installation.sh
$ ./libuvc_installation.sh
Wait until Librealsense script completed message is displayed (may take some time)
Connect RealSense device
Run rs-enumerate-devices from the terminal to verify the installation

But the problem is still there. 
I use usb-devices then I have seen the camera is changed into Movidius Ltd. 

[image]

I opened realsense-viewer but no device was showed up. 
I test with PC Windows 11, there is no problem with the camera and cable. 
May you suggest the way to fix it ? 
I have seen similar issues with NVIDIA Jetson but it was fixed long time ago. May you add the fix to RB5 Qualcomm too ?

closed 2026-10-23 by MartyG-RealSense