realsenseai/librealsense #12529: <arg name="filters" default=""/>

HHU-YHJ, 2023-12-19:
device:D435i
description:when use D435i camera in ros,I have a question for paramter named "filters" in my launch file which is below.How can i use this param?????????
[image]

I know it from website :https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/readme.md
In the example displayed on the above website, it outputs the image after filtering and provides a GUI interface that adjusts the filter parameter. I want to ask after the filtering is enabled in this interface, and the topic topics published by ROS, such as:/camera/depth/image_rect_raw, will it change?

MartyG-RealSense, 2023-12-19:
Hi @HHU-YHJ  In the RealSense ROS1 wrapper, you can apply a post-processing filter by putting the term **filters:=** in your roslaunch command and then the name of the filter after it.  For example, to apply a **decimation** filter:

`roslaunch realsense2_camera rs_camera.launch filters:=decimation` 

Once a decimation filter is applied by enabling it in the roslaunch command then its strength value (**Magnitude**) can be configured with dynamic_reconfigure during runtime after launch, as demonstrated at https://github.com/IntelRealSense/realsense-ros/issues/583

To apply a different filter, instead of 'decimation' use a filter name from the list below of filters supported in the ROS1 wrapper.

disparity
spatial
temporal
hole_filling

In the ROS1 wrapper, dynamic_reconfigure parameters can be accessed and changed in a GUI by inputting the **rosrun** command below after launch has completed.

`rosrun rqt_reconfigure rqt_reconfigure`

[image]

HHU-YHJ, 2023-12-19:
ok,I will try it later,thanks you very much for your profession guide!!!

closed 2023-12-29 by HHU-YHJ