realsenseai/librealsense #12530: OUTIL CALIBRAGE D430

yvespon, 2023-12-19:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D430|
|**Serial Number**|051422072694|
|**Firmware Version**|5.15.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|051323023365|
|**Firmware Update Id**|051323023365|

Please provide a description of the problem

MartyG-RealSense, 2023-12-20:
Hi @yvespon  Can you provide more information about your problem, please?   Thanks!

yvespon, 2023-12-20:
Bonjour
, je suis Français et ne parle pas anglais . J'ai 64 ans et débute
complètement cette nouvelle activité. Je pratique l'imprimante 3D depuis
juin et j'arrive à être performant avec Freecad et Elegoo Neptune 2
J'ai acheté une caméra intel D430 et réussi à charger l'application et
faire fonctionner la caméra avec  #Intel.RealSense.Viewer#
Mais après avoir fait plusieurs réglages , je me suis apperçu qu'il y avait
de l'ombre dans l'application #Intel.RealSense.Viewer# ....
Impossible de remettre comme au début ...Voir image jointe, c'est un livre
debout . En vert le livre photographié en rouge c'est l'ombre .
En faite il y a de très nombreux réglages , et malheureusement je n'ai pas
enregistré de sauvegarde......

J'ai cherché .... Rien de concluant ...
J'ai vu qu'il fallait calibrer la caméra ...
Enfin pour finir comme je suis débutant je ne sais pas par ou commencer

Avez vous, si possible, des tutos vidéo pour débuter et pour les
paramétrages que je puisse progresser

J'ai l'impression que c'est compliqué

Merci d'avance
Yves

[image: CAMERA.png]

MartyG-RealSense, 2023-12-28:
Hi @yvespon  Do you require further assistance with this case, please?  Thanks!

yvespon, 2023-12-28:
bonjour

pour l'instant je ne m'en occupe pas , je verrai plus tard

Merci

Yves

MartyG-RealSense, 2023-12-29:
Vous êtes les bienvenus. Merci beaucoup pour la mise à jour. Comme vous n’avez actuellement pas besoin d’aide supplémentaire, je clôturerai ce dossier. Merci encore!

closed 2023-12-29 by MartyG-RealSense