realsenseai/librealsense #12532: Sementation fault

jiaqizheng2000, 2023-12-20:
----------------------------------------------------------------------------------------------------

PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/

### Issue Description
when I run python manage.py
the get Segmentation fault

MartyG-RealSense, 2023-12-20:
Hi @jiaqizheng2000  Have you installed the librealsense SDK on your Linux 11 (Bullseye) computer, please?

jiaqizheng2000, 2023-12-20:
Yes，I have installed it following the guidance in github.-------- 原始邮件 --------发件人： MartyG-RealSense ***@***.***>日期： 2023年12月20日周三 15:06收件人： IntelRealSense/librealsense ***@***.***>抄送： jiaqizheng2000 ***@***.***>, Mention ***@***.***>主    题： Re: [IntelRealSense/librealsense] Sementation fault (Issue #12532)
Hi @jiaqizheng2000  Have you installed the librealsense SDK on your Linux 11 (Bullseye) computer, please?

jiaqizheng2000, 2023-12-20:
Yes, I have installed it following https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide#prerequisites

jiaqizheng2000, 2023-12-20:
and I can open it by command realsense-viewer, and see the image

MartyG-RealSense, 2023-12-20:
Is your Python program a **pyrealsense2** one?  If it is, have you installed the RealSense pyrealsense2 wrapper after you installed librealsense?

On a PC computer you can install the pyrealsense2 wrapper with the command **pip install pyrealsense2**

jiaqizheng2000, 2023-12-20:
I have installed pyrealsense2

jiaqizheng2000, 2023-12-20:
is there some problem with VERSION="11 (bullseye)"? Do I need to change to Buster? Will it work?

MartyG-RealSense, 2023-12-20:
The Raspberry Pi version of Bullseye does not work with RealSense, while Buster does.

I do not know whether the Debian Bullseye has the same problem though.

jiaqizheng2000, 2023-12-20:
ok, I will try The Raspberry Pi version of Buster to see whether it works. Could you send me a guidance or just need to do the same process as before.

jiaqizheng2000, 2023-12-20:
ok, thank you very much, I will try it tomorrow

jiaqizheng2000, 2023-12-24:
I have a problem with Buster using raspberry pi 4, it shows Unable to read partition  as FAT

jiaqizheng2000, 2023-12-24:
E: Unable to locate package libglui-dev
E: Unable to locate package libglui2c2

There are two errors when I install the package, I just skip them, does they matter?

MartyG-RealSense, 2023-12-24:
I believe that these packages are related to GLFW graphics support and are installed when using Ubuntu 18. 20 and 22.  So they may not be necessary on your Debian based installation.

jiaqizheng2000, 2023-12-24:
Ok,thanks.

And I also have some issues with protobuf.
when I run ./autogen.sh, it shows as below.
+ autoreconf -f -i -Wall,no-obsolete
configure.ac:30: error: possibly undefined macro: AC_PROG_LIBTOOL
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1

MartyG-RealSense, 2023-12-24:
The protobuf version in the instructions is very old because the guide is old.  Using protobuf v3.14.0 instead of v3.5.1 may work better for you.

jiaqizheng2000, 2023-12-24:
ok, and I have installed pyrealsense2 following the guide, but when I import pyrealsense2, it still shows no module named pyrealsense2

MartyG-RealSense, 2023-12-24:
Does it make a difference if you use **import pyrealsense2.pyrealsense2 as rs** instead of import pyrealsense2 as rs

jiaqizheng2000, 2023-12-24:
No, it doesn't work. And I saw some online guide said that need to move some *.so、so.2、so.2.50、so.2.50.0 file to /usr/lib/python3 from librealsense/build/wrappers/python, but I didn't see them when I have done pyrealsense2 installation.

MartyG-RealSense, 2023-12-24:
Do you have any files with the ending .so in the build/wrappers/python folder, please?

As you are using Python 3, the pyrealsense2.so file will have a complex auto-generated filename based on the Python 3 version that you are using.  So if Python 3.7 was being used then the pyrealsense2.so file might have a filename like **pyrealsense2.cpython-37m-arm-linux-gnueabihf.so**

jiaqizheng2000, 2023-12-24:
No，I didn't find any files ending with .so in the librealsense/build/wrappers/python folder

jiaqizheng2000, 2023-12-25:
[image]
[image]

MartyG-RealSense, 2023-12-25:
If you run **pip install pyrealsense2** again and then try your manage.py program, do you still get the segmentation fault?

jiaqizheng2000, 2023-12-26:
when I run pip install pyrealsense2, or pip3 install pyrealsense2, it will raise the error:

 Could not find a version that satisfies the requirement pyrealsense2 (from versions: )
No matching distribution found for pyrealsense2

[image]

MartyG-RealSense, 2023-12-26:
Which Python 3 version are you using, please?  That pip install error could occur if a Python 3 version older than 3.7 (such as 3.6) was being used.

jiaqizheng2000, 2023-12-26:
I am using python 3.7.2

MartyG-RealSense, 2023-12-26:
Is your computer a desktop / laptop PC or does it use Arm architecture?  If it is based on Arm then the minimum Python requirement would be 3.9 instead of 3.7.

jiaqizheng2000, 2023-12-26:
It works in my computer，but I want to use it in raspberry pi 4B which I am using Buster 2021-5-21

MartyG-RealSense, 2023-12-26:
I've had previous reports that the Arm files of pip install pyrealsense2 work with Arm in general but not Raspberry Pi in particular, so that may be the reason why pip install is not working for you.  In that case, building pyrealsense2 from source code would be the only option, unfortunately.

You have already tried building the wrapper from source on your Pi, but it is not clear why the .so files are not being generated in the build/wrappers/python folder.  You do not seem to be doing anything incorrect.

jiaqizheng2000, 2023-12-26:
Ok，thanks for your kind help.

MartyG-RealSense, 2024-01-02:
Hi @jiaqizheng2000  Do you require further assistance with this case, please?  Thanks!

jiaqizheng2000, 2024-01-02:
No，thanks

closed 2024-01-02 by jiaqizheng2000