realsenseai/librealsense #12537: FW update failed

perovehuddly, 2023-12-22:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D4XX USB2 Recovery|
|**Product Line**|D400|
|**Firmware Update Id**|033423022101|

Please provide a description of the problem
 Automatic upload of SW with realsense.viewer realsense 435 failed multiple times

MartyG-RealSense, 2023-12-22:
Hi @perovehuddly  If the camera is identified as **Intel RealSense D4XX USB2 Recovery** then this indicates that it is in recovery mode, likely due to a failed update.  

The RealSense firmware update tool **rs-fw-update** can recover cameras that are in recovery mode.

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update

A pre-built executable version of this tool for Windows can be obtained if you install the full RealSense SDK using the **Intel.RealSense.SDK-WIN10** installer file in the 'Assets' file list of the Releases page.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

Instructions for running the rs-fw-update program in the Windows command prompt interface and initiating recovery can be found at https://github.com/IntelRealSense/librealsense/issues/8500#issuecomment-791540583

As you are using RealSense Viewer version 2.54.2, the bin file for firmware **5.15.1.0** should be downloaded from the link below before using rs-fw-update.

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

MartyG-RealSense, 2024-01-05:
Case closed due to no further comments received.

closed 2024-01-05 by MartyG-RealSense

perovehuddly, 2024-01-11:
Hi... I have not had time to test your suggestion yet (this is a side
project), I'll come back to it later.