realsenseai/librealsense #12539: Error on attempting On-chip calibration

beratna-steve, 2023-12-24:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2023-12-25:
Hi @beratna-steve  Please first check whether permission to access the camera hardware and permission for applications to access the camera are both set to 'On' in the Windows camera privacy settings.  This can be done using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

MartyG-RealSense, 2024-01-02:
Hi @beratna-steve  Do you require further assistance with this case, please?  Thanks!

beratna-steve, 2024-01-02:
I do not!  I didn't have enough light.  Once I got more light on the wall,
it worked beautifully.

closed 2024-01-02 by MartyG-RealSense