realsenseai/librealsense #12542: ASIC and projector temperatures

avadosf, 2023-12-27:
Hello,
I am interested in how ASIC and projector temperatures affect depth accuracy. I have studied the information from here. https://github.com/IntelRealSense/librealsense/issues/10476
But there are still questions about recommended temperatures for ASIC and projector separately.

MartyG-RealSense, 2023-12-27:
Hi @avadosf  The official recommended maximum operating temperature (**ASIC** temperature) is 35 degrees C, though in practice it could go as high as 42 degrees before problems with the image might start manifesting above that point.

When the **projector** temperature exceeds 60 degrees C, the firmware driver's laser safety mechanism will activate.  It will first halve the Laser Power to try to bring the temperature down.  If the temperature continues to exceed 60 degrees then the laser will be shut off.  This does not prevent the camera from operating but is likely to negatively affect accuracy as the infrared dot pattern projection is disabled.  

This often causes depth image quality to be reduced due to the pattern's absence making it more difficult for the camera to analyze surfaces for depth information.

avadosf, 2023-12-27:
What will cause problems when the ASIC heats up? My guess is throttling.

MartyG-RealSense, 2023-12-27:
The problems that occur will likely vary in each situation and there is not information on exactly what will occur.  For electronics in general though, severe overheating can cause short-circuits that impair the hardware's ability to operate correctly.

avadosf, 2023-12-27:
thanks for the information

closed 2023-12-27 by avadosf