realsenseai/librealsense #12543: poll_for_frames still limiting refresh rate to slowest stream

Starblazer75, 2023-12-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version | Ubuntu 20 | 
| Kernel Version (Linux Only)    | 5.10.104-tegra | 
| Platform                                 | Jetson Orin Nano |
| SDK Version                            | 2.x.x | 
| Language                            | Python | 
| Segment			| Robot | 

When coding in Pyrealsense2, I have the depth, color, accel, and gyro streams turned on. When using pipeline.poll_for_frames(), the rate that it returns a valid frameset is limited to the slowest stream even though I am using poll instead of wait. This ends up returning a frameset with all the streams instead of individual frames of each stream. Is there any way to get each individual frame instead of framesets? I'm not seeing the difference between wait_for_frames and poll_for_frames when both of them essentially have the same refresh rate. 

This is running on a ROS2 Humble node refreshing at 500Hz. I have confirmed it is the different streams causing this because when I only enable accel and gyro, it runs at 200Hz but if I enable color (30Hz), it will go down to 30Hz even though I am looking at the accel/gyro frames.

```
def timer_callback(self):
    frames = self.pipeline.poll_for_frames()

    accel_frame = frames.first_or_default(rs.stream.accel)
    gyro_frame = frames.first_or_default(rs.stream.gyro)

    if accel_frame.is_motion_frame() and gyro_frame.is_motion_frame():

        self.counter += 1

        timer = Clock().now().to_msg()
        time = timer.sec + (timer.nanosec / 1000000000)
        time -= self.start_time

        self.get_logger().info(f'{self.counter / time}')

def __initialize_camera():
    # Initializes the camera for all the frames being analyzed
        pipeline = rs.pipeline()
        config = rs.config()
        config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 200)
        config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)
        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 90)
        config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 60)
        pipeline.start(config)

        return pipeline
        
```

This isnt everything, but assume that pipeline in initialize_camera is the same as the one in timer_callback. This code counts the amount of valid motion frames and prints the Hz.

MartyG-RealSense, 2023-12-28:
Hi @Starblazer75  Having depth, color and IMU simultaneously enabled can cause problems in the RealSense SDK.  A workaround is to set up two separate pipelines, with IMU on one pipeline and depth & color on the other pipeline.  The best example of a Python script for implementing this is at https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238

MartyG-RealSense, 2024-01-05:
Hi @Starblazer75  Do you require further assistance with this case, please?  Thanks!

Starblazer75, 2024-01-05:
Hi! Thanks for the help, the problem has been resolved with the comment.

closed 2024-01-05 by Starblazer75

MartyG-RealSense, 2024-01-05:
It's excellent to hear that your problem is solved.  Thanks very much for the update!