realsenseai/librealsense #12545: Realsense Viewer open depth stream exception occured in debug mode!!!

asla1, 2023-12-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I | 
| Firmware Version                   |5.15.1 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  /                                   | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.2                       | 
| Language                            |  C                     | 
| Segment			|  others                        | 

### Issue Description

I compiled the latest master branch of the librealsense SDK using the VS2022 compiler. The debug version I compiled crashes immediately when I run the realsense viewer and start the depth stream with a connected D435I device. The same thing happens when I switch the depth resolution and frame rate. However, starting the RGB or IR streams works normally, as shown in the picture.
[image]

The same phenomenon occurs when I use both VS2015 and VS2019.

However, it works normally in the release version.

I want to know if this is a known issue. Please help me to resolve it!

MartyG-RealSense, 2023-12-28:
Hi @asla1  There was a previous report of this issue at https://github.com/IntelRealSense/librealsense/issues/10299 with the earlier **2.49.0** version of the Viewer when the SDK is built in debug mode.  In that case I suggested disabling **GLSL** support in the Viewer.  Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

asla1, 2023-12-28:
@MartyG-RealSense  Thank you very much! It did help me!

MartyG-RealSense, 2023-12-28:
That's great to hear, @asla1 - thanks very much for the update!

closed 2024-01-05 by MartyG-RealSense