realsenseai/librealsense #12546: Capturing Intel RealSense IMU data for calibration

Mahesha999, 2023-12-30:
I am trying to calibrate IMU from Intel Realsense. Kalibr requires input in rosbag. We can create rosbag from csv as specified [here](https://github.com/ethz-asl/kalibr/wiki/bag-format) using `bagcreater` tool.  However bagcreater requires input csv in following format:

```
timestamp,omega_x,omega_y,omega_z,alpha_x,alpha_y,alpha_z
1385030208736607488,0.5,-0.2,-0.1,8.1,-1.9,-3.3
 ...
1386030208736607488,0.5,-0.1,-0.1,8.1,-1.9,-3.3
```

I was thinking how can I get csv in this format. First task was to log IMU data. I used Intel realsense-viewer to record IMU data. I run it for few seconds to quickly get the rosbag it created for analysis. The rosbag contained many topics (corresponding to each data stream), out of which following two seems to of interest:

```
 /device_0/sensor_2/Accel_0/imu/data                    1269 msgs    : sensor_msgs/Imu  
 /device_0/sensor_2/Gyro_0/imu/data                     2528 msgs    : sensor_msgs/Imu             
```

I was planning to extract these topic using [bagextractor](https://github.com/ethz-asl/kalibr/wiki/bag-format#bagextractor) and then merge them and the create a new bag out of them using [bagcreater](https://github.com/ethz-asl/kalibr/wiki/bag-format#bagcreater) in the format required by kalibr. But, as you can see they contain different number of messages. Gyro logged 2528 messages whereas accel logged 1269 messages. How can I merge them correctly? Or is there different tool that I should be using to capture Intel RealSense IMU data in the rosbag format required by Kalibr? Or should I be using the [realsense API] (https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#api) to log both gyro and accel for each timestamp?

PS: First few rows of CSVs exported from these two topics looks as follows:

**/device_0/sensor_2/Accel_0/imu/data**

```
timestamp,omega_x,omega_y,omega_z,alpha_x,alpha_y,alpha_z
1703886880345048064,-0.6841690540313721,2.3858649730682373,-0.07504915446043015,0.0,0.0,0.0
1703886880350096640,-0.39618974924087524,2.5638885498046875,-0.2687807083129883,0.0,0.0,0.0
1703886880355147008,-0.05235987901687622,2.717477560043335,-0.45902159810066223,0.0,0.0,0.0
```

**/device_0/sensor_2/Gyro_0/imu/data**

```
timestamp,omega_x,omega_y,omega_z,alpha_x,alpha_y,alpha_z
1703886880343395072,0.0,0.0,0.0,-3.5696206092834473,6.531228542327881,7.00194787979126
1703886880353453824,0.0,0.0,0.0,-3.7657535076141357,6.1095428466796875,6.158576011657715
1703886880363513088,0.0,0.0,0.0,-3.255807638168335,6.050703048706055,5.776116847991943
```

Notice how gyro data contains all zeroes for accel columns and vice versa.

MartyG-RealSense, 2023-12-30:
Hi @Mahesha999  I would not recommend calibrating your RealSense camera's IMU using Kalibr.  There is instead an official Python tool provided by Intel for doing so called **rs-imu-calibration.py** that does not require a bag file.

https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-imu-calibration/rs-imu-calibration.py

https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

Mahesha999, 2023-12-30:
> Hi @Mahesha999 I would not recommend calibrating your RealSense camera's IMU using Kalibr. There is instead an official Python tool provided by Intel for doing so called **rs-imu-calibration.py** [...]

Aah, I wasn't fully clear what I was trying to achieve. I indeed calibrated IMU using `rs-imu-calibration.py`.  The final output after calibration process was following:

```
Direction data collected.
Would you like to save the raw data? Enter footer for saving files (accel_<footer>.txt and gyro_<footer>.txt)
Enter nothing to not save raw data to disk. >20dec

Writing files:
accel_20dec.txt
gyro_20dec.txt
[1.01568832e-03 5.89697726e-05 9.94564821e-05]
[1000 1000 1000 1000 1000 1000]
using 6000 measurements.
[[ 1.00794371 -0.00341102 -0.0045452 ]
 [-0.03062543  1.01120542 -0.03208355]
 [-0.00357108 -0.02002484  1.01206203]
 [-0.00248397 -0.20700389  0.15847565]]
residuals: [62.58381636 22.04040843 85.83482748]
rank: 4
singular: [448.50318393 432.87155646 421.80205615  77.37480686]
norm (raw data  ): 9.714921
norm (fixed data): 9.804833 A good calibration will be near 9.806650
rs-imu-calibration.py:725: DeprecationWarning: tostring() is deprecated. Use tobytes() instead.
  outfile.write(imu_calib_table.astype('f').tostring())
Would you like to write the results to the camera? (Y/N)Y
Writing calibration to device.
  Device PID:  0B5C
  Device name:  Intel RealSense D455
  Serial number:  231122300625
  Firmware version:  5.15.1
SUCCESS: saved calibration to camera.
Done.

```
But I wanted to obtain following parameters: Gyro Noise, Accel Noise, Gyro Walk and Accel Walk. I was unable to figure out these parameters from above output. Can you please tell me if there is any way to obtain these values from some official tool? If there is no such official tool, then can you lease provide some solution hint for the original question, since It seems that Kalibr provides these values very clearly.

MartyG-RealSense, 2023-12-30:
There is not an official tool for calculating the random walk of the IMU but some RealSense users have done it using a technique called an **Allan variance plot**.  A link to an example reference for this method can be found below.

https://discourse.ros.org/t/i-will-calculate-noise-coefficients-from-inertial-sensors-calibration-data/24779

**Noise variance** values for gyro and accel can be obtained with the official **rs-enumerate-devices** tool by launching it in calibration information mode with the command **rs-enumerate-devices -c** and looking at the 'Motion Intrinsic Parameters' section of the intrinsics listing, as described at https://github.com/IntelRealSense/librealsense/issues/8899#issuecomment-826709184

Mahesha999, 2023-12-30:
Yes I have tried `rs-enumerate-devices` too. It reported as following:

```
Motion Intrinsic of "Gyro"	  MOTION_XYZ32F
Bias Variances: 	0.000000000000000  0.000000000000000  0.000000000000000  
Noise Variances: 	0.000000000000000  0.000000000000000  0.000000000000000  
Sensitivity : 
     1.000000       0.000000       0.000000       0.000000  
     0.000000       1.000000       0.000000       0.000000  
     0.000000       0.000000       1.000000       0.000000  

Motion Intrinsic of "Accel"	  MOTION_XYZ32F
Bias Variances: 	0.000000000000000  0.000000000000000  0.000000000000000  
Noise Variances: 	0.000000000000000  0.000000000000000  0.000000000000000  
Sensitivity : 
     1.000000       0.000000       0.000000       0.000000  
     0.000000       1.000000       0.000000       0.000000  
     0.000000       0.000000       1.000000       0.000000 
```

Seems that official sdk report noise variances as all zeroes. But turns out kalibr does report it a bit different / very tiny values. Also kalibr also utilizes allan variance plot for calculating random walk. So going to try it out for more precise values. So in the end I really wanted to log the data as desired. It tried to build to build two approaches, one with callback and other without. Here is code for both and data captured by them [drive link](https://drive.google.com/drive/folders/1XuMlpHEgcsHUtuOTkDKOIY9FhH-VHn6A?usp=sharing).

Looking at data generated, 
1. I found following issue with non callback approach, some timestamps repeat couple of times. Also some gyro and accel data repeat, not just consecutively but also in distant records. 
2. Issue with callback approach was that it lacks precision in gyro and accel data
3. Issue with both approach that they does not get timestamp in the same format returned by realsense viewer. Realsense viewer returns timestamp as 19 digit number `1703886880350096640`, while callback approach returns it as `1703952937329.38232421875000000000`. Will merely scaling this value to match 19 digit number do?

MartyG-RealSense, 2023-12-31:
Thank you for the script link but I am unable to test scripts on my computer, unfortunately.

The RealSense SDK has an official C++ callback example called **rs-callback** that you could test as it can provide IMU data.

https://github.com/IntelRealSense/librealsense/tree/master/examples/callback

I do not know the reason for the difference between the size of the timestamp in the Viewer and in your own script, so could not say whether scaling the value would be a correct action.  I do apologize.

MartyG-RealSense, 2024-01-07:
Hi @Mahesha999  Bearing in mind the information in the comment above, is there anything else that I can help with please?  Thanks!

Mahesha999, 2024-01-09:
Nope. Scaling values to match 19 digit number did the trick:

```
file << std::fixed << std::setprecision(0) << accel_frame.get_timestamp() * 1e6f << ",";
```

MartyG-RealSense, 2024-01-09:
It's great to hear that your solution worked.  Thanks very much for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2024-01-09 by MartyG-RealSense