realsenseai/librealsense #12551: lagging and frame drop with realsense d435 on Jetson Xavier

eladpar, 2024-01-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.14.0 | 
| Operating System & Version |   JetPack 5.0.2  | 
| Platform                                 | Jetson Xavier NX  |
| SDK Version                            |  2.54.1                       | 

### Issue Description
Hey, so I've went through some similar questions and the last one was about a year ago.
It seems (using realsense viewer) that the camera has some lags and a frame drop rate of about 10 (meaning 30%) 
with with the default preset.
my librealsense was built with cuda and its 2.54.1 and FW is 5.14
but it seems to happen with FW 5.13
im wondering if its directly linked to the librealsense or its just how it is?
if so, is there a different camera that will yield better results?

MartyG-RealSense, 2024-01-01:
Hi @eladpar  You can disregard the 'Frame Drops Per Second' graph as it is not a reliable measure of performance.  The important information is the **frames per second** (FPS) of the stream in the 'i' information overlay at the top of the stream panel.  Your image shows the FPS to be 30.01, indicating that there is not a lag problem with the stream.

eladpar, 2024-01-01:
> Hi @eladpar You can disregard the 'Frame Drops Per Second' graph as it is not a reliable measure of performance. The important information is the **frames per second** (FPS) of the stream in the 'i' [...]

Thanks for the quick reply,  so let me rephrase:
My problem is the video feed gets stuck for a second every second or two.
I'm trying to use the camera in a deepstream application with c++ realsense libraries,  and the problem happens there too, it gets stuck too often..

MartyG-RealSense, 2024-01-01:
Firmware 5.15.1 is the correct firmware to use with 2.54.2.

The RGB stream should not be getting stuck regularly.

Are you using the official 1 meter USB cable supplied with the camera or a longer USB cable of your own choice, please?

If you are using the official cable, does the problem still occur if you go to the **More** option near the top of the side-panel and select **Reset Hardware** from its drop-down menu to reset the camera.

eladpar, 2024-01-01:
> Firmware 5.15.1 is the correct firmware to use with 2.54.2.
>
> The RGB stream should not be getting stuck regularly.
>
> Are you using the official 1 meter USB cable supplied with the camera or a longer [...]

Official cable.
Tried resetting,  but to no avail. 
It might be the whole jetson freezes.. 
Is there a last known SW and FW for jetsons

MartyG-RealSense, 2024-01-02:
The same SDK and firmware versions that are used on PC can be used with Jetson boards too.  The Arm architecture of single-board computers such as Jetson can sometimes cause complications though.

Which Ubuntu version is your board using?  JetPack 5 is designed for use with Ubuntu 20.04.  If Ubuntu 18.04 is being used then JetPack 4.6 should be used with it.

MartyG-RealSense, 2024-01-09:
Hi @eladpar  Do you require further assistance with this case, please?  Thanks!

eladpar, 2024-01-11:
sadly its still isnt working im wondering if you can reproduce the same setup?
im intending on trying the d405 this week to see if its specifically d435

MartyG-RealSense, 2024-01-17:
Hi @eladpar  Do you have an update about this case that you can provide, please? Thanks!

eladpar, 2024-01-21:
Hey so I checked d405 and it seems fine
still the 435 laggs and I cant see even a few uninterrupted seconds of streaming

MartyG-RealSense, 2024-01-21:
Hi @eladpar  Can you confirm that you enabled the depth stream before opening the Calibration Data interface, please?  The Write Table button is grayed out otherwise if the depth stream has not been started first.

eladpar, 2024-01-21:
> Hi @eladpar Can you confirm that you enabled the depth stream before opening the Calibration Data interface, please? The Write Table button is grayed out otherwise if the depth stream has not been started first.

This Worked!
Any idea why?

MartyG-RealSense, 2024-01-21:
The camera hardware stores calibration data in a Calibration Table.  If the table becomes corrupted then it can cause a camera not to work correctly.  Writing calibration data to the camera creates a new calibration table to replace the damaged one.

eladpar, 2024-01-21:
Great Thanks!

closed 2024-01-21 by eladpar