realsenseai/librealsense #12552: ROS 2 Humble Docker in Raspberry cannot find camera

JLBicho, 2024-01-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | 1.5.8.1 | 
| Operating System & Version |   Raspbian Bullseye 11 aarch 64 | 
| Kernel Version (Linux Only)    |  5.15.76-v8+                                        | 
| Platform                                 | Raspberry Pi 4 4GB |
| SDK Version                            |  2                          | 
| Language                            |  ROS 2 Humble    | 
| Segment			|  Robot            | 

### Issue Description

I have a ROS 2 Humble docker image with realsense2 and librealsense2 installed from debian packages (sudo apt install). Everything related to ROS works. The Raspberry Pi detects the Realsense with lsusb. 

When I try to use docker with the following command:
```bash
docker run -it --name humble_realsense --net=host --env="ROS_DOMAIN_ID=42" --privileged -v /etc/udev/*:/etc/udev/* \
    -v /dev/bus/usb:/dev/bus/usb \
    -v /dev/:/dev/  ros2humble_realsense:latest bash
```

The container does not find the camera:

```
# ros2 launch realsense2_camera rs_launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-01-02-19-00-38-029139-raspberrypi4gb-71
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [72]
[realsense2_camera_node-1] [INFO] [1704222039.488293615] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1704222039.488679241] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1704222039.488760018] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1]  02/01 19:00:39,518 ERROR [547532761312] (librealsense-exception.h:52) /dev/video10 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  02/01 19:00:39,518 ERROR [547532761312] (sensor.cpp:661) acquire_power failed: /dev/video10 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  02/01 19:00:39,610 WARNING [547532761312] (rs.cpp:312) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1704222039.610634600] [camera.camera]: Device 1/7 failed with exception: /dev/video10 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  02/01 19:00:39,622 ERROR [547532761312] (librealsense-exception.h:52) /dev/video12 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  02/01 19:00:39,622 ERROR [547532761312] (sensor.cpp:661) acquire_power failed: /dev/video12 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  02/01 19:00:39,711 WARNING [547532761312] (rs.cpp:312) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1704222039.711763733] [camera.camera]: Device 2/7 failed with exception: /dev/video12 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  02/01 19:00:39,722 ERROR [547532761312] (librealsense-exception.h:52) /dev/video18 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  02/01 19:00:39,723 ERROR [547532761312] (sensor.cpp:661) acquire_power failed: /dev/video18 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  02/01 19:00:39,812 WARNING [547532761312] (rs.cpp:312) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1704222039.812702478] [camera.camera]: Device 3/7 failed with exception: /dev/video18 is no video capture device Last Error: Invalid argument
[ERROR] [realsense2_camera_node-1]: process has died [pid 72, exit code -11, cmd '/opt/ros/humble/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_72waf_w2 --params-file /tmp/launch_params_dfds8f9e'].
```

However, the camera (or at least the IR camera) works fine with python3+OpenCV:

```python
# python3
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
ca>>> cap = cv2.VideoCapture(0)
[ WARN:0] global ./modules/videoio/src/cap_gstreamer.cpp (616) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
>>> ret, frame = cap.read()
>>> ret
True
>>> frame
array([[[ 82,  82,  82],
        [ 81,  81,  81],
        [ 81,  81,  81],
...
```

[image]

I think this issue is related to permissions or USB access in the Raspberry, because the same docker in my laptop works fine with the camera.

Has anyone had the same issues, or can give me any hint? 

Thank you!

MartyG-RealSense, 2024-01-03:
Hi @JLBicho  My research of your question found a case regarding problems with using a RealSense camera with Docker on Pi where a suggested solution was to use **sudo** admin permissions.

```
sudo docker run -it --privileged -v /etc/udev/*:/etc/udev/* \
-v /dev/bus/usb:/dev/bus/usb \
-v /dev/:/dev/ \
-v /tmp/.X11-unix:/tmp/.X11-unix \
-e DISPLAY=$DISPLAY -h $HOSTNAME -v $HOME/.Xauthority:/home/.Xauthority realsense:latest bash
```

JLBicho, 2024-01-03:
Hello @MartyG-RealSense, thank you for the quick reply!

I've tried the launching with 'sudo' and now a new error occurs:

```
$ sudo docker run -it --name humble_realsense_sudo --net=host --env="ROS_DOMAIN_ID=42" -v /etc/udev/*:/etc/udev/*     -v /dev/bus/usb:/dev/bus/usb     -v /dev/:/dev/  ros2humble_realsense:latest bash
root@raspberrypi4gb:/# source /ros_entrypoint.sh 
root@raspberrypi4gb:/# ros2 launch realsense2_camera rs_launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-01-03-11-46-22-567763-raspberrypi4gb-65
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [66]
[realsense2_camera_node-1] [INFO] [1704282384.581281165] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1704282384.583020971] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1704282384.583219839] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1]  03/01 11:46:24,589 ERROR [547474106592] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video0 Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,591 ERROR [547474106592] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video1 Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,592 ERROR [547474106592] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video2 Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,593 ERROR [547474106592] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video3 Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,594 ERROR [547474106592] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video4 Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,596 ERROR [547474106592] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video5 Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,597 ERROR [547474106592] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video6 Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,598 ERROR [547474106592] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video7 Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,599 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,599 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,599 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,600 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,600 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,600 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,600 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,600 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,601 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,601 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,601 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,601 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,601 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:24,601 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1] [WARN] [1704282384.619769803] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1]  03/01 11:46:30,623 ERROR [547474106592] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video0 Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:30,623 ERROR [547474106592] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video1 Last Error: Operation not permitted
[... cut ...]
[realsense2_camera_node-1]  03/01 11:46:48,760 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:48,760 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1]  03/01 11:46:48,760 ERROR [547474106592] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: Operation not permitted
[realsense2_camera_node-1] [WARN] [1704282408.783816517] [camera.camera]: No RealSense devices were found!
```

Any idea on this one?

Thanks!

MartyG-RealSense, 2024-01-03:
There was a Pi and ROS2 case at https://github.com/IntelRealSense/realsense-ros/issues/2545#issuecomment-1741687576 that had this **Mipi device capability could not be grabbed** error.  Installing librealsense from source code with a build script called **libuvc_installation.sh** resolved the problem in their particular case, as described at https://github.com/IntelRealSense/realsense-ros/issues/2545#issuecomment-1741993966

2b-t, 2024-01-03:
I just happened to notice that contrary to @MartyG-RealSense code snippet, you seem to be missing the **`--privileged`** flag (and neither gave he container access rights to the Realsense with dedicated `device_cgroup_rules`). So the operation not permitted error might as well just be caused by that. So try simply adding `--privileged` before giving Marty's solution a go.

JLBicho, 2024-01-03:
Hello,

I've tried both thing, privileged with cgroups and compiling from source the libuvc and still the same error with MIPI.

I'll keep trying things and let you know. If you happen to come around some similar error, please let me know.

Thank you!

ciandonovan, 2024-01-05:
@JLBicho what does your `ls -l /dev` look like? Is your docker user in the `video` group?

If you're willing to run the development branch, [this](https://github.com/IntelRealSense/librealsense/pull/11900) pull request allows you to pass into the container individual `/dev/videoX` devices without needing any extra privileges or volume/bind mounts.

`docker run -it --name humble_realsense --net=host --env="ROS_DOMAIN_ID=42" --device=/dev/video0:/dev/video0:rw --device=/dev/video1:/dev/video1:rw ros2humble_realsense:latest bash`

Obviously if you have more video devices just add more `--device` arguments in the same format.

JLBicho, 2024-01-11:
@ciandonovan, thank for the suggestion.

However, I still get the following error:

```[realsense2_camera_node-1]  11/01 17:23:21,655 ERROR [547247417568] (librealsense-exception.h:52) Mipi device capability could not be grabbed Last Error: No such file or directory
```

I don't think it has to do with the mounting of the device, rather some problem with the installation of librealsense.

MartyG-RealSense, 2024-01-17:
Hi @JLBicho  Do you have an update about this case that you can provide, please?  Thanks!

JLBicho, 2024-01-18:
Hello @MartyG-RealSense,

I haven't been able to try more things.

I'm trying to get a D400 to check if the problem is related only to the L515 or else. I'm also planning on changing the O.S. of the Rpi.

I'll let you know any advances.

MartyG-RealSense, 2024-01-19:
Thanks very much.  I look forward to your next report.  Good luck!

MartyG-RealSense, 2024-02-02:
Hi @JLBicho  Do you require further assistance with this case, please?  Thanks!

JLBicho, 2024-02-02:
Hi @MartyG-RealSense !

For now I'm a little stuck. I'm not able to make it work with either L515 or D435.

I'll try to reinstall the O.S. with Ubuntu instead of Raspbian.

Anyway, thank you for your support. I'll let you know if I finally get it working.

If you wish to close this issue, please do so.

MartyG-RealSense, 2024-02-03:
Thanks very much for the update.  As you suggested, I will close the issue.  Please feel free to re-open it at a future date or create a new issue if you need to do so.  Good luck!

closed 2024-02-03 by MartyG-RealSense

sladomic, 2024-05-07:
I might have the same issue, but it worked in all old versions and still does. Only when I upgrade to 2.55.1.6486 it stops working.

I use a python:3.7-slim image on a NUC with Ubuntu installed.

MartyG-RealSense, 2024-05-08:
Hi @sladomic  When you upgraded to librealsense SDK 2.55.1, did you also update the camera's firmware driver to **5.16.0.1** which is the recommended firmware for 2.55.1.  Using older firmware versions than the recommended can result in errors.  And the previous SDK version 2.54.2 was mostly bug-fixes and so was not very different to 2.54.1, whilst 2.55.1 has a large number of feature and API changes.

sladomic, 2024-05-08:
Hi @MartyG-RealSense

Thanks for the fast reply. I haven‘t updated the firmware driver since I use a L515, which isn’t supported anymore officially.

However I can still use it when not upgrading to 2.55.1.

MartyG-RealSense, 2024-05-08:
Thanks very much for the information about your camera model.  Yes, L515 support is removed in 2.55.1 onwards.