realsenseai/librealsense #12553: D435i showing as D410i; showing error ISP update failed

xuxu8933, 2024-01-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |  Ubuntu 20.04 | 
| Platform                                 | PC.  |
| SDK Version                            |  2.54.2.0                        | 

### Issue Description
I recently try to use my realsense d435i, but the rgb camera won't be opened. The realsense viewer report error `isp update failed` and the camera is recognized as d430i. 
I have tried following options:
- different port, cables or operating systems
- downgrade the firmware to 5.12.15.50

After all, the same error reported when connect the camera. Could you help me on this issue?

MartyG-RealSense, 2024-01-03:
Hi @xuxu8933  Firmware version **5.15.1** is the correct one to use with RealSense Viewer version 2.54.2.  

A camera is identified by the firmware as D430i if it is unable to detect the RGB sensor.  This can occur because of a problem with the firmware or - in extremely rare cases - the internal cable connecting the RGB sensor to the camera's circuit board becoming disconnected.  A problem with the firmware installation is therefore a more likely cause.

In a previous case of the **isp update failed** message in 2023 at https://github.com/IntelRealSense/librealsense/issues/11963#issuecomment-1641846056 where a D415 was recognized as a D410 (a D415 without a detected RGB sensor), my Intel RealSense colleagues recommended two courses of action:

1.  Opening up the camera and replacing the **Vision Processor D4** circuit board.  As you have a D435i, the **V2** version of the board would have to be used.  Doing so would invalidate the camera's sales warranty though if it is still within its 1 year warranty period.

2.  Alternatively, mail the D435i camera to Intel so that my Intel RealSense colleagues can attempt to recover the camera or replace it.  If you wish to pursue this option, may I ask which country you are located in please?

xuxu8933, 2024-01-03:
Hi @MartyG-RealSense, if possible we would like to repair or replace it. We are located in Germany.

MartyG-RealSense, 2024-01-03:
Because you are in Europe, Intel will need an **EORI** number to ship to you.  If you do not already have an EORI number, the link below has information about applying for one in Germany.

https://verwaltung.bund.de/leistungsverzeichnis/EN/leistung/99122042001000

xuxu8933, 2024-01-09:
> Because you are in Europe, Intel will need an **EORI** number to ship to you. If you do not already have an EORI number, the link below has information about applying for one in Germany.
> 
> https://verwaltung.bund.de/leistungsverzeichnis/EN/leistung/99122042001000

Hi @MartyG-RealSense The EORI number is DE3298116. Please let me know if you need more info.

MartyG-RealSense, 2024-01-09:
Thank you very much for the EORI number.  I will talk to my Intel RealSense colleagues regarding sending your D435i camera by mail to Intel to attempt recovery.

MartyG-RealSense, 2024-01-10:
My colleagues agreed that your camera should be returned by post to Intel to attempt recovery.  Please send me an email at the address below so that I can provide you with the postal address.

martyx.grover@intel.com

MartyG-RealSense, 2024-02-20:
Hi @xuxu8933  After a camera was shipped to you, can this case be closed now please or do you require further assistance?  Thanks!

MartyG-RealSense, 2024-02-27:
Hi @xuxu8933  Have you received a replacement camera, please?  Thanks!

xuxu8933, 2024-02-27:
Hi, the camera is received and working now. Thank you for your help.

closed 2024-02-27 by MartyG-RealSense