realsenseai/librealsense #12560: Synchronizing Emitter ON-OFF pattern using hardware sync cable

1005088h, 2024-01-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D434IF | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |    Ubuntu 22  | 
| Kernel Version (Linux Only)    |  5.15.0-89-generic                                         | 
| Platform                                 | PC/docker |
| SDK Version                            |  2.54.2 }                          | 
| Language                            |  C++/Rust  | 
| Segment			|  Robot  | 

### Issue Description
I want to synchronize two D434IF cameras both the camera exposure and the emitter ON-OFF pattern when the emitter_on_off option is turned on. I connect the two camera with hardware sync cable following the step https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration. 

What I achieved:
I use the realsense-viewer and view the two camera IR images and randomly pause. I can see sometimes the two IR image are in synced(ON-ON and OFF-OFF at the same time instance when paused), and sometimes out-of-synced(ON-OFF and OFF-ON at the same time).

[image]
OFF-OFF(in synced)

[image]
ON-ON(in synced)

[image]
ON-OFF(out of synced)

To confirm that the hardware sync is working only for camera exposure but not emitter. I randomly restart the slave 60 times.
32 are in synced and 28 are out of synced. It's seems the assumption is true.

What I want to achieve is 
Make sure the emitter ON-OFF pattern always in synced. Do you know is there any solution for this?

MartyG-RealSense, 2024-01-08:
Hi @1005088h  If Inter Cam Sync Mode value '259' is used with the Slave camera then two frames per trigger are sent with laser ON-OFF.  If mode '260' is used then two frames per trigger are sent with laser OFF-ON.

Unlike slave mode '2', in modes 4 and higher the camera listens indefinitely for a trigger pulse, then performs the capture and starts waiting again until the next trigger pulse.  With mode '2', the camera only listens for a trigger for a certain time period on each frame and then initiates an unsynced capture if a trigger is not detected within that timeframe.

Modes 4 and above are known as **genlock** hardware sync.  They are no longer officially supported by Intel, with modes 1 and 2 being the only officially supported and validated modes, but modes 4+ still work in the RealSense Viewer and the RealSense SDK.

1005088h, 2024-01-08:
Hi @MartyG-RealSense, thanks for the response. Can I confirm that I should set one camera to mode '1' as a master and the other to mode '259' or '260' as a slave.  

> Hi @1005088h If Inter Cam Sync Mode value '259' is used with the Slave camera then two frames per trigger are sent with laser ON-OFF. If mode '260' is used then two frames per trigger are sent with laser OFF-ON.

To my understanding, the 'master' will follow the 'slave' mode for the ON-OFF / OFF-ON pattern according to the '259' or '260 '. In this case, the master also listen to the slave?

> When using a genlock mode, the slave camera's FPS should be set to 2 times the trigger's frequency. 

Can you elaborate more on this? Why the slave camera's FPS should be set to 2 times the trigger's frequency?  The master and slave are in different frame rate?

> So if the master camera is set to 30 FPS then the slave should be set to 60 FPS.

I think there is not 60 FPS for resolution 848x480, correct me if I am wrong. I can only see 6/15/30/90 Hz. If this the case, I want a 30hz both from master and slave, what can I do?

To my understanding, if I want the emitter to flash in 30hz in sync mode, 15hz ON and 15hz OFF. I should do the setting something like this?

| Option | Master   | Slave  |
|------------------------|-----------------------|-------------------- |
| InterCamSyncMode  | 1  |  259 |
| EmitterOnOff  | 1  |  1 |
| GlobalTimeEnabled  | 1  |  1 |
| Framerate  | 30  |  60 |

MartyG-RealSense, 2024-01-08:
If you are not using an external trigger signal produced by a signal generator device then one of the cameras should be set to Inter Cam Sync Mode '1' to designate it as a Master camera.

It is recommended that the Master camera is enabled first and the Slave camera enabled second.

The slave camera follows the sync timing of the master.  The master does not listen to the slave and instead provides the sync trigger pulse.

In regard to why the slave camera's FPS is set to 2x the master, the reason is complicated.  I will quote the genlock system's white-paper document (which is a different white paper to the Multi Camera one that you have been using).

*********

The native frame rate (of the slave camera) sets an upper limit on the frame rate that can be supported by an external trigger.  When sending a trigger pulse train, the trigger signal cannot be repeated faster than half the native camera mode frame rate.  In other words, once a trigger is received the camera will be blind to any other trigger signals for 2x the frame time of the "native mode" of the camera.

********

The genlock sync white-paper is no longer available because Intel no longer officially supports genlock mode, but I have an archived PDF document copy that you can download using the link below.

[External Synchronization.pdf](https://github.com/IntelRealSense/librealsense/files/13861706/External.Synchronization.pdf)

D435i should support 60 FPS at 848x480.  It is 1280x720 resolution that only supports up to 30 FPS.

Regarding your table, Yes, set master to 1 and slave to 259 if you are generating your trigger signal from the master camera instead of an external signal generator.  

Yes, set master to 30 FPS and slave to 60 FPS.  

Global Time will be enabled by default on RealSense 400 Series cameras so you should not need to manually set it to '1'.  

Likewise, you should not need to set Emitter On Off to '1' as far as I am aware if you are using mode 259 as the mode should handle the on-off for you.

As the emitter on-off is on a per-frame basis, I would think (though I am not certain) that it would be more difficult to sync the on-off of the master and slave cameras up if one of the cameras had frame drops or lag.

When Global Time is enabled and there are multiple cameras, it should attempt to calculate a common timestamp for all of the cameras, as described at https://github.com/IntelRealSense/librealsense/pull/3909 but I do not know the effect that Global TIme would have on emitter sync when using mode 259 / 260 as these mode numbers were hardly ever used.

MartyG-RealSense, 2024-01-09:
In modes 259 and 260, two frames per trigger are released by the slave camera, one frame with the emitter off and the other with the emitter on.

I believe though that multiple frames are only generated by the slave camera, and not both master and slave.  I would assume that the On and Off frames will both have the same timestamp since they relate to the same frame, and the timestamp of those frames would sync to that of the master camera.

If you wanted both cameras to generate an on and off frame, they would probably both have to be set to slave (259) and the master trigger signal be generated by an external signal generator device instead.

MartyG-RealSense, 2024-01-20:
Do you require further assistance with this case, please?  Thanks!

1005088h, 2024-01-21:
I tested with 259 and 260 modes, and both could not sync the emitter ON-OFF pattern. I will try to restart the slave until they are synced. Currently, we don't have a plan to use an external clock. Just a proposal, our company is going to use a lot of real sense in synced mode, is it possible to add a synced emitter feature to the driver and firmware? It will be beneficial for our product development.

MartyG-RealSense, 2024-01-21:
I have passed your feature request to my Intel RealSense colleagues.

MartyG-RealSense, 2024-01-22:
Hi @1005088h  My colleagues have informed me that multiple camera sync of emitter on-off will not be a supported feature, unfortunately.

MartyG-RealSense, 2024-01-28:
Hi @1005088h  Bearing in mind the reply in the above comment, do you require further assistance with this case please?  Thanks!

1005088h, 2024-01-28:
> Hi @1005088h  Bearing in mind the reply in the above comment, do you require further assistance with this case please?  Thanks!

Hi @MartyG-RealSense , thanks for the information, we can close this issue.

MartyG-RealSense, 2024-01-28:
Okay, thanks very much for the update!  As you suggest, I will close the issue.  Thanks again!

closed 2024-01-28 by MartyG-RealSense