realsenseai/librealsense #12562: L515 Cannot open

CCCCCCCT, 2024-01-09:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:572

```

I have already run version 2.21.0 and it can run, but my current device L515 does not support it. rs-capture.exe also works. I wonder if anyone can help me?

MartyG-RealSense, 2024-01-09:
Hi @CCCCCCCT  For the earliest support of L515, please try running version 2.35.2.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.35.2

CCCCCCCT, 2024-01-12:
I successfully ran 2.50.0, thank you very much for your help.

MartyG-RealSense, 2024-01-12:
You are very welcome.  I'm pleased that I could help.  Thanks very much for the update!

closed 2024-01-20 by MartyG-RealSense