realsenseai/librealsense #12565: [140041002829568] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

ranjithlesoko, 2024-01-10:
[140041002829568] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
if run roslaunch realsense2_camera opensource_tracking_tk_online.launch 
work  but in roslaunch realsense2_camera rs_camera.launch not work showin this error and stuck 

[140041002829568] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2024-01-10:
Hi @ranjithlesoko  The **opensource_tracking_tk_online.launch** file imports settings from rs_camera.launch, so it should be okay to use this launch file instead of rs_camera if it works for you.

The contents of opensource_tracking_tk_online.launch can be read at https://github.com/IntelRealSense/librealsense/issues/11753#issuecomment-1545629996

The official Intel-provided version of the above launch file is called **opensource_tracking.launch**

`roslaunch realsense2_camera opensource_tracking.launch`

MartyG-RealSense, 2024-01-17:
Hi @ranjithlesoko  Do you require further assistance with this case, please?  Thanks!

ranjithlesoko, 2024-01-19:
i sorted all previous issue now i got this (Confidence, 0) sensor isn't supported by current device  the rs_camera.launch not work

[ INFO] [1705644299.204512178]: Initializing nodelet with 12 worker threads.
[ INFO] [1705644299.354141764]: RealSense ROS v2.3.2
[ INFO] [1705644299.354171455]: Built with LibRealSense v2.54.2
[ INFO] [1705644299.354185045]: Running with LibRealSense v2.54.2
[ INFO] [1705644299.373222028]:  
[ INFO] [1705644299.387684826]: Device with serial number 102422073962 was found.

[ INFO] [1705644299.387705236]: Device with physical ID /sys/devices/pci0000:00/0000:00:08.1/0000:13:00.3/usb4/4-2/4-2:1.0/video4linux/video0 was found.
[ INFO] [1705644299.387715286]: Device with name Intel RealSense D435I was found.
[ INFO] [1705644299.387933192]: Device with port number 4-2 was found.
[ INFO] [1705644299.387951802]: Device USB type: 3.2
[ INFO] [1705644299.387967933]: Resetting device...
[ INFO] [1705644305.479581711]:  
[ INFO] [1705644305.494351624]: Device with serial number 102422073962 was found.

[ INFO] [1705644305.494381485]: Device with physical ID /sys/devices/pci0000:00/0000:00:08.1/0000:13:00.3/usb4/4-2/4-2:1.0/video4linux/video0 was found.
[ INFO] [1705644305.494397555]: Device with name Intel RealSense D435I was found.
[ INFO] [1705644305.494584270]: Device with port number 4-2 was found.
[ INFO] [1705644305.494603270]: Device USB type: 3.2
[ INFO] [1705644305.495917733]: getParameters...
[ INFO] [1705644305.519655387]: setupDevice...
[ INFO] [1705644305.519670458]: JSON file is not provided
[ INFO] [1705644305.519681928]: ROS Node Namespace: camera
[ INFO] [1705644305.519691628]: Device Name: Intel RealSense D435I
[ INFO] [1705644305.519699468]: Device Serial No: 102422073962
[ INFO] [1705644305.519708519]: Device physical port: /sys/devices/pci0000:00/0000:00:08.1/0000:13:00.3/usb4/4-2/4-2:1.0/video4linux/video0
[ INFO] [1705644305.519717419]: Device FW version: 5.15.1
[ INFO] [1705644305.519726039]: Device Product ID: 0x0B3A
[ INFO] [1705644305.519733659]: Enable PointCloud: Off
[ INFO] [1705644305.519746090]: Align Depth: Off
[ INFO] [1705644305.519753980]: Sync Mode: Off
[ INFO] [1705644305.519773450]: Device Sensors: 
[ INFO] [1705644305.520909379]: Stereo Module was found.
[ INFO] [1705644305.524983769]: RGB Camera was found.
[ INFO] [1705644305.525099732]: Motion Module was found.
[ INFO] [1705644305.525134332]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1705644305.525158923]: num_filters: 0
[ INFO] [1705644305.525174883]: Setting Dynamic reconfig parameters.
[ INFO] [1705644305.560895734]: Done Setting Dynamic reconfig parameters.
[ INFO] [1705644305.561281183]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1705644305.562336479]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1705644305.562357230]: setupPublishers...
[ INFO] [1705644305.563914718]: Expected frequency for depth = 30.00000
[ INFO] [1705644305.582565006]: Expected frequency for color = 30.00000
[ INFO] [1705644305.594253494]: setupStreams...
[ INFO] [1705644305.619557777]: SELECTED BASE:Depth, 0
[ INFO] [1705644305.626012197]: RealSense Node Is Up!
[ WARN] [1705644305.717918280]: 
stuck on this  not opening rviz

MartyG-RealSense, 2024-01-19:
Most users get the **(Confidence, 0) sensor isn't supported by current device! -- Skipping** message in their log so it is not something to be concerned about.   I believe that it relates to a feature of the **RealSense T265 Tracking Camera** model, and so would not be usable by RealSense 400 Series cameras.  So your RViz problem should not be related to this message.

Wrapper 2.3.2 should be used with librealsense version 2.50.0 or 2.51.1, not 2.54.2.

On the rs_camera.launch file, the IMU is disabled by default.  So it should be enabled in the roslaunch instruction.

`roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation`

ranjithlesoko, 2024-01-20:
ok try to make the version 2.51.1

ranjithlesoko, 2024-01-20:
[ INFO] [1705730996.911927500]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1705730996.911943101]: num_filters: 0
[ INFO] [1705730996.911953201]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ WARN] [1705730996.934716599]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1705730996.941862454]: Done Setting Dynamic reconfig parameters.
[ INFO] [1705730996.942237293]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1705730996.943254398]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1705730996.943272018]: gyro stream is enabled - fps: 200
[ INFO] [1705730996.943284018]: accel stream is enabled - fps: 63
[ INFO] [1705730996.943295729]: setupPublishers...
[ INFO] [1705730996.944904488]: Expected frequency for depth = 30.00000
[ INFO] [1705730996.964834527]: Expected frequency for color = 30.00000
[ INFO] [1705730996.977022114]: Start publisher IMU
[ INFO] [1705730996.977407074]: setupStreams...
 20/01 11:39:56,978 WARNING [140586343036672] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ INFO] [1705730996.992189237]: SELECTED BASE:Depth, 0
[ INFO] [1705730996.999658110]: RealSense Node Is Up!
[ WARN] [1705730997.071429988]:  i got this error

MartyG-RealSense, 2024-01-20:
If you are referring to the 'IMU Calibration is not available' message, this does not prevent the IMU from being used by ROS.

ranjithlesoko, 2024-02-01:
sorry @MartyG-RealSense  for late replay i completed all process done  in my pc now i go to jetson nx i got lot of error  first i want to know jetpack 5.1 L4T 35.2.1 support or not  i done  realsense-viewer work fine but in ros-Wrapper i got less then 15 fps how to improve this on  jetson

MartyG-RealSense, 2024-02-01:
The **patch-realsense-ubuntu-L4T.sh** Jetson patch script supports JetPack 5.0.2 (L4T 35.1).  However, the librealsense SDK can be used with Jetsons with an unsupported JetPack version if you either (a) do not apply a kernel patch, or (b) build librealsense from source code with CMake with the flag **-DFORCE_RSUSB-BACKEND=ON** included in the CMake build instruction.

When building from source code for Jetson, the flag **-DBUILD_WITH_CUDA=ON** should also be included.  This will build librealsense with CUDA support enabled for improved performance when using depth-color alignment or pointclouds.  Enabling it in the librealsense build automatically enables it in the ROS wrapper too.

ranjithlesoko, 2024-02-02:
ok i try to flash jetpack 5.0.2

ranjithlesoko, 2024-02-02:
which version libsense 2.50 or 2.52

MartyG-RealSense, 2024-02-02:
JetPack 5.0.2 support was added in librealsense 2.54.1.

closed 2024-02-16 by MartyG-RealSense