realsenseai/librealsense #12569: Realsense Depth camera D455 giving unstable gyro data

Muthukumar4796, 2024-01-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC NVIDIA GPU  |
| SDK Version                            |  { legacy / 2.}                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I am using Realsense Depth camera D455 for IMU data, but IMU data is not stable.
when sensor is not moving it should give stable values but z values are increasing for every second.

MartyG-RealSense, 2024-01-12:
Hi @Muthukumar4796  The raw data of a RealSense IMU is 'noisy' by nature.  You can calibrate the IMU using a Python tool.

https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-imu-calibration/rs-imu-calibration.py

https://www.intelrealsense.com/wp-content/uploads/2020/07/IMU_Calibration_Tool_for_Intel_RealSense-Depth_Cameras_Whitepaper.pdf

Muthukumar4796, 2024-01-13:
Thank you for responding.
I calibrated our Realsense D455 cameras and have attached the results. Could you please check them? Also, could you provide me with insights and procedures on how to use the accl_2.txt and gyro_02.txt calibrated files to obtain noise-free data?
[image]

MartyG-RealSense, 2024-01-13:
The calibration results appear to be very good.  It would be difficult to obtain better results than the values that are listed, so I would recommend not repeating the calibration.

When using the RealSense Viewer tool, you can take advantage of the saved IMU calibration by going to the Motion Module section of the options side-panel, expanding open the Controls category and ensuring that the **Enable Motion Correction** option is enabled (blue).

[image]

closed 2024-01-26 by MartyG-RealSense