realsenseai/librealsense #12570: Cannot build with flag BUILD_NETWORK_DEVICE=ON

yovamo, 2024-01-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                        | 
| Platform                                 | Jetson Nano  |
| SDK Version                            | 2.54.2                         | 
| Language                            |  python }                          | 
| Segment			|  Robot }                          | 

### Issue Description
In attempt to build latest available SDK (v2.54.2) on the Jetson Nano with the following configuration
`cmake ../ -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false -DBUILD_NETWORK_DEVICE=ON -DBUILD_WITH_CUDA=true        `
I've received the following WARNING for the BUILD_NETWORK_DEVICE:
```
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.54.2
-- Setting Unix configurations
-- Info: Building with CUDA requires CMake v3.8+
-- The CUDA compiler identification is NVIDIA 10.2.300
-- Check for working CUDA compiler: /usr/local/cuda/bin/nvcc
-- Check for working CUDA compiler: /usr/local/cuda/bin/nvcc -- works
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found CUDA: /usr/local/cuda (found version "10.2")
-- CUDA_LIBRARIES: /usr/local/cuda/include /usr/local/cuda/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/aarch64-linux-gnu/librt.so;/usr/local/cuda/lib64/libcusparse.so;/usr/local/cuda/lib64/libcublas.so
-- CHECK_FOR_UPDATES depends on BUILD_GRAPHICAL_EXAMPLES flag, turning it off..
-- Performing Test SUPPORTS_CXX14
-- Performing Test SUPPORTS_CXX14 - Success
-- using RS2_USE_V4L2_BACKEND
-- Found usb: /usr/lib/aarch64-linux-gnu/libusb-1.0.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'libudev'
--   Found libudev, version 237
-- Found Udev: /usr/include
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Fetching recommended firmwares:
-- disable link time optimization for fw project
-- ... https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-5.15.1.0.bin
-- ... https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- ... https://librealsense.intel.com/Releases/L5xx/FW/L51X_FW_Image-1.5.8.1.bin
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_NETWORK_DEVICE

```
After building and install, I tried to command rs-server in Jetson nano, but I get:
`-bash: rs-server: command not found`

Running rs-enumerate-devices shows the camera is connected correctly.

Running sudo systemctl restart rs-server.service gives: Failed to restart rs-server.service: Unit rs-server.service not found.

MartyG-RealSense, 2024-01-12:
Hi @yovamo  The rs-server networking interface was removed in librealsense 2.54.1 and so the **-DBUILD_NETWORK_DEVICE=ON** CMake flag will not work any more.  If your project requires the network interface then librealsense version **2.53.1** or **2.51.1** will need to be installed instead of 2.54.2.  

It is planned that a new networking system to replace the removed rs-server one will be introduced in an upcoming librealsense release.

yovamo, 2024-01-12:
Thanks @MartyG-RealSense. This can be closed!

closed 2024-01-12 by yovamo